JMRI/JMRI #4616: New CMRI features in 4.10

JLang1951, 2017-12-23:
I installed 4.10 the other day and I like all of the new features for monitoring and configuring CMRI nodes but I have noticed one problem that I want to report. When I pull up the list asignments for my SIMNI nodes the user names for the I/O's show up in the comments column and not the user names column. Not a big deal but thought that I would point it out. Here is my panel if someone wants to look at it.
[JDL_NYC_Panel.zip](https://github.com/JMRI/JMRI/files/1584197/JDL_NYC_Panel.zip)
NCE Simulated
CMRI Simulated.

Thanks John L.

CCatania01, 2017-12-23:
The list assignments from the main C/MRI menu display is incorrect.  Listing bit assignments for a node from the Node Configuration Manager, displays the information correctly.
Will be corrected in a 4.11.x release.

KenC57, 2017-12-23:
I must have been looking at the node config, what other display was there??
-ken c

CCatania01, 2017-12-24:
From the C/MRI menu, List Assignments.  This is the original function where you selected the node address to display the port assignments.
Chuck

CCatania01, 2018-01-26:
Corrected.  PR #4824

CCatania01, 2018-02-03:
Would JLang1951 please review, and close.
Thanks,
Chuck Catania

closed 2018-02-03 by JLang1951