JMRI/JMRI #4618: NmraConsistManager now requires a CommandStation

rhwood, 2017-12-23:
See [discussion on the Users groups](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/143976;_ylc=X3oDMTM4ZmE1am1lBF9TAzk3MzU5NzE0BGdycElkAzY2MDI3NTgEZ3Jwc3BJZAMxNzA1MDYzNTc2BG1zZ0lkAzE0Mzk4NgRzZWMDZnRyBHNsawN2dHBjBHN0aW1lAzE1MTQwNDI1MjQEdHBjSWQDMTQzOTc2).

It appears that changes in the NmraConsistManager to require a CommandStation now cause some connections that do not provide CommandStation objects to fail.

This (so far) is know to affect TMCC users in 4.10.

@pabender This appears to be a side effect of #4410. Do have specific advice for addressing this?

pabender, 2017-12-23:
The NmraConsistManager has ALWAYS required an NMRA Command Station object to function.  TMCC does not now,and never has, provided this.  

Similarly, the DCC consist manager has always required that an ops mode programmer be present, and TMCC  does not, and never has, provided an OpsMode programmer.

The suggestion in this case is to set the consist selection in preferences to internal and then set the WiThrottle server to use the DCC Brand Specific consists.

Eventually, the plan is to remove the option from the WiThrottleServer.  It uses its own implementation of a consist manager, but there are other things that need to be modified without changing the external behavior before that can happen.

pabender, 2017-12-23:
Incidentally, the specific issue here is with the WiThrottle implementation of the consist manager (WiFiConsistManager), which looks for a default for the CommandStation object at initialization time.  The Internal manager does not provide that (it does provide an internal OpsMode Programmer).

This bit of behavior is unchanged in WiFiConsistManager, though it was refactored as a first step towards removing it during the process that lead to #4410.

KenC57, 2017-12-23:
TMCC doesn't have a consist manager, only throttle and turnout.
-ken c

pabender, 2018-01-08:
Closed via #4620

closed 2018-01-08 by pabender