JMRI/JMRI #4621: TMCC does not save changes to connection port

KenC57, 2017-12-24:
I've stepped through a bunch of this and when I inspect things, the port name from the selection shows up. But when it writes the file, is has '(none selected)' instead. Also the connection name and prefix keeps changing by getting +1 to the existing value. It seems it keep creating a new connection memo somewhere along the way and that gets a 'one more' than the prior value.

Does this behavior sound like something anyone else has seen??
-ken c

pabender, 2017-12-24:
On 12/23/2017 08:30 PM, Ken Cameron wrote:
> I've stepped through a bunch of this and when I inspect things, the
> port name from the selection shows up. But when it writes the file, is
> has '(none selected)' instead. Also the connection name [...]
Yes.

Check and make sure that the ConnectionConfig only creates a new adapter 
in setInstance if the adapter is null.
Also, verify that the ConnectionConfigXml has over-ridden both versions 
of getInstance (the one that takes a parameter and the one that does not.)

It's likely the ConnectionConfigXml that is the issue.

Paul

KenC57, 2017-12-24:
Paul,

Yes it seems something is creating instead of reusing. The hard part of
tracing is very little code was written for the TMCC system. Almost all the
config parts are the abstract/common code. So it would look like either
everything else is overriding the abstract that's doing something wrong or
this is one of the few and trips up. I'm tired enough today so I'm not
digging further tonight. Will resume in the morning.
-ken c

pabender, 2017-12-29:
@KenC57 

The cause for this is almost certainly the missing method in the ConnectionConfigXml file.  I've seen this cause the problem you describe.

I am also wondering if that could contribute to:

    [junit] Testcase: testLaunchTmcc(apps.DispatcherPro.DispatcherProTest):	Caused an ERROR
    [junit] Unexpected ERROR or FATAL messages emitted
    [junit] java.lang.AssertionError: Unexpected ERROR or FATAL messages emitted
    [junit] 	at apps.tests.Log4JFixture.tearDown(Log4JFixture.java:38)
    [junit] 	at jmri.util.JUnitUtil.tearDown(JUnitUtil.java:118)
    [junit] 	at apps.DispatcherPro.DispatcherProTest.tearDown(DispatcherProTest.java:157)
    [junit] 	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
    [junit] 	at java.lang.Thread.run(Thread.java:748)

The runtime log is actually:

    [junit] INFO  - Starting with profile TMCC Simulator.3f033c2b [Time-limited test] apps.Apps.?()
    [junit] ERROR - Proceeding to initialize class jmri.jmrit.decoderdefn.DecoderIndexFile but initialization is marked as complete [Time-limited test] jmri.InstanceManager.?()
    [junit] java.lang.Exception: Thread "initialize decoder index"
    [junit] 	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:257)
    [junit] 	at jmri.InstanceManager.getDefault(InstanceManager.java:206)
    [junit] 	at apps.Apps.lambda$new$3(Apps.java:371)
    [junit] 	at java.lang.Thread.run(Thread.java:748)
    [junit] WARN  - Cleaning up frame "DispatcherPro" (a class jmri.util.JmriJFrame) in apps.DispatcherPro.DispatcherProTest. [Time-limited test] jmri.util.JUnitUtil.?()

So this isn't the only issue, but it is certainly possible that the issue with TMCC preferences sometimes generates an error window when initialization the application ( and I have seen this for every application we have setup these tests for, not just the example above )

KenC57, 2017-12-30:
Paul, you guessed it!! It was one of the getInstance() missing, the one with the object being passed that was missing. Now when we get master cleared again, I can fix it.

KenC57, 2018-01-03:
Is fixed via PR#4663 and will be 4.11.2

closed 2018-01-03 by KenC57