JMRI/JMRI #4628: Unable to build Windows installer/launcher on macOS

rhwood, 2017-12-24:
makensis on macOS creates a bad build for the JMRI installer for Windows and, presumably, the JMRI launcher for Windows. This failure is not reproducible using makensis on various versions of Linux.

rhwood, 2017-12-24:
Changing the verbosity of the NSIS builds in build.xml from `-V2` to `-V4` provides an interesting hint: the System.dll for NSIS on macOS is ~1/2 the size of that file on CentOS.

My Windows test system is 32 bit. It might be that macOS builds of the JMRI Windows installer only work on 64 bit systems.

dheap, 2017-12-24:
My macOS builds don't work properly on Win7 x64. Haven't tried Win7 x86.

The installer launches and seems fine until it gets to the Select Options window, where it goes non-responsive.

rhwood, 2018-03-16:
OP is no longer concerned.

dheap, 2018-03-16:
As far as I know, this is unresolved and I miss the ability to build Windows Installers for testing local changes on my VMs.

dheap, 2018-04-18:
@mattharris I've done some test builds of windows-package, both with the master `LaunchJMRI.exe`  and a new one built with `ant launchjmri.exe` Either way I get an installer hang:
[image]
The logs, etc are here:
[package-windows-tests.zip](https://github.com/JMRI/JMRI/files/1922588/package-windows-tests.zip)
The actual builds are in here:
https://www.dropbox.com/sh/hvg00jieab4a46i/AADKei6Jvy_XyshwLFSbzg2ba?dl=0

Can you see anything untoward in the logs, etc.?

dheap, 2018-11-15:
I've now got this working fine:

- MacOS 10.10.5 Yosemite
- [Homebrew](https://brew.sh) `brew install makensis --with-large-strings`
- Entry in "local.properties" `nsis.home=/usr/local/bin/`

[Homebrew's makensis](https://formulae.brew.sh/formula/makensis) installed  version 3.03.

closed 2021-04-14 by stale[bot]