JMRI/JMRI #4632: Create Test Release 4.11.2

bobjacobsen, 2017-12-25:
Second in the usual sequence of test releases leading to 4.12 in June or July 2018. Made from HEAD of master.

bobjacobsen, 2018-01-18:
Now planning to build late (California time) Saturday January 20th from master, including all PRs ready to merge then.

bobjacobsen, 2018-01-21:
The release-4.11.2 branch has been created. 

From now on, please document your changes in the [jmri4.11.3.shtml](https://github.com/JMRI/website/blob/master/releasenotes/jmri4.11.3.shtml) release note file.

Maintainers, please set the 4.11.3 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](http://jmri.tagadab.com/jenkins/job/TestReleases/job/4.11.2/)

silverailscolo, 2018-01-21:
The macOS installer ran fine. But a part of PR #4795 slipped into this edition. If possible please include full PR in additional release to prevent startup error loading SPROG Simulator.

bobjacobsen, 2018-01-21:
> But a part of PR #4795 slipped into this edition.

I'd like to understand what happened in a bit more detail.  Could you point to one of the parts (i.e. file&line or commit hash) that got included?

It could have been because another PR was based on the `silverailscolo:eb-sprog-simulator` branch, and development overlapped.  Perhaps #4736? It was merged on Jan 12, so if you're started work on the #4795 change and pushed that into the `silverailscolo:eb-sprog-simulator` branch it would have been included.  But I'd like to confirm that.

silverailscolo, 2018-01-21:
@bobjacobsen 3 of my commits were merged into master as part of your PR #4799. I always have to watch out not to switch branches while there I have some uncommitted edits in IntelliJ, as they will move over to the new branch. So the same might have happened as you were switching branches assisting me.

bobjacobsen, 2018-01-23:
This has been released.

closed 2018-01-23 by bobjacobsen