JMRI/JMRI #4635: SPROG Simulator

silverailscolo, 2017-12-25:
Upon user request, add Simulator conection to SPROG.

Please assing to me, hoping to be assisted by Joff Hamilton.

crosland, 2017-12-28:
@silverailscolo Hi Egbert, I am the SPROG designer. I have never looked at JMRI simulators but I can help with questions about the communication protocol.

Andrew

silverailscolo, 2017-12-28:
Thanks, Andrew.
I've developed a step by step pattern to add a (dummy) simulator, but the essential part indeed is adding the system specific replies to messages.
Once I have the Sim basics operating I hope to come back to you for advice on that.

We could also arrange for you to fill in this code block as a new GitHub branch after the basic option has been merged.
> @silverailscolo <https://github.com/silverailscolo> Hi Egbert, I am the SPROG designer. I have never looked at JMRI simulators but I can help with questions about the communication protocol.
> 
> Andrew
>

silverailscolo, 2018-01-03:
@crosland could you please supply the syntax or - if possible - code suggestions for the Spog Simulator replies in this [branch](https://github.com/silverailscolo/JMRI/blob/c000bfe55dc4b80582c182b920083dbab477976d/java/src/jmri/jmrix/sprog/simulator/SimulatorAdapter.java#L255)?

This is a switch case construct for each received message, eg.
```
            case '?':                                    // this is the char/string making the request
                                                         // in the case of SPROG it's probably a hex value
                log.debug("Read_Sprog_Version detected");
                String replyString = "V0.1 - simulator"; // just my initial suggestion
                reply = new SprogReply(replyString);     // fake version number reply
                i = replyString.length();
                reply.setElement(i++, 0x0d);             // add EndOfMessageChar
                break;
```

silverailscolo, 2018-01-04:
@crosland, how does the SPROG confirm (reply to) a command? For now I use "OK" but that is not picked up. Below is the current set of messages to respond to, as seen in the SPROG Console:
```
    private SprogReply generateReply(SprogMessage msg) {

        SprogReply reply;
        char command = msg.toString().charAt(0);
        switch (command) {

            case 'O': // Send packet
                reply = new SprogReply(SPR_OK); // confirm
                break;

            case 'A': // Aquire Throttle command detected
                reply = new SprogReply(SPR_OK); // confirm
                break;

            case '>': // Set Throttle command detected
                reply = new SprogReply(SPR_OK); // confirm
                break;

            case '+': // TRACK_POWER_ON detected
                trackPowerState = true;
                reply = new SprogReply(SPR_OK); // confirm
                break;

            case '-': // TRACK_POWER_OFF detected
                trackPowerState = false;
                reply = new SprogReply(SPR_OK); // confirm
                break;

            case '?': // Read_Sprog_Version detected
                String replyString = "V0.1 - simulator";
                reply = new SprogReply(replyString); // fake version number reply
                break;

            default: // non-reply message detected
                reply = new SprogReply(""); // is there a default SPROG error reply?
        }
        return reply;
    }
```

crosland, 2018-01-04:
I can't find the branch silverailscolo in github desktop. I'm not a git "power user" so I may not be doing the right thing.

In the meantime, here are the string definitions from the SPROG source:
"\n\rSPROG II Ver 4.3\n\r" -reply to "?" version request command
"P> " - prompt
"= " - Reply to CV read,followed by value in decimal
 - Also reply to address, speed and mode commands if no argument given. Address and speed reply in decimal, Mode reply is hex
"!O" - Overload during programming
"!E" - Error response followed by an error code
"OK" - Successful CV write
"No Ack" - No Ack during CV read or write

The SPROG always replies to each message in sequence. After the above replies, the prompt will be sent.

rhwood, 2018-01-04:
- View this PR on GitHub
- Find the link that reads “open this on GitHub Desktop” in the bottom of the list of checks
- Click it, allow GitHub Desktop to open, and the branch will be on your desktop

crosland, 2018-01-07:
@rhwood Sorry but I still don't understand what I am to do. If I try to go to https://github.com/JMRI/JMRI/pull/4635 then I am redirected to this issue page. There is no PR page that I can see and no link to open the branch.
@silverailscolo You sent me links e.g. https://github.com/silverailscolo/JMRI/blob/c000bfe55dc4b80582c182b920083dbab477976d/java/src/jmri/jmrix/sprog/simulator/SimulatorAdapter.java#L255 but the edit icon is inactive and the tool tip saye I must be on a branch to make or propose changes to the file.
I would rather make the edits on a local branch so that I can build and check for typos befor committing.

Sorry if I am being thick and have missed something.

silverailscolo, 2018-01-08:
When you go [here](https://github.com/silverailscolo/JMRI/tree/eb-sprog-simulator) or move up from the earlier link I sent, a _Clone_ button appears at top right. That's needed for you to receive a copy of my branch for you to check, compile and suggest PR's (to my branch, not to JMRI master directly).
Currently trying to make the simulated Sprog replies show up in the Sprog Console/Monitor, but they're not listening the same way the EasyDCC listens for the Simulator.

crosland, 2018-01-09:
@silverailscolo Sorry, you are going to have to give me clear explicit instructions. I cloned your branch to my machine. The jmrix/sprog/simulator path does not exist. Do I need to select a particular branch, rather than master? Which one? There are too many to guess from.

silverailscolo, 2018-01-09:
No problem, @crosland. The Sprog Simulator - like any code work - is in a branch named ``eb-sprog-simulator``, that was based on ``master`` (``eb-`` allows me to quickly spot my own stuff, ``-sprog-etc`` helps me remember what this is about...).
You can jump to the online version [from here](https://github.com/silverailscolo/JMRI/tree/eb-sprog-simulator).
The option to select the branch in GitHub Desktop is at top center of the working pane (and in your IDE as well). Perhaps you need to pull to get the latest stuff (that's just to the right in this screen shot):
[image]
BTW, it's now essentially operating, so feel free to pick it up!

silverailscolo, 2018-01-29:
Still needs to reply to hex messages to silence the repeated Keep Alive/Time out response.

silverailscolo, 2019-05-05:
SPROG Simulator was added in 4.11.2 with PR #4795. No issues reported since.

closed 2019-05-05 by silverailscolo