JMRI/JMRI #4636: NCE Macro Editor Reproducible Errors

Cliff-Anderson, 2017-12-26:
## Statement of the Problem:
Conditions defined here force NCE Macro Editor to consistently reproduce some errors:
1.	Macro Save action attempts to write 16 bytes of data, but tells NCE that it wants to write 129 bytes.  The Macro Backup action does not replicate this error!
2.	NCE Command Station replies with error code “4” which Macro Editor ignores  
3.	NCE Simulator reports an ERROR message to session.log, but fails to reply to the Macro Editor with the proper error code “4” response. 
4.	The NCE Monitor sees the reply of the error code “4” from the NCE Command Station as an unrecognized reply and simply displays `[34]  Reply: 34`  
5.	The NCE Macro Editor will persist in attempting to write the last 4 bytes of Macro data, but neither the NCE Simulator nor the NCE Command Station seem to place that data where it can be recovered.  
6.	The NCE: Edit NCE Macro window refreshes the displayed Macro command list to exactly what it was before any modifications were made.  
a.	Perhaps a better error recovery alternative would help?
7.	Depending upon the connection, a few seconds or even a few minutes after the erroneous write attempt, either the NCE Simulator behaves strangely or else the NCE Command Station software usually crashes.
a.	The ambiguity of only “usually” crashing may indicate that even more things are lurking?
8.	Macro Read data event performs exactly as if it is expected, however it will send eight **nuisance** ERROR messages to the session.log file each Macro it reads.  That means 256 * 8 = 2048 unneeded messages for the Macro Backup action.  

## Fair Disclosure:

- This report culminates an activity initiated by a request for help.  Command Station Macros were not in my normal toolbox.  The JMRI PanelPro tool called Routes has worked **extremely well** for the layouts that I have be involved with.  Command Station macros, have proven to be fragile, at least in some club layout environments.  
- There are three open threads in the Yahoo Users Group pertaining to Macro problems with NCE, and one unresolved thread on the [Jmri-reports] at SourceForge.  It is not clear to me that the developer activity that this report might cause would be assured to close any one of those threads.  
- The JMRI provided NCE Macro tools allowing Macro Backup and Macro Restore of all the macros to or from a single text file appear to work well and may even have provided some users with a “workaround” when complicated editing is not required.  
- But, since letting go of an unsolved puzzle is not one of my strong points, the “bothersome hangnail” request has progressed to this report.  
- Only the NCE PowerPro with EIA232 interface and the 2007 update chip were available for my part in this adventure.  The testing of the multitudinous implementation of NCE Interfaces and NCE Macros is beyond the scope of this note.  

## Test Configuration:
### Software: 
PanelPro JMRI version 4.11.1+Rfe44680 for Windows, 
Java SE Runtime Environment 8u151 (64 bit)
or
NetBeans IDE 8.2, 
Java SE Development Kit 8 Update 112 (64 bit), 
PanelPro version 4.11.1ish+cliffaa-rr+20171221T0708Z+Rab31aff064 (and in fact later Fetches)

### Equipment: 
Win10 64 bit laptop, 
NCE PowerPro Command Station EPROM revision = 6.2.0, 
USB – 232 portal

## Configuration Profiles used:
Two configuration profile folders were created just for the tests described.  They are called “Macro_Test_NCE_PowerPro_232” and “Macro_Test_NCE_Simulator” and each has nothing other than the minimum required for these tests.  Both folders are attached here as zip files.  
It should be noted that the session.log file for each of the Config Profiles has been copied into the respective folder.  
Annotated versions of both the monitorLog.txt and the session.log file are provided along with the untouched versions.  

## The Supporting Context:
After several attempts to duplicate the situation described in the request for help, and alternative plan eventually was formulated.  A Macro Backup file was created that contains eleven macro definitions for Macros # 0 through # 10 and eleven more for Macros # 245 through # 256.  All of these macros contain trivial Commands to alternately close and throw turnout # 1.  The macro backup file provides a repeatable baseline condition and the text file is easy for me to compare for changes by looking at the text in subsequent Backup files.  After starting the NCE Monitor logging, the Backup file is loaded to ensure that each test is independent of all prior conditions and memory states.  

## The Test Steps:
It is important for me to admit that the time stamps on the NCE monitorlog.txt file and the session.log file consistently show lags in reporting the same events.  Time stamp lags were greater when running NetBeans in Debug mode.  My days of programming multi-thread software are but a dim memory.  A procedural practice was adopted for these tests, to delay each manual action for at least a full minute to help me identify gaps in each of the time stamp sequences.  Even so, some events appear to me to be reported much closer together than I remember doing them.  No further attempt has been made here to compensate for or provide correction to any of the apparent time stamp discrepancies between the two log files.  This topic is beyond my reach.  

Each of the test Configuration Profiles starts PanelPro with the System Console and the NCE Monitor.  No scripts, panels, or other JMRI features are employed.  

After the PanelPro has completed the startup and is running, it is necessary to manually start the NCE Monitor logging feature.  

The PanelPro menu item NCE -> Edit Macros is used to open the window labeled “NCE: Edit NCE Macro” and then the “Restore” button is used to read the file named “232 test macro backup.txt” for the known macro memory baseline state.  Both folders contain identical copies of the backup text file.  

Then Macro # 5 is consistently selected and the “Get” button is used.  After a few seconds, the memory data fills in the text windows for the Commands.  At least one of the forum threads led me to believe that adjacent macro data could be contaminated.  The extra steps taken here also allowed for verification or refutation of that premise.  Thus far, the occurrences of NCE Command Station software crashes have precluded the checking of adjacent memory contamination.  No such contamination has been observed with the NCE Simulator tests or from the very very few times the Command Station did not crash.  One can hope that either this question will become mute once we no longer have crashes or if it still occurs, we will have a better understanding of how it could happen.  

When a few of the Macro # 5 turnout commands have been changed, the “Save” button is used.  

The resulting actions vary in subtle ways between the PowerPro 232 interface and the NCE Simulator interface, but the essence is that the data for the modified Macro # 5 is formatted into a string of bytes to be sent to the NCE connection.  Because a maximum of 16 data bytes per write is allowed by the NCE command station, two consecutive writes are done to accommodate the 20 byte length of the Macro data string.  The first write should send the first 16 bytes and the second write should sent the last four bytes.  From what is observable while NetBean is running PanelPro in Debug mode, the entire 20 byte string is handed off to another Thread for that actual write process.  

## What goes Wrong:
Unfortunately, the first write process has formatted the counter byte that tells NCE how many bytes of data are to follow has the value of 129 = 0x81 when it should have a value of 16 = 0x10.  

Both the Simulator and the NCE Command Station respond poorly to this problem, but in different ways and with the Edit Macro function misunderstanding or just missing the reply in either of the two cases.  

### NCE Command Station:
If connected to the NCE Command Station, the reply “4” is reported as an unresolved message in the monitorLog.txt file in the form of `[34]  Reply: 34` but is not interpreted.  
That reply is ignored in the session.log which has no ERROR message pertaining to the receipt of the unrecognized reply type code: 4.  
Then several seconds later, some kind of strange and unpredictable behavior usually starts.  The NCE Command Station software usually crashes.  In some cases the strange behavior is not apparent for at least 10 or seemingly more seconds.  

### NCE Simulator:
[... cut ...]
After that interval, the two log files seem to disagree.  The monitorLog.txt file indicates that the default “are you there?” handshake has been initiated successfully.  The session.log file indicates that the connection has timed out.  

## In Either Case:
The seemingly unrelated fact that legitimate reading the memory throws EIGHT illegitimate ERROR messages for each macro may be considered a nuisance, but it can and did **bury useful** information.  

[Macro_Test_NCE_PowerPro_232.zip](https://github.com/JMRI/JMRI/files/1586528/Macro_Test_NCE_PowerPro_232.zip)
[Macro_Test_NCE_Simulator.zip](https://github.com/JMRI/JMRI/files/1586530/Macro_Test_NCE_Simulator.zip)

pabender, 2017-12-26:
This is related to known issues with the NCE macro editor.

See, for example, this bug report from sourceforge:
https://sourceforge.net/p/jmri/bugs/557/

@KenC57 provided background in response to the bug report above.

dheap, 2017-12-26:
The problem of the NCE Command Monitor throwing a spurious error message of the form "ERROR - Unhandled command code:" for EVERY response from the command station has been with us from the day we switched from the RxTx serial libraries to the PJC serial libraries.

While a nuisance, these messages do not indicate a real error and have absolutely no deleterious effect on the functioning of either JMRI or the NCE Command Monitor.

The problem has been mentioned but not addressed. This is probably because it does not create a functional problem whereas there have been many high-priority functionality issues to address. JMRI is a purely volunteer group and no one is obliged to take on any issue. It's a case of priorities (and also individual familiarity with any particular part of the code) for those of us who choose to contribute.

Dave

dheap, 2017-12-26:
@Cliff-Anderson Note that my previous comment re "ERROR - Unhandled command code:" messages is not intended to dismiss your valid reporting of the known issues with the NCE Macro Editor.

dheap, 2017-12-27:
I'm looking into the spurious "ERROR - Unhandled command code:" messages now.

dheap, 2017-12-27:
#4638 cover sat least some of the spurious errors

dheap, 2017-12-27:
The spurious error logs were actually as the result of some FindBugs-inspired changes around the same time as, but unrelated to the switch from RxTx to PJC.

Cliff-Anderson, 2017-12-29:
Thank you @dheap for fixing the spurious error logs.  

After blundering through the NetBeans Debug mode with numerous dead ends, two possible improvements are suggested in what follows.  I am not proficient in Java and am badly out of date with multi-thread skills, so I am reluctant to do any more than just "suggest."  

The less important suggestion is provided first, as it is the easier on to verify. 

Item # 3 The NCE Simulator fails to reply with the error code “4” response:
File named “SimulatorAdapter.java” starting at line number 422 with a suggested insertion after line number 424 and a modification to the line currently numbered 425
```
422    private NceReply writeMemory(NceMessage m, NceReply reply, int num, boolean skipbyte) {
423        if (num > 16) {
424            log.error("Nce write memory command was greater than 16");
424a           reply.setElement(0, NCE_BYTE_OUT_OF_RANGE);   // Report error to Macro Editor
425//          return null;
425            return reply;
426        }
```

Calling function at line number 334 appears to ignore the returned value, but not the argument named “reply” .

The fourth item on the original list "The NCE Monitor sees the reply with error code “4” as an unrecognized reply 0x34"  remains unsolved as does the persistent ERROR messages that appear in the session.log file with each Save instruction.  

**But**, even better possibilities may be on the horizon with the second suggestion.

From the original list of reproducible problems, "Item # 1 Macro Save action attempts write 16 bytes of data, but tells NCE that it wants to write 129 bytes."  It is now clear that the transmitted byte length is erroneously being picked up from the first byte of the Macro Data and is not consistently 129 as I thought at the time.  
 
In the file named “NCEMacroEditPanel.java” starting at line number 1528 and a **(suggested modification)** at line number 1534:

```
1528   // Write N bytes of NCE memory 
1529    private void writeSerialMemoryN(int nceMacroAddr, byte[] x, int len) {
1530        replyLen = NceMessage.REPLY_1;   // Expect 1 byte response
1531        waiting++;
1532        byte[] bl = NceBinaryCommand.accMemoryWriteN(nceMacroAddr, len);
1533        for (int i = 0; i < len; i++) {
1534//          bl[3 + i] = x[i];
1534            bl[4 + i] = x[i];	// Hardcoded offset was miscalculated 1535        }
1536        NceMessage m = NceMessage.createBinaryMessage(tc, bl, NceMessage.REPLY_1);
1537        tc.sendNceMessage(m, this);
1538    }
```
Please do not ask me how many times I skipped over line 1534 without recognizing that the first byte of Macro Data was getting put into the byte length location.  Personally, the use of "Hardcoded" values was always distasteful, but I am a user here and not a code contributor.  

Both of these suggestions seem to work OK using NetBeans, but I am well aware that making changes to the master code is much more complicated.

Cliff-Anderson, 2017-12-29:
Sorry, the wrong button got clicked.

Cliff

DanielBoudreau, 2018-07-26:
Fixed in 4.13.1

closed 2018-07-26 by DanielBoudreau