JMRI/JMRI #4640: LayoutBlock: (2605) fails causing thread death in startup

KenC57, 2017-12-27:
This gives the case of where loading a file via the startup prevents the main JMRI window from opening.
I'm attaching the file that causes this. You need a LocoNet and NCE connection to simulate this. You have to rename this file to the xml since we can't upload a .xml file.
[CCMR_Orig.txt](https://github.com/JMRI/JMRI/files/1589951/CCMR_Orig.txt)

KenC57, 2017-12-28:
The exception thrown is related to the list, maxtSet, is an immutable list and at 2605 it tries to change it. I think the right thing is that at 2583 it needs to copy the provided list instead of equating the list.

```
2017-12-28 09:06:33,451 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.UnsupportedOperationException
	at java.util.Collections$UnmodifiableList.set(Unknown Source)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.addThroughPath(LayoutBlock.java:2605)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.addThroughPath(LayoutBlock.java:2419)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.addThroughPath(LayoutBlock.java:2351)
	at jmri.jmrit.display.layoutEditor.LayoutBlock.addNeighbour(LayoutBlock.java:1559)
```

geowar1, 2017-12-28:
Looking at the sources… On line 2580 they allocate/initialize set and then check its size on line 2582 (would that ever fail?) and then never use it again… I think they intended to modify set and pass it to addThroughPathPostChecks instead of maxt and just forgot to do that.

rhwood, 2017-12-28:
The ConnectivityUtil method that generates the list should not be making an immutable list. This is discussed in another issue and on the PR that makes that change.

KenC57, 2017-12-28:
PR #4643 seems to fix it by using local copies of the lists. Also 80+ doc warnings while I was there.

PR #4646 (2017-12-29): Fix API for getting a sequence of LayoutTurnouts and expected state
The API for getting a sequence of LayoutTurnouts and their expected state in a panel is fundamentally flawed in that the sequence of LayoutTurnouts is not retained by ConnectivityUtil, but the expected state of those turnouts is retained, but only until another sequence is requested.

Furthermore 7be74e8 broke this by allowing the sequence to be mutable, while requiring that the associated expected states by immutable, which breaks panel loading.

This PR fixes that by adding an API for holding an object and its expected state [... cut]
closed 2017-12-29 by rhwood