JMRI/JMRI #4650: Confusing Digitrax Decoder families

rhwood, 2017-12-29:
Are the two Digitrax decoder families `Basic STD` and `Basic STD*` supposed to be separate in the attached image or is the `*` in the second family an error?

[image]

bobjacobsen, 2017-12-29:
A long time ago, it was intentional. Digitrax had a series of name for decoders as the feature set advanced. The difference between Basic STD and Basic STD* is, IIRC, whether the strobe lights are configurable or not. Since families & files & CV sets were related then, that difference created a separate family. 

A diff of the files might show the difference, if the white space hasn’t gotten too munged. 

With more modern JMRI’s capabilities, it should be possible to combine quite a few of the historical families if somebody wants to. 

Bob

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2018-03-16 by rhwood