JMRI/JMRI #4655: fontname="Dialog.plain" is being added to .xml file when saved on raspberrypi.

ClockLoop, 2017-12-30:
The attribute -->   fontname="Dialog.plain"     is being added to every locoicon and blockcontentsicon inside an .xml file that i created on windows and then edit on a raspberrypi.

I compared my original file to the modified file and those are being added when I save the file on a raspberrypi only.   I can do the same edits on a pc and those attributes will not be added.

These extras cause an error when they are present.

[image]
[image]
[image]

pabender, 2017-12-30:
looking at the code, the font name attribute is created if the font is not the default font, but it must be missing from the file that defines how the XML panel files look.

Try running the TurnOffXmlValidation.py script  before loading  the panel and let us know if that solves the problem.  If it does, that will confirm where the problem is, and we can fix it before, hopefully before the next test release.

ClockLoop, 2018-01-02:
Ill test that now.

ClockLoop, 2018-01-02:
I opened jmri ran the TurnOffXmlValidation.py script, I then opened the xml file that I was not able to open before, and it opens and operates normally now.
So confirmed, it does solve the problem.

pabender, 2018-01-10:
@ClockLoop 

Can you please attach a copy of the XML file that includes the 

`fontname="Dialog.plain"
`

attributes.

I would like to fix this, but I need a test file.

closed 2021-04-14 by stale[bot]