JMRI/JMRI #4662: Frequent test failures with "reply complete in unexpected state" errors

rhwood, 2017-12-31:
The "reply complete in unexpected state" error message is written by two classes:
https://github.com/JMRI/JMRI/blob/31d7ee34e6058a81ec74863334a191d0d1fc36f5/java/src/jmri/jmrix/dcc4pc/Dcc4PcTrafficController.java#L493
or
https://github.com/JMRI/JMRI/blob/9e1b89e7e21c95fafa00861489db96e2663204f5/java/src/jmri/jmrix/srcp/SRCPTrafficController.java#L182

One of these classes is in a test that frequently fails without more context.

pabender, 2017-12-31:
While we have been seeing this lately, it is occurred after the tests are completed.  In other words, this isn't the cause of the test failure.

See, for example,: 
https://ci.appveyor.com/project/JMRI/jmri/build/11053

Which was a successful test run showing the error message.

So, while this is something we need to track down and eliminate, it is not the cause of a failing test run.

Now, as for why this is occurring, there is a communication thread that is still running at the time the time the test run finishes.  This ultimately can be traced back to AbstractMRTrafficController.  We have just eliminated most of the occurrences of the error messages, but we obviously missed a few where the method generating the message was overridden. 

I'll take a look at this in the next few days and see if I can get the tests to properly terminate the communication threads for these classes.

pabender, 2018-01-01:
This particular error IS still produced in AbstractMRTrafficController

https://github.com/JMRI/JMRI/blob/e8be5d55e83945290cbdcaef5973519677dfe302/java/src/jmri/jmrix/AbstractMRTrafficController.java#L1037

That greatly increases the number of possible culprits for generation of this message.  I am working on making the message tell us which class generated it.

pabender, 2018-01-01:
It looks like the culprit is actually one of the EasyDCC tests.

This was in the log for one of the Travis tests in #4667:

ERROR - reply complete in unexpected state: 10 was O
 in class easydcc.EasyDccTrafficController [easydcc.EasyDccTrafficController Receive thread] 

Note that at least the appveyor test and one of the Travis tests did show the message, but neither of them failed.

silverailscolo, 2018-01-01:
Sorry this is a simple Simulator reply, returning O (capital o). Could not figure out how to send a second reply line that provides the content of the action, as expected from the Control Station according to the documentation.

rhwood, 2018-03-16:
OP is no longer concerned.

rhwood, 2020-07-20:
I think this has been resolved

closed 2020-07-20 by rhwood