JMRI/JMRI #4666: Help: Decoder Pro 3 JMRI and LENZ

Emath2003, 2018-01-01:
Hello,
I am hoping I can get some help. (I have posted in the JMRI Yahoo group as well).
I have two separate Lenz systems, an LZV100 and an LV101/LZ100.
I have a persistent problem where the LZV 100 will not program locomotive through Decoder Pro 3. ( I am using Decoder Pro 3 ver 4.4 but have tested this across multiple versions up to the current version).
The LV101/LV100 works correctly and will program and read back CVs with Decoder Pro 3.

Windows 7.
The Lenz systems are both version 3.6. (update was done by Lenz)
I use the LI USB Ethernet connection (same connection for both).
Decoder Pro 3 preferences are:
System Manufacturer, Lenz;
System Connection, Lenz LIUSB Ethernet;
IP Address 192.168.1.200 (checked on the router ping is good and I can run the layout);
Connection Prefix, X;
Connection name, XPressnet;
Communication Port 5550.

Both systems will run engines and will program on the programming track with an LH100 hand controller.
When using Decoder Pro 3 with the LZV100 I get Error 311, 310, and 306.
"Communications Error between PC and Command Station (311)" No information on what this means, and it is not even listed on the JMRI site as an error.
I also get "Programming Operation Sequence Error (310)"

Wiring, power supplies, cables, track, and programming track have all be verified and swapped across both systems to eliminate those variables. Tested with multiple engines with the same result.

Below are three logs:
First: Startup log for LV101/LZ100 (This system works perfectly)
Second: Startup log for LZV100 (This system will do everything but program with Decoder Pro 3, it will program with an LH100) NOTE: There are errors on startup.
Third: Log section for the LZV100 when I try to read CV1 using Decoder Pro 3 (this section of the log follows right after the "Second" section.

Any help would be greatly appreciated.
Thanks,
E

***LV 101 LZ 100 LOG***
QUOTE
2017-12-30 15:11:08,257 util.Log4JUtil INFO - ****** JMRI log ******* [main]
2017-12-30 15:11:08,272 util.Log4JUtil INFO - This log is appended to file: C:\Users\Standard Account\JMRI\log\messages.log [main]
2017-12-30 15:11:08,272 util.Log4JUtil INFO - This log is stored in file: C:\Users\Standard Account\JMRI\log\session.log [main]
2017-12-30 15:11:08,288 apps.AppsBase INFO - DecoderPro version 4.4-R1ccf76b starts under Java 1.8.0_151 on Windows 7 x86 v6.1 at Sat Dec 30 15:11:08 EST 2017 [main]
2017-12-30 15:11:11,392 gui3.Apps3 INFO - Starting with profile My_JMRI_Railroad.3f60ba7c [main]
2017-12-30 15:11:11,580 node.NodeIdentity INFO - Using jmri-90B11C8C37F7-3f60ba7c as the JMRI Node identity [main]
2017-12-30 15:11:12,406 configurexml.PerformFileModelXml INFO - Load file C:\Users\Standard Account\JMRI\My_JMRI_Railroad\Turnouts.xml [main]
2017-12-30 15:11:16,603 util.FileUtilSupport INFO - File path program: is D:\Decoder Pro 3 Ver 4.4\JMRI\ [main]
2017-12-30 15:11:16,603 util.FileUtilSupport INFO - File path preference: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2017-12-30 15:11:16,603 util.FileUtilSupport INFO - File path profile: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2017-12-30 15:11:16,603 util.FileUtilSupport INFO - File path settings: is C:\Users\Standard Account\JMRI\ [main]
2017-12-30 15:11:16,603 util.FileUtilSupport INFO - File path home: is C:\Users\Standard Account\ [main]
2017-12-30 15:11:16,603 util.FileUtilSupport INFO - File path scripts: is D:\Decoder Pro 3 Ver 4.4\JMRI\jython\ [main]
UNQUOTE
***

***LZV100 Log***
QUOTE
2017-12-30 15:43:40,713 util.Log4JUtil INFO - ****** JMRI log ******* [main]
2017-12-30 15:43:40,728 util.Log4JUtil INFO - This log is appended to file: C:\Users\Standard Account\JMRI\log\messages.log [main]
2017-12-30 15:43:40,728 util.Log4JUtil INFO - This log is stored in file: C:\Users\Standard Account\JMRI\log\session.log [main]
2017-12-30 15:43:40,744 apps.AppsBase INFO - DecoderPro version 4.4-R1ccf76b starts under Java 1.8.0_151 on Windows 7 x86 v6.1 at Sat Dec 30 15:43:40 EST 2017 [main]
2017-12-30 15:43:43,848 gui3.Apps3 INFO - Starting with profile My_JMRI_Railroad.3f60ba7c [main]
2017-12-30 15:43:44,036 node.NodeIdentity INFO - Using jmri-90B11C8C37F7-3f60ba7c as the JMRI Node identity [main]
2017-12-30 15:44:14,175 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [Timer-0]
java.lang.NullPointerException
at jmri.jmrix.lenz.liusbethernet.LIUSBEthernetAdapter$1.run(LIUSBEthernetAdapter.java:114)
at java.util.TimerThread.mainLoop(Unknown Source)
at java.util.TimerThread.run(Unknown Source)
2017-12-30 15:44:14,190 lenz.XNetInitializationManager WARN - Command Station disconnected, or powered down assuming LZ100/LZV100 V3.x [main]
2017-12-30 15:44:14,705 configurexml.PerformFileModelXml INFO - Load file C:\Users\Standard Account\JMRI\My_JMRI_Railroad\Turnouts.xml [main]
2017-12-30 15:44:18,933 util.FileUtilSupport INFO - File path program: is D:\Decoder Pro 3 Ver 4.4\JMRI\ [main]
2017-12-30 15:44:18,933 util.FileUtilSupport INFO - File path preference: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2017-12-30 15:44:18,933 util.FileUtilSupport INFO - File path profile: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2017-12-30 15:44:18,933 util.FileUtilSupport INFO - File path settings: is C:\Users\Standard Account\JMRI\ [main]
[... 52 lines cut ...]
2017-12-30 16:28:29,832 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 21 times. [Receive]
2017-12-30 16:28:31,938 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:31,938 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 22 times. [Receive]
2017-12-30 16:28:34,153 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:34,153 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 23 times. [Receive]
2017-12-30 16:28:36,462 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:36,462 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 24 times. [Receive]
2017-12-30 16:28:38,865 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:38,865 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 25 times. [Receive]
2017-12-30 16:28:41,376 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:41,376 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 26 times. [Receive]
2017-12-30 16:28:43,981 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:43,981 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 27 times. [Receive]
2017-12-30 16:28:46,696 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:48,053 lenz.XNetProgrammer ERROR - Communications error in REQUESTSENT state while programming. Error: 01 02 03 [AWT-EventQueue-0]
2017-12-30 16:28:48,053 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 02 03. Retransmitted 28 times. [Receive]
2017-12-30 16:28:50,861 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:50,861 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 29 times. [Receive]
2017-12-30 16:28:53,765 jmrix.AbstractMRTrafficController INFO - Message added back to queue: 22 15 01 36 [Transmit]
2017-12-30 16:28:53,765 jmrix.AbstractMRTrafficController ERROR - Automatic Recovery from Error Message: 01 08 09. Retransmitted 30 times. [Receive]
UNQUOTE
***
emath2003@yahoo.com

Thanks!

pabender, 2018-01-01:
> On Dec 31, 2017, at 7:26 PM, Emath2003 <notifications@github.com> wrote:
> 
> LZV100 Log
> 
> QUOTE
> 2017-12-30 15:43:40,713 util.Log4JUtil INFO - ****** JMRI log ******* [main]
> 2017-12-30 15:43:40,728 util.Log4JUtil INFO - This log is appended to file: C:\Users\Standard Account\JMRI\log\messages.log [main]
> 2017-12-30 15:43:40,728 util.Log4JUtil INFO - This log is stored in file: C:\Users\Standard Account\JMRI\log\session.log [main]
> 2017-12-30 15:43:40,744 apps.AppsBase INFO - DecoderPro version 4.4-R1ccf76b starts under Java 1.8.0_151 on Windows 7 x86 v6.1 at Sat Dec 30 15:43:40 EST 2017 [main]
> 2017-12-30 15:43:43,848 gui3.Apps3 INFO - Starting with profile My_JMRI_Railroad.3f60ba7c [main]
> 2017-12-30 15:43:44,036 node.NodeIdentity INFO - Using jmri-90B11C8C37F7-3f60ba7c as the JMRI Node identity [main]
> 2017-12-30 15:44:14,175 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [Timer-0]
> java.lang.NullPointerException
> at jmri.jmrix.lenz.liusbethernet.LIUSBEthernetAdapter$1.run(LIUSBEthernetAdapter.java:114)
> at java.util.TimerThread.mainLoop(Unknown Source)
> 

There is an error thrown while initializing the connection to the LIUSBEthernet. Until that is fixed, I can't guarantee anything else will work.

Since this log was produced with version 4.4, I need you to go back and get a log from either 4.10 or 4.11.1 ( preferably 4.11.1).  I need to see where the logs indicate an issue, but there may be a line number difference between 4.4 and 4.11.1.

Two questions:
1) are you using one LIUSB-Ethernet or two? 
2) are you using automatic configuration for the IP address?

Paul

Emath2003, 2018-01-01:
Dear Paul, 
Thank you for your quick reply. 
1: I only have one LIUSB-Ethernet. 
2: I am not using automatic configuration for the IP Address. The LIUSB-Ethernet has a static address of 192.168.1.200, the ping test works in the router, and it works with the LV101/LZ100. 

Here is the log from Decoder Pro 3 Version 4.11.1

LZV100 log

2018-01-01 16:03:59,526 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2018-01-01 16:03:59,576 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Standard Account\JMRI\log\messages.log [main]
2018-01-01 16:03:59,576 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Standard Account\JMRI\log\session.log [main]
2018-01-01 16:03:59,586 apps.AppsBase                         INFO  - DecoderPro version 4.11.1+Rfe44680 starts under Java 1.8.0_151 on Windows 7 x86 v6.1 at Mon Jan 01 16:03:59 EST 2018 [main]
2018-01-01 16:04:02,706 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f60ba7c [main]
2018-01-01 16:04:02,956 node.NodeIdentity                     INFO  - Using jmri-90B11C8C37F7-3f60ba7c as the JMRI Node identity [main]
2018-01-01 16:04:33,080 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [Timer-0]
java.lang.NullPointerException
	at jmri.jmrix.lenz.liusbethernet.LIUSBEthernetAdapter$1.run(LIUSBEthernetAdapter.java:94)
	at java.util.TimerThread.mainLoop(Unknown Source)
	at java.util.TimerThread.run(Unknown Source)
2018-01-01 16:04:33,110 lenz.XNetInitializationManager        WARN  - Command Station disconnected, or powered down assuming LZ100/LZV100 V3.x [main]
2018-01-01 16:04:34,040 apps.PerformFileModel                 INFO  - Loading file C:\Users\Standard Account\JMRI\My_JMRI_Railroad\Turnouts.xml [main]
2018-01-01 16:04:39,650 util.FileUtilSupport                  INFO  - File path program: is D:\Decoder Pro 3 Ver 4.11.1\JMRI\ [main]
2018-01-01 16:04:39,650 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2018-01-01 16:04:39,650 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2018-01-01 16:04:39,650 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Standard Account\JMRI\ [main]
2018-01-01 16:04:39,650 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Standard Account\ [main]
2018-01-01 16:04:39,650 util.FileUtilSupport                  INFO  - File path scripts: is D:\Decoder Pro 3 Ver 4.11.1\JMRI\jython\ [main]

Thanks

pabender, 2018-01-04:
I can see why the null pointer exception occurred, and I'm about to submit a fix for that, but I can't see why you aren't getting connected to your command station (which is ultimately the problem).

How is your LIUSB-Ethernet physically connected to the LZ100 or the LZV100?  
is it connected directly to the back?
is it connected using the LMAB terminal block or the DIN connector?
Are you using the same same cables with both command stations?
What lights are on on the LIUSB-Ethernet?

Emath2003, 2018-01-15:
Answers below: 

How is your LIUSB-Ethernet physically connected to the LZ100 or the LZV100?
is it connected directly to the back?
is it connected using the LMAB terminal block or the DIN connector?

A: The LIUSB-Ethernet is connected by an LA 152 using the LMAB terminals. However, I also used a separate “test bench” with a separate LA 152, separate Xpressnet cable, not connected to the layout, separate programming track, and a separate power supply.  Same result, LZ100/LV101 works, LZV100 does not.  I have also bypassed the LA152

Are you using the same same cables with both command stations? 

A: Yes, but I have changed the cable, and have separate cables on the test bench and still the same result, LZ100/LV101 works, LZV100 does not. Also, I tried both systems, on the test bench, to a Windows XP computer (with an earlier version of Decoder Pro), same result, LZ100/LV101 works, LZV100 does not.
 
What lights are on on the LIUSB-Ethernet?

A: The Xpressnet light is on solid red most of the time. The data light flashes yellow momentarily when a command is given. 

If there is anything you can think of to try, please let me know and we will try it out. 
Thanks

Emath2003, 2018-02-02:
Paul, 
Any thoughts on what else I might be able to try.  I am still unable to program and get the above error messages. Is it possible that there is a problem with the LZV100 itself, or with the LI USB Ethernet that is unique to the LZV100. Is there anything that any other logs might reveal. 
Thanks, 
Eric

pabender, 2018-02-02:
JMRI doesn't actually care if the command station is an LZV100 or an LZ100, so this has to be a wiring problem somewhere ( possibly inside the command station)

Can you verify that the LIUSB is on?  It is powered by the L and M connections of the XpressNet connection ( which should carry 12VDC).

Also, if you have an LH90 available, can you take the cord off of that, plug the DIN connector into the command station directly and the other end into the LIUSB.  Once you have that setup, you can try to connect again.

Emath2003, 2018-02-04:
Paul,
Thank you for your above post. We spent the last two days testing the systems and while we have not yet found a solution, report as follows: 

In answer to your questions: 

Q: Can you verify that the LIUSB is on? It is powered by the L and M connections of the XpressNet connection (which should carry 12VDC).

A: Yes, LM reads 12 volts tested by multimeter.  LIUSB/ETHERNET connector is receiving power and continues to work with the Ethernet connection to both the LZV100 and the LZ100/LV101. 

Q: Also, if you have an LH90 available, can you take the cord off of that, plug the DIN connector into the command station directly and the other end into the LIUSB. Once you have that setup, you can try to connect again.

A: We do have an LH 90 and did this connection as well. This was done so that the only xpressnet device or connection was the LIUSB/ETHERNET (all the rest of the xpressnet was disconnected). The result was the same, the system will run locomotives from DecoderPro3, but will not program and still provides the same error on startup. 

We then went on to perform further testing. 

1: LIUSB/ETHERNET Connected by Ethernet

We reconfirmed that the LIUSB/ETHERNET connector, when connected by Ethernet, loads without error and both operates and programs on the programming track with the LZ100/LV101. We also tested again with the LZV100, which re-confirmed that the same error previously posted in the logs appears at startup, and that DecoderPro3 will operate the railroad, but will not program on the programming track. Both of the foregoing were verified with the command stations connected to the layout as well as a test bench that is completely separate and used no common wires with the layout and has a separate power supply. 

So the LZV100/LV101 continues to work without issue with the LIUSB/ETHERNET, whereas the LZV100 gives the startup error shown in the logs above in this same string of posts, and will not program on the programming track. 

However, we also found that using the single CV programmer in DecoderPro3, the LZV100 would actually write the cv and then remain in program mode (light on the LZV100 flashing). What it would not do is make the “click” and come back to regular mode and read back the CV. We do not know whether this is because the LZV100 is not sending the information back, or whether the LIUSB/ETHERNET is not properly handling the information. 

We also tried resetting the LIUSB/ETHERNET, changing the IP address, and the behavior was unchanged. 

2: LIUSB

We have an LIUSB (Part 23150) (Which does not have an Ethernet connector, just a USB and xpressnet).
We tested both the LZV100 and the LZ100/LV101 with this connector using the test bench and a Windows 7 computer. With both DecoderPro3 and the lenz software both the LZV100 and the LZ100/LV101 work to operate the layout and to program on the programming track. 

3: LIUSB/ETHERNET connected by USB
The testing above prompted us to go back to the LIUSB/ETHERNET connector and use the USB part of this connector. When we set this up, the LZV100 would again operate locomotives, but would not program. When we connected this to the LZ100/LV101 that system would operate locomotive, but would frequently need to send a CV command twice to obtain a read back of the CV. 

We carefully observed the lights on the LIUSB/ETHERNET. We could see that in programming mode the command would be sent (light flash on the LIUSB/ETHERNET), the command station go into program mode (light on the front of the command station flash), the locomotive would buzz and move very slightly on the tracks, however the LZV100 would just keep flashing whereas the LZ100/LV101 would “click” followed by the light on the LIUSB/ETHERNET flashing, DecoderPro3 displaying the read back cv and the command station light being steady on. We then tried the same test using the Lenz Software (server) for the LIUSB/ETHERNET and this software displayed the same behavior with the LZV100 not working and the LZ100/LV101 operating correctly. 

4: Reconnected to the layout.

We reconnected the LZV100 to the layout and reconfirmed that when connected to the Ethernet connector (again totally separate wiring, power source, different LA 152, and different electrical, data, xpressnet, and ethernet cables from the test bench) the behavior was unchanged. The LZV100 gave the same start up error as in the logs previously posted, operates locomotives, but will not program on the programming track through DecoderPro3. We also reconfirmed that the system will program locomotives on the programming track with the LH100.   

Based on our tests, we believe that the following can be eliminated:

The router and associated cables: LZ100/LV101 works completely, and we have changed the cables as well as used the separate test bench. The router sees the LIUSB/ETHERNET and that device is accessible at its designated address. We also disabled the security software for the testing. 

Power Supply: Tested with two different power supplies

Xpressnet and cables: Tested with different isolated LA152s and with no LA152 using the cable for the LH90. 

Locomotive: tested across several locomotives all of which display the same behavior, and the LH100 is able to program all locomotives. 

Track: Tested on the layout and a test bench with separate track and track wiring (all wiring is different on the test bench). 

[... cut ...]
The fact that the LZV100 will not program locomotives with the LIUSB/ETHERNET using either the Ethernet or the USB connection (and using DecoderPro or the Lenz software), but will program with the LH100, while the LZ100/LV101 works with all connections is puzzling.  

If the software can be eliminated (since it does work with the LZ100/LV101), what we are left with is that we cannot determine whether the problem is with the LIUSB/ETHERNET or with the LZV100 itself or whether the problem is hardware or firmware related for either of these devices. Do you have any advice as to how we can test as between these two components.

pabender, 2018-02-06:
Ok, somehow I missed in all this that you are able to control locomotives.  That means you don't have a basic communication problem.  

I do see what is going on now however.

Here is a copy of a log from my LIUSB-Ethernet connected LZV100 reading CV1 with no locomotive on the programming track:

```
00:32:10.423: [packet: 22 15 01 36]  Service Mode Request: Read CV 1 in Direct Mode
00:32:10.424: [61 02 63]  Broadcast: Service Mode Entry
00:32:10.425: [61 02 63]  Broadcast: Service Mode Entry
00:32:11.770: [01 05 04]  The Command Station is no longer providing the LI a timeslot for communication
00:32:11.772: [01 02 03]  Error occurred between the interface and the command station.
00:33:11.780: [packet: 22 15 01 36]  Service Mode Request: Read CV 1 in Direct Mode
00:33:11.785: [01 07 06]  Timeslot Restored
00:33:21.781: [packet: 21 10 31]  REQUEST: Service Mode Result
00:33:21.788: [61 13 72]  Service Mode: Data Byte Not Found
00:33:21.790: [packet: 21 81 A0]  REQUEST: Normal Operations Resumed
00:33:21.943: [61 01 60]  Broadcast: Normal Operations Resumed
00:33:21.948: [61 01 60]  Broadcast: Normal Operations Resumed
00:33:21.952: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
```
What you will notice in the log is that the command station removes the communication timeslot (the 01 05 04 message) immediately before the 01 02 03 error is reported.  

Because you sent console logs, and not XPressNet logs, the 01 02 03 error was present in your logs, but the 01 05 04 message wasn't (because to the programmer, that isn't an error).

It turns out that knowing both messages are present is really important for proper operation here.  The XPressNet programmer has been treating the 01 02 03 as an error that needed to be brought up to the user.  To make this work correctly, since the interface is (for some reason) sending both messages to us, we need to suppress sending that message on to the interface in the case where no timeslot is available.

Now, why this message is occuring with your LZV100 and not with the LZ100, I can't explain.  I've never noticed that the LIUSB-Ethernet did this, but I have a fix going through the build process now.  I'll let you know when there is something available for you to test.

pabender, 2018-02-06:
@Emath2003 

Please try one of the package builds from

http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

All of the current builds have the correction to the programmer required to make this work with your LZV100

Paul

Emath2003, 2018-02-19:
Paul, 
I downloaded the version from the link you provided and loaded it. 
The startup log is attached below, as is the xpressnet monitor log when I both tried unsuccessfully to read cv 1 with DecoderPro, and when I then immediately did so, with no changes of any kind to the system, successfully with an LH-100. I annotated the logs accordingly. 
After testing this version I can report no change from the behavior noted above, and I am still unable to read any cv values on the programming track with DecoderPro. 
Hope the logs help, and continued thanks for all of your assistance with this. 

***Startup Log***
2018-02-19 10:36:39,861 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2018-02-19 10:36:39,892 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Standard Account\JMRI\log\messages.log [main]
2018-02-19 10:36:39,892 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Standard Account\JMRI\log\session.log [main]
2018-02-19 10:36:39,908 apps.AppsBase                         INFO  - DecoderPro version 4.11.3ish+jenkins+20180210T0447Z+R8101bf73 starts under Java 1.8.0_151 on Windows 7 x86 v6.1 at Mon Feb 19 10:36:39 EST 2018 [main]
2018-02-19 10:36:42,939 gui3.Apps3                            INFO  - Starting with profile My_JMRI_Railroad.3f60ba7c [main]
2018-02-19 10:36:43,251 node.NodeIdentity                     INFO  - Using jmri-90B11C8C37F7-3f60ba7c as the JMRI Node identity [main]
2018-02-19 10:37:13,518 lenz.XNetInitializationManager        WARN  - Command Station disconnected, or powered down assuming LZ100/LZV100 V3.x [main]
2018-02-19 10:37:15,127 apps.PerformFileModel                 INFO  - Loading file C:\Users\Standard Account\JMRI\My_JMRI_Railroad\Turnouts.xml [main]
2018-02-19 10:37:20,529 util.FileUtilSupport                  INFO  - File path program: is D:\Decoder Pro 3 ver 4.11.3\JMRI\ [main]
2018-02-19 10:37:20,529 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2018-02-19 10:37:20,529 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Standard Account\JMRI\My_JMRI_Railroad\ [main]
2018-02-19 10:37:20,529 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Standard Account\JMRI\ [main]
2018-02-19 10:37:20,529 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Standard Account\ [main]
2018-02-19 10:37:20,529 util.FileUtilSupport                  INFO  - File path scripts: is D:\Decoder Pro 3 ver 4.11.3\JMRI\jython\ [main]
2018-02-19 10:37:38,722 jmrix.AbstractMonPane                 WARN  - File selection dialog box did not provide a path to the specified file.  Log will be saved to C:\Users\Standard Account\JMRI\My_JMRI_Railroad\monitorLog.txt [AWT-EventQueue-0]
2018-02-19 10:37:38,722 jmrix.AbstractMonPane                 WARN  - startLogButtonActionPerformed: getSelectedFile() returns monitorLog.txt monitorLog.txt [AWT-EventQueue-0]
2018-02-19 10:37:38,722 jmrix.AbstractMonPane                 WARN  - startLogButtonActionPerformed: is attempting to use returned file path and file name C:\Users\Standard Account\JMRI\My_JMRI_Railroad\monitorLog.txt [AWT-EventQueue-0]
*** End Startup Log***

***Xpressnet Monitor***
10:41:43.557: [packet: 21 24 05]  REQUEST: Command Station Status
10:41:43.557: [62 22 00 40]  Command Station Status: Manual power-up Mode; 
10:42:02.285: [packet: 22 15 01 36]  Service Mode Request: Read CV 1 in Direct Mode
10:42:02.288: [61 02 63]  Broadcast: Service Mode Entry
10:42:02.290: [61 02 63]  Broadcast: Service Mode Entry
10:42:03.212: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
10:42:03.212: [packet: 21 10 31]  REQUEST: Service Mode Result
10:42:03.212: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout

***At 10:43:32 Decoder Pro screen said "Timeout talking to command station (306)" ***  

10:43:32.341: [packet: 21 81 A0]  REQUEST: Normal Operations Resumed
10:43:32.481: [61 01 60]  Broadcast: Normal Operations Resumed
10:43:32.481: [61 01 60]  Broadcast: Normal Operations Resumed
10:43:32.497: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
10:43:43.560: [packet: 21 24 05]  REQUEST: Command Station Status
10:43:43.560: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
10:44:13.561: [packet: 21 24 05]  REQUEST: Command Station Status
10:44:13.561: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
10:44:43.561: [packet: 21 24 05]  REQUEST: Command Station Status
10:44:43.561: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
10:45:13.562: [packet: 21 24 05]  REQUEST: Command Station Status
10:45:13.562: [62 22 00 40]  Command Station Status: Manual power-up Mode; 
10:45:43.562: [packet: 21 24 05]  REQUEST: Command Station Status
10:45:43.563: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout

*** Here I was able to use an LH 100 and read CV 1 in program mode ***

10:45:59.891: [61 02 63]  Broadcast: Service Mode Entry
10:45:59.891: [61 02 63]  Broadcast: Service Mode Entry
10:46:01.287: [01 05 04]  The Command Station is no longer providing the LI a timeslot for communication
10:46:09.907: [61 01 60]  Broadcast: Normal Operations Resumed
10:46:09.907: [61 01 60]  Broadcast: Normal Operations Resumed
10:46:09.907: [01 07 06]  Timeslot Restored
10:46:13.564: [packet: 21 24 05]  REQUEST: Command Station Status
10:46:13.564: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
10:46:43.565: [packet: 21 24 05]  REQUEST: Command Station Status
10:46:43.565: [01 04 05]  Command Successfully Sent/Normal Operations Resumed after timeout
***End Xpressnet Monitor***

pabender, 2018-02-19:
Sent from my iPad

> On Feb 19, 2018, at 10:57 AM, Emath2003 <notifications@github.com> wrote:
> 
> Paul,
> I downloaded the version from the link you provided and loaded it.
> The startup log is attached below, as is the xpressnet monitor log when I both tried unsuccessfully to read cv 1 with DecoderPro, and when I then immediately did so, with no changes of any kind to the system, successfully with an LH-100. I annotated the logs accordingly.
> After testing this version I can report no change from the behavior noted above, and I am still unable to read any cv values on the programming track with DecoderPro.
> Hope the logs help, and continued thanks for all of your assistance with this.
> 
> Startup Log
> 2018-02-19 10:37:13,518 lenz.XNetInitializationManager WARN - Command Station disconnected, or powered down assuming LZ100/LZV100 V3.x [main]
> 
I am still concerned about this.  This is an indication we are never receiving a reply to a version request from the command station.

Until we figure out why this is occurring, I can’t guarantee anything else will work properly.
> Xpressnet Monitor
>
> 10:41:43.557: [packet: 21 24 05] REQUEST: Command Station Status
> 10:41:43.557: [62 22 00 40] Command Station Status: Manual power-up Mode;
> 10:42:02.285: [packet: 22 15 01 36] [...]

This is actually the correct behavior.

Look at the differences between the time stamps between the request for service mode results and the request for normal operations resumed.  That 90 seconds is specified in the XpressNet standards as the time we are to wait before we try to recover. 

Now, the question is why do we not receive a reply to our request for a result from the command station, and that I can’t answer or reproduce.

I can tell you this, all the messages that start with a 01 in the raw data are generated by the 01 04 05 message states either 1) that the command was sent to the command station, but no meaningful error or reply was received from the command station or 2) that the command station has started normal operations.

I am honestly baffled by this, and unfortunately can’t replicate the symptoms.
> *** Here I was able to use an LH 100 and read CV 1 in program mode ***
> 
That actually doesn’t help, because we can’t see the communication between the command station and the LH100.

Paul

closed 2021-04-14 by stale[bot]