JMRI/JMRI #4670: Plan to get back to parsable system prefixes

bobjacobsen, 2018-01-01:
This is my (@bobjacobsen) plan to put JMRI back at its original parseable An* system names (hopefully) during the JMRI 4.13 development cycle so that we're migrated by a year from now.

AFAIK, this is compatible with other parseable formats, such as a URI or period separated style. Is somebody wants to work on that, they should please put their plans in another Issue and reference it here.

The legacy prefixes we have to deal with (as far as I can tell from the code; note that people have certainly configured others as one-offs) are:
 - Anyma DX512 is “DX”
 - DCC++ is “DCCPP"
 - DCC4PC is “DP"
 - JSON client is “json”
 - MRC is “MR"
 - Marklin CS2 is “MC"
 - Raspberry PI is “PI"
 - TAMS is “TM”

For an example of the problems, consider MR, which could either be the MRC prefix or could be M (CAN, including MERG and OpenLCB) Reporters, which can exist. But there aren’t that many MRC systems doing complicated things, and it seems unlikely that MRC and MERG will be on the same layout, so let’s just treat MR as a special case during this migration and see if that can work.

The most important components to get right are those with ProxyManagers, where objects from different systems have to be resolved properly.  These are
- Lights
- Reporters
- Sensors
- Turnouts

Although there are _lots_ of NamedBean subtypes that run through a single (i.e. internal) manager, some of which don't really enforce any kind of name. In the long run, we want to get system names back to being parsable, but in the short run we have to deal with what's in people's panel files.  A quick scan of our test XML files found problematic examples of multiple types:
 - Memory
 - SignalHead
 - SignalMast
 - Logix
 - Block

For a full clean-up, we want to reach a state where the naming is handled properly at all levels:
 - When configuring the connection
 - When creating an item in e.g. scripts
 - In the panel file schema check

But, as a migration point, we want these to be soft warnings for quite a while before we make them hard failures. To the extent that we can, we should automatically resolve these, but that’s probably not always going to be possible.

---

The first round of changes are detailed in PR #4669.  They allow the special case "legacy" prefixes above to continue to work properly (though at some cost in CPU load for everybody), and organize the parsing code better.  This also includes what we originally started out to do, which is better comparisons for NamedBean system names, e.g. so that C/MRI and OpenLCB system names sort properly.

---

The next step, unless people raise unanticipated issues, is to stop making the situation worse by changing the future use of prefixes. There are two parts to this.  

First, for those Bean types that rely on system definitions:

- [ ] Change the default letters for new systems.  This has (AFAIK) no adverse effect on anybody with an existing connection, but keeps the number of non-conforming ones from increasing. The new ones where chosen to be (currently, mostly) unique, but perhaps not so mnemonic; comments and preferences on the new prefix letter assignments are definitely welcomed! 
    - [ ] Anyma DX512 change “DX” to “D1”  (As in all the letter-digit assignments below, this could also be just the letter, but then it gets hard to write self-explanatory documentation...)
    - [ ] DCC++ change “DCCPP" to “D2"
    - [ ] DCC4PC change “DP" to “D3”
    - [ ] JSON client change “json” to “J2"  ("j" would also be OK, but upper/lower has been a discussion point in the past)
    - [ ] MRC change “MR" to “M" ("M2" would be more to pattern, but it seems unlikely that anybody will be confused by adding an MRC system to a MERG or OpenLCB installation)
    - [ ] Marklin CS2 change “MC" to “M2”
    - [ ] Raspberry PI change “PI" to “P2" ("P" is powerline, and "P1" looks too much like "PI"; note the name of the device is "Raspberry Pi", with a lower case "i")
    - [ ] TAMS change “TM” to “T” (Could be "T2", but it seems unlikely that anybody will be confused by adding a TMCC system to a TAMS one)
    - [ ] document in Names.html, the hardware pages, release note
- [ ] Add checks to the preferences so that non-conforming prefixes can’t be set; existing non-parsable prefixes will remain OK, but the user can’t add a new system connection preference with a non-parsable prefix nor change an exiting prefix to a non-parsable one.

Second, for those Bean types (e.g. SignalMasts, Logix, etc) that don't rely on system definitions for their prefixes:

- [ ] Ensure that all the "add" operations in the tables create a proper system prefix (usually, but not always, "I"), along with the correct type letter.

This two are not yet in PRs, but the goal is to have all this infrastructure in place by 4.11.2 in a week or so. 

---

After that, hopefully-but-not certainly in 4.11.3, is the start of migrating existing prefixes. Basic idea is to be permissive of people using non-parsable Light, Sensor, Reporter and Turnout (LRST) prefixes now, but to let them know changes are coming, provide basic migration tools, and to keep more of them from being created.  Non-LRST names are a mess, but one problem at a time….

- [ ] Add detection & warning infrastructure for system prefixes:
    - [ ] Creation of a NamedBean with a non-parseable system name is stored in a application-common list; if that list is non-empty, the application is running in "legacy" mode.
    - [ ] During this stage, the first time a legacy name is found, a warning message is put in the log; note that it can be found in either the preferences (at startup) or in a panel file or script operation later.
    - [ ] At normal shutdown, the legacy list is written out as a "legacy-name.csv" file containing the prefix, type letter and suffix information for each found legacy-name NamedBean.
    - [ ] Provide a migration script that combines a legacy-name.csv file, a panel file, and a replacement prefix to create a new panel file with all the system names changed.
    
The idea is to get some broad experience with that approach. The conversion tool is not as trivial as some seem to think, as system names can be embedded in e.g. other systems names for e.g. SignalHeads and SignalMasts, user names don't have to be separate from system names; etc. (Note that this tool doesn't handle problems in scripts; I'm not sure that they can be reliably automatically updated like this, but perhaps a semi-automated approach is possible) (Another proposed approach would be to fix the system names on loading, but that requires finding _all_ the places in the code where resolution is done; that doesn't seem realistic given that you can e.g. take a name from a Memory value)

- [ ] If not in legacy mode, when writing panel files reference a new version of the schema that enforces parsable system names.

---

Get all that working, get some experience with migrations by asking people via jmriusers to try them.  Then, at the start of the JMRI 4.13 cycle:

- [ ] At start-up, if a non-parseable LRST prefix is found a suppressible warning dialog is popped (if not headless). This is a once-per-run thing, and checking the checkbox allows you to suppress it. It tells you that you need to update that prefix, and points you to more info on how to it both in preferences and also panel files, etc.

- [ ] And finally, at some later point after JRMI 4.14 (e.g. more than a year from now) remove all that migration code.

Would very much appreciate thoughts on the overall plan.

A couple additional minor items, noticed in passing, that are marginally on point and that somebody might want to handle (these could also eventually be separate Issues):
- [ ] The tables inside a Route editor should have their system name column changed over to NamedBean type, so they sort the same way
- [ ] In general, remove the last remaining SystemNameComparator uses in favor of direct operations on NamedBeans

rhwood, 2018-01-01:
The JSON client has never be operable (its a stub that I started but never completed), so the best way to deal with it might be to remove it.

bobjacobsen, 2018-01-02:
I've had several email conversations with users who came to different conclusions, would appreciate guidance.  The question is whether it's better to change e.g. "DCCPP" to "D2" or just "D", change "DP" (for DCC4PC) to "D3" or just "D", etc.

- Having all the defaults being unique means there are less likely to be collisions:  Very few, if any, of these systems have a reason for connecting more than one; people who do connect more than one likely know what they're doing.  And _that_ means that the documentation can reflect what you actually see:  The pages an talk about "sensors with names like D2S12" and be pretty sure such things exist.  Using just "D", just "M", etc raises the chance of collisions so somebody reading "sensors with names like DS12" will have to mentally translate that to "Well, on my layout that's D2S12" more often. (Note: JMRI already updates the last digit to avoid collisions when adding a new system, that's not the question here)

- But the vast majority of people have just one connection, and the extra digit in D2 is just a burden for them.

Of the roughly half-dozen users I've talked to, most seem to agree with the second point.  Only one (who doesn't have multiple connections, but might some day) preferred the first point.

Should these defaults all be set to single letters?

silverailscolo, 2018-01-02:
Thanks for sharing this plan with timing.
It seems wise to seriously consider whether -from a support perspective- enforcing the digit would indeed reduce the number of users asking for help. If that is to be expected, we -as users- will be able to adapt and get used to this, just like we no longer have any trouble using a (first) system letter, even though in practice there is no use for it on my simple home layout...
Now when the plan includes a step to make any system connection prefix distinguishable in the System Name, I expect the user confusion to translate the documentation on "D2.123" to match their private "D.123" won't be so common since they would just be looking for the delimiter, making the case for point 1 less convincing as long as the delimiter is to be introduced in the same version.

As for the usefulness of 2 connections to the same system, I again would like to suggest a use case where a panel file is being developed with a simulator connection, disabling one while using the other. Personally, I find assigning (and remembering) meaning using a (second) letter far easier than using digits.

devel-bobm, 2018-01-02:
My two cents:

Implementing multiple connection types using the same single character prefix by default is just plain ambiguous.

Remember, one of the reasons for this issue's existence is because people are using ambiguous system name prefixes and that is causing difficulty!  Propose ambiguity to solve ambiguity?  Seems like the wrong direction to go...

bobjacobsen, 2018-01-02:
> Implementing multiple connection types using the same single character prefix by default is just plain ambiguous.
>
> Remember, one of the reasons for this issue's existence is because people are using [...]

I don’t think I understand your point.  Could you clarity?

“L” for LocoNet, “N” for NCE, etc is not fixed: Users can configure them to something else. Doesn’t happen often, but it certainly does happen.  

There’s no ambiguity about it at all:  That connection has “D” as its system letter, not “L”.  The system names will be DT1, DS1, DR1, etc; also not ambiguous.

That’s different from the current ambiguity: What’s the system prefix in DCTTS12?  

So I think I must be misunderstanding your point.

devel-bobm, 2018-01-02:
Just my opinions; since I am not volunteering to work this set of issues, I accept the choices of the coders who take it on.  @bobjacobsen , thanks for spearheading this!

The ambiguity I see, and have a problem with, is that what started out as a "system type" letter no longer is a "system type" letter.  It is simply a randomly-chosen letter (or four) that means something to someone, that probably relates to a system connection type.  It is impossible to know what it means without seeing the connection-specifiction(s) and perhaps the context in which it was used.

So far as I can tell, the "system prefix" in "DCTTS12" would be "DCTT", with the object type "S12" specifying the 12th (or 13th) "sensor".  Could be the 122nd "D" system connection, or the 22nd "DC" system connection or the second "DCT" connection or the first NCE cab bus system connection, or ...

Give us the configuration of connection(s), and we (and JMRI) should be able to figure out what that system name means.  And figure out if it is erroneous (not related to a specific connection).

It seems to me that a lightly-made decision on system names will simply kick the problem down the road a bit longer.  I envision that having the same "default system prefix" character for multiple connection types is kicking the problem down the road.

Both developers and many of the users who provide JMRI support have been recommending that our users focus their efforts on "user names", rather than "system names".  As such, the system name ought not to be a big hurdle for anyone, regardless of what that system name looks like.  

It seems reasonable to me to make things easier on the developers, where we can, when we can; this seems like a time when we can make an attempt.

bobjacobsen, 2018-01-03:
Sorry I haven’t made what’s happening here clear.

The change is not to having letters that can be reconfigured.  That happened in 2006.  Most of the time, people use the default letters.  You really can assume that “L” refers to a LocoNet on a LocoNet layout.  But sometimes “K” does too, because that’s how the 2nd LocoNet is referenced.

The change is that you can no longer refer to a LocoNet as LNRHfoo or any other random string as a prefix:  The system prefix is being forced _back_ to a single Letter, optionally followed by digits.  

The question, then, is what do use a default for the various system connections?  There are more than 26 of them….

> The ambiguity I see, and have a problem with, is that what started out as a "system type" letter no longer is a "system type" letter. It is simply a randomly-chosen letter (or four) that means [...]

Well, that overstates it a bit.  It could be randomly chosen, but I don’t know of anybody who’s actually thrown dice.  I don’t know of anybody with just one connection who’s using anything other than the default.

But there is a question of what to use for the default.  There are more than 26 connection types, so we can’t dedicate a letter to each.  What do we do about Marklin, MERG, MRC and Maple?  DCC4PC and DCC++? TMCC and Tams?  Should they share a letter (M, D and T respectively) or have digits added (i.e. M1, M2, M3…)

> So far as I can tell, the "system prefix" in "DCTTS12" would be "DCTT", with the object type "S12" specifying the 12th (or 13th) "sensor". Could be the 122nd "D" system connection, or the 22nd "DC" system connection or the second "DCT" connection or the first NCE cab bus system connection, or …

The correct answer was “DC”. That was a Turnout named TS12.  “DC” referred to a 2nd Maple setup….

I agree that there was no way to figure out that it was DC vs DCT vs DCTT.  That’s why I’m coding enforcement of the 12-year old convention that system prefixes are one letter with optional digits. Then you know for sure.

> Give us the configuration of connection(s), and we (and JMRI) should be able to figure out what that system name means. And figure out if it is erroneous (not related to a specific connection).

We’ve been over this.  People intuitively think they “should be able to figure out" but it’s not possible, not without parsable names.  That’s the point about DCTTS12 above. Hence the return to parsable names.

> It seems to me that a lightly-made decision on system names will simply kick the problem down the road a bit longer. I envision that having the same "default system prefix" character for multiple connection types is kicking the problem down the road.

Is that a vote for e.g. M2, M3, etc? We can certainly do that.

> Both developers and many of the users who provide JMRI support have been recommending that our users focus their efforts on "user names", rather than "system names". As such, the system name ought [...]

On this, we agree.  Hence the change.

But if I still misunderstand, I’d be happy to hear a specific example of what you think the problem is.

Bob

rhwood, 2018-01-03:
I recommend that we use the default letter for a prefix followed by the next available number for that letter. Treat a single letter connection prefix as if it were followed by the number 1.

Example: Given an existing connection "L", the default for the next LocoNet (or other L...) connection would be "L2". In the absence of any connection types with a prefix "L...", the default for a LocoNet (or other L...) connection would be "L1".

bobjacobsen, 2018-01-03:
Not entirely sure, but I think the following is a recommendation that the default for DCC++, DCCPC et all should be “D”, Marklin, MERG and MRC should be “M”, etc.

> I recommend that we use the default letter for a prefix followed by the next available number for that letter. Treat a single letter connection prefix as if it were followed by the number 1. [...]

That’s how it’s been working.  If you have a connection that uses L, the next one that defaults to L will be L2.  

> In the absence of any connection types with a prefix "L...", the default for a LocoNet (or other L...) connection would be "L1".

Why?  Most people have only one connection.  Isn’t LT123 easier to work with than L1T123?

rhwood, 2018-01-03:
> Why?  Most people have only one connection.  Isn’t LT123 easier to work with than L1T123?

I don't think LT123 is any more difficult than L1T123, but I think that working with two different patterns (LT123 and L2T123) is more difficult than working with a single pattern (L1T123 and L2T123)

pabender, 2018-01-03:
Sent from my iPhone

> On Jan 2, 2018, at 8:18 PM, Randall Wood <notifications@github.com> wrote:
> I don't think LT123 is any more difficult than L1T123, but I think that working with two different patterns (LT123 and L2T123) is more difficult than working with a single pattern (L1T123 and L2T123)
> 
It is still a single pattern:

[A-Z]([0-9]*)[RSTL]ADDRESS

Where ADDRESS varies based on the system in use.

Paul

dheap, 2018-01-03:
> It is still a single pattern:
> 
> [A-Z]([0-9]*)[RSTL]ADDRESS
> 
> Where ADDRESS varies based on the system in use.

Which allows for:

- MT123
- M1T123
- M578T123

I don't see confusion or ambiguity problems with that.

dheap, 2018-01-03:
Markdown interpreted the regex in previous comment as a URL, edited to fix.

devel-bobm, 2018-01-03:
To my way of thinking, the first character of the pattern ` [A-Z]([0-9]*)[RSTL]ADDRESS ` is ambiguous.  At one time, we could infer a specific "connection type" from a system name.  But that is no longer the case, for good reason.  With current JMRI, the best one can say that it encodes a system connection, and one must jump thru some reasonably-well-defined hoops to figure out the "connection type" (if one has the appropriate connection information).

That proposal would continue the "connection type" ambiguity.  So be it.  Code wins.  But there are other options.

An alternative proposal: something like ` [A-Z][A-Z]([0-9]*)[RSTL]ADDRESS `, with each connection type having a unique, fixed, "prefix" character pair.  If concerned about capacity for additional connection types, one could implement three for four alpha characters as the "prefix", instead of two.

KenC57, 2018-01-03:
Bob M,

While a two char prefix should likely out last us, it is really the same as
a single prefix char. So instead of 26 prefixes, you'd have 26*26, just a
much bigger, but still finite list.

Personally I don't believe trying to 'infer' from the prefix is right. You
need the context of the profile to know for sure. I've got a panel somewhere
that has LT* but it runs on a NCE connection with the 'L' prefix. Yes he had
started with Digitrax and changed.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

bobjacobsen, 2018-01-03:
> An alternative proposal: something like [A-Z][A-Z]([0-9]*)[RSTL]ADDRESS, with each connection type having a unique, fixed, "prefix" character pair. If concerned about capacity for additional connection types, one could implement three for four alpha characters as the "prefix", instead of two.

That’s a really interesting idea.

To make it really “unambiguous”, you’d probably have to encode a bit more than just the manufacturer. For LocoNet, you might have LC for LocoNet with command station, LS for LocoNet standalone, LP for LocoNet PR3 to cover some of the larger variation.  And there would still be lots of combinations left over!

Getting this in place would be complicated.  Most existing panel files aren’t compatible, so the migration would be _much_ larger with this proposal.  And [A-Z]\d*[RSTL] doesn’t distinguish from [A-Z][A-Z]\d*[RSTL]; they can’t coexist. But making the second letter lower case: [A-Z][a-z]\d*[RSTL] would allow them to coexist, simplifying migration.  Though names like LpT123 do look odd to me.

devel-bobm, 2018-01-03:
@kenc57,
 
System names beginning with "L" (or "LT) running on an NCE system would be a candidate for "migration" to whatever designator would be chosen for "NCE Cab Bus Interaface via NCE-USB Interface Device" (or whatever two-or-more-character prefix and associated long-winded system connection description actually applies).

@bobjacobsen offers an interesting idea with "lower-casing" the standardized second character: '[A-Z][a-z][RSLT]ADDRESS'.  It offers an advantage in "migration", as identifying un-migrated system prefixes ought to be fairly obvious.

A quick check shows that I can change a LocoNet connection from prefix "L" to "La", and it is persisted in the connection portion of a panel file in mixed case.  But if I add turnout 140 via the Turnouts table (current-ish JMRI code), something forces the system name to "LAT140".  I presume that similar upper-casing is applied to other connection-related object types.  If we go with mixed-case connection type prefixes, then we have a bit of work to do to "unwind" that forced-upper-casing.

For the sake of expansion, we could make the pattern be '[A-Z][a-z]([a-z]*)[RSLT]ADDRESS', giving the ability to expand beyond two required characters, where appropriate, while keeping the option of two-character system prefixes.  Extra complexity, but extra flexibility.

pabender, 2018-01-03:
On 01/03/2018 12:26 AM, Bob Jacobsen wrote:
> To make it really “unambiguous”, you’d probably have to encode a bit
> more than just the manufacturer. For LocoNet, you might have LC for
> LocoNet with command station, LS for LocoNet standalone, LP [...]

But what happens when someone switches from a PR3 to a LocoBuffer or 
switching from a simulator to a real connection?  Or to ask that another 
way, at what point do we consider something significant enough to 
warrant using the second letter?

And then there's the whole question of how does this work with the 
simple server, where the connection prefix is supposed to match the 
connection prefix on the remote side.

I personally still prefer prefixes of the form [A-Z]([0-9]*).  I don't 
find that ambiguous, even if "L" could be loconet on one person's 
installation and NCE or Lenz, or... on someone elses.

Parsing ambiguity for system names only occurs within the context of a 
single profile.  We don't need to make the names globally parsable, 
which appears to be the goal of adding the second letter to the system 
name.

In other words, the names have a scope of a single JMRI profile. The 
profile gives us the context required to determine what a given system 
name means.  They don't really have meaning outside of that JMRI 
profile, and we shouldn't assume that they do.

Paul

devel-bobm, 2018-01-03:
System connection ambiguity applies whenever the system connection information is separate from the data which implies it.  Such ambiguity applies to jython code.  

It may apply in a hypothetical case (something I have thought of but never tried) where an XML file has been split apart.  I envision a user having one XML file for connection, turnout, sensor, reporter, light definitions, and one XML file for panel defintion, and one XML file for logix definition.  There are cases where hand-editing XML is far-easier than working thru the JMRI GUI, and having the different XML files may be advantageous in such cases.

How does one transition from a PR3 acting as a LocoNet interface to a LocoBuffer, all other things being equal?  It depends on how the user has his configuration set up.

If the user has only one "connection", there's only one prefix in-use.  When he modifies the conneciton from PR3 to LocoBuffer-USB, the system prefix is changed from, say "Dp" to "Ru".  When JMRI restarts, the "migration" process offers to change system names from "Dp" to "Ru".  If the user refuses, then none of his pre-defined "Dp*" objects will "connect".

(Or, perhaps, the PR3 (interface mode) and LocoBuffer-USB capabilities are similar enough that they both implement prefix "Li", and no migration is needed.  The user simply changes the connection hardware specified in the profile, and restarts JMRI, without any further "migration".)

In a second case, where the user has more than one non-disabled connection, and wishes to "flip" between the pre-existing PR3 connection (with prefix "Dp" or "Li") and the pre-existing LocoBuffer-USB conenction (with prefix "Ru" or "Li2"), then the user must intentionally perform some "migration" step to convert objects with system names starting with "Dp" to "Ru" (or "Li" to "Li2").  Then the user must re-start JMRI.

Similarly with LocoNet simulator.  Either the simulator connection has a unique system name prefix and the user must "migrate" his system names to/from a prefix suitable for the "loconet simulator" , or the simulator uses the same prefix as other common LocoNet interfaces and the user simply changes the connection type, without need for "migration" of existing system names.

NCE connections may have a similar case where different hardare might possibly share the same "system name prefix".  I know that there are some subtle differences, and this may prevent using a common system name prefix.  The same is true for LocoNet hardware interfaces!  Further study would be required.

The "migration" is more difficult for those system names which are not directly managed by JMRI, such as jython.  But the user of jython has always had this problem when changing system names, so I don't see this as a show-stopping issue.

The question of the "simple server" might be argued to be a case where system name parsing ambiguity has gone beyond the scope of a single profile.  The "simple server" is beyond the scope of my knowledge.  Do "User names" work across the "simple server" interface?  If not, should they?  If so, should/can we "ban" use of system names across the simple server?  Would it make any sense to "advertise" any "system name changes" made during a "migration" over the "simple server"?

pabender, 2018-01-03:
On 01/03/2018 02:31 PM, Bob Milhaupt wrote:
>
> System connection ambiguity applies whenever the system connection 
> information is separate from the data which implies it. Such ambiguity 
> applies to jython code.
>

Sorry, but you're conflating two different concepts.  When we talk about 
ambiguity in parsing, we are specifically talking about some subset of 
an input stream that can have two or more means within the same context 
(When I do this with students, we draw parse trees for a specific 
grammar and end up creating two different parse trees for the same input.)

If you take something out of context, you NEVER have any grantee that 
some name will even have a valid meaning.  This is the argument you are 
trying to make.

> The question of the "simple server" might be argued to be a case where
> system name parsing ambiguity has gone beyond the scope of a single
> profile. The "simple server" is beyond the scope of my [...]

The Simple Server and the JmriClient sent and receive system names. The 
client is able to tell the server it wants to create a new device on 
because the system names are unambiguous.  It never sends user names 
because user names don't necessarily have a meaning on the server AND 
you can have more than one client that reference different objects using 
the same user name.

bobjacobsen, 2018-01-03:
(We were originally talking about default system prefixes, but this has now morphed to be about restricting non-default prefixes, hence forcing a pattern)

> System connection ambiguity applies whenever the system connection information is separate from the data which implies it.

But why is it a problem? Seems like a good thing to me. 

With that definition, a lot of what JMRI does is dedicated to _creating_ “ambiguity”. You don’t need to know the specific type of a Turnout to manipulate it. 

The purpose of a _name_ is to identify something. A name isn’t the same as a complete description of type and state: it’s just a (mostly) unique identity in a convenient form. 

“Bob Jacobsen” doesn’t carry a lot of information about my type and configuration. Even using my drivers license number as an ID instead carries the information that I (at least at one time used to) live in California and have (or at least once had) a license, but that says nothing about other dimensions. 

The LT123 name (assuming parsable) tells you this is a Turnout, but that’s about it. And for a dozen years, that’s been fine. You’d need to look that object up to know if it has feedback sensors, has associated operations, what it’s current state is etc. The vast majority of JMRI scripts and jmri.jmrit tools simply don’t care what type Turnout is there. 

We certainly can put more type & configuration detail into the name. A full description is probably possible in a couple hundred bytes, which we could maybe code down to 10-25. What does that gain?

Even if we just code coarse connection type in two letters, what does that gain?

Here’s an example of what it costs: three times in the last two weeks people have sent me pieces of layout logic in a panel file. (Yes, they could have sent the entire profile, but I don’t have their hardware). I set up Internal systems with the same letters, created a bunch of objects of a different type, and away we went. Yes, I could use some (non-existent) tools to do massive renaming, but then I’d just have to undo that to send the file back. 

It seems to me that “system ambiguity” is a good thing. What am I missing?

bobjacobsen, 2018-01-03:
> In a second case, where the user has more than one non-disabled connection, and wishes to "flip" between the pre-existing PR3 connection (with prefix "Dp" or "Li") and the pre-existing [...]

The above pretty much settles this question for me:  I _really_ _really_  don’t want to have to do a migration step every time I switch between using a panel file on a real hardware and on a simulator. Nor when I switch between two different hardware implementations.

*) If the migration is hard and/or imperfect, I’m going to be annoyed at this a couple times a week.

*) If the migration is trivial, I’m _still_ going to be annoyed at it a couple times a week, because we’re now going to a lot of trouble to preserve a distinction (“Ru vs Dp is an important thing to know!”) that the migration itself demonstrates is trivial.

PR #7919 (2020-01-12): XML Schema enforces parseable system names
This creates a new layout-4.19.2 XML Schema that enforces proper parseable system names for Turnouts, Lights and Sensors.  Newly-written files will reference this schema, so will have to have proper names; earlier PRs are enforcing that internally.

For more info, see:
Issue #4670 
PR #7781
PR #6490
PR #5664
PR #4758
PR #4753

This closes #4670
closed 2020-01-12 by bobjacobsen