JMRI/JMRI #4673: Controlling TWO  locos with one transponding, results in reporter table swapping.

ClockLoop, 2018-01-01:
I found that when there are multiple transponding locos being controlled by a single throttle or multiple throttles, the transponding ends up confusing which throttle is controlling which loco and will put the wrong loco number into the reporters table.

I tested with a single transponding locomotive and two different physical throttles, the dcs51 and the dt500, and the jmri virtual throttles.  They all make this happen once you try to control the second slot.

So technically what I am seeing is if I have a single transponding loco working fine and reporting fine, but when I try to control another loco even if it dosen't even exist on my layout, (i just enter the loco into my physical throttle and try to give it commands)  This will cause the reporter to report into the location that my real transponding loco is in, the report that belongs to the "fake loco",  (which dosen't actually exist)

I have a DT500 throttle, 1-DCS51 throttle configured as a slave, a 1-DCS240 command station, 1-PS2012 power supply, and 5-DS64 and 4-BXP88.
JMRI is build 4.11.1+Rfe44680 on windows7.

I am seeing the transponder reports in the loco net messages, what tells the bxp88 which throttle is controlling the transponding train?

devel-bobm, 2018-01-02:
Here are some thoughts - I am far from a transponding expert, so do not consider my thoughts to be "authoritative".

Is your problem with JMRI or with the Digtrax Throttle's display of the transponding information?  (I.e. the "Find" button on a DT400x , DT402x, or DT500x?)  That feature should function without ANY JMRI interaction.  Any problem you have with the Digitrax throttle's display of transponding information is likely a Digitrax problem.

Similarly the "debouncing" issue - that seems like a track pick-up issue, not a JMRI issue.

So far as I know, JMRI does not attempt to relate a loco address to a throttle via transponding - there just is not enough information available in the LocoNet message.

Pugwash1, 2018-01-02:
The loco reported is normally the last message seen by jmri. So if loco is reporting for a section 1234 present, 5678 present then only 5678 will show. The bouncing absent present messages are cross talk. Nothing to do with jmri. The 88's need to as close to the blocks that it's monitoring as possible. The electrical  challenge can be seen here https://louko.com/alo/DCC/Transponding/

bobjacobsen, 2018-01-02:
To add to some of the earlier contents, the Reporter table just shows the last message each Reporter is reporting.  Sometimes messages come so fast that what you see there is only part of the story.

The jython//ReporterFormatter.py sample script tries to put that info into a more usable form, showing more than one decoder.

ClockLoop, 2018-01-02:
"The bouncing absent present messages are cross talk. Nothing to do with jmri. The 88's need to as close to the blocks that it's monitoring as possible. The electrical challenge can be seen here https://louko.com/alo/DCC/Transponding/"

I have run very long lines of solid copper wire to my bxp 88 arrays.    I see now I should have placed the bxp88s closer to the zones.   I do notice zones closer to the bxp88 transpond much more often.

I guess I have some layout work to do before I can really determine if this is a loconet digitrax issue, bxp88 issue, wire length issue, jmri issue, or my zones have too much track.

Thanks for the help.   I will move the bxp88's to be much closer to their zones.   One died on me along with a sdxn136 decoder so i need to pull it apart somewhat anyway.

Then I can test this again.  Gotta order more bxp88s tho.

ClockLoop, 2018-01-02:
Having a bit of bxp88 code would help me here!   A way to configure the bxp88 in the loconet menu would at least let me read back the bxp88 switches so i can see if its talking at the right address, and has the right settings.

I don't know java so I am useless here in jmri if I wanna contibute.  I only know embedded code so I can make devices to talk to the loconet or show info.

Perhaps I will, like some readout display that shows the status and settings of all devices on the loconet in a big, live spreadsheet.

devel-bobm, 2018-01-02:
Investigations I've done lead me to believe that the BXP88 does not support OpSw programming in the same way that DS64, SE8c, BDL16x, and PM4x devices can be programmed.  I have made a support request with Digitrax that this feature be added.

ClockLoop, 2018-01-02:
If this is truly crosstalk, then here is the log result of crosstalk.
I do have engine 1 running, while engine 6 is running on a entirely different track and loop.  The only commonalities between train 1 and train 6 and their tracks, is the wires feeding the tracks are bunched together and go longer than 3 foot alongside eachother, and at some points tightly pressed together from going through holes and things.
I am now rewiring my entire layout, relocating the bxp88's within 1 foot of the track, then rerunning the power, ground and loconet cables.

`17:28:41.506: [D0 20 28 00 01 26]  Transponder address 1 (short) present at LR41 (Loop1-A) (BDL16x Board ID 3 RX4 zone E).
17:28:42.056: [D0 00 28 00 01 06]  Transponder address 1 (short) absent at LR41 (Loop1-A) (BDL16x Board ID 3 RX4 zone E).
17:28:42.246: [D0 20 28 00 06 21]  Transponder address 6 (short) present at LR41 (Loop1-A) (BDL16x Board ID 3 RX4 zone E).
17:28:42.666: [B2 14 50 09]  Sensor LS41 (Loop1-A) is High.  (BDL16 # 3, DS9; DS54/DS64/SE8c # 6, AuxA/A1/DS01).
17:28:42.826: [D0 00 28 00 06 01]  Transponder address 6 (short) absent at LR41 (Loop1-A) (BDL16x Board ID 3 RX4 zone E).
17:29:02.436: [D0 20 28 00 01 26]  Transponder address 1 (short) present at LR41 (Loop1-A) (BDL16x Board ID 3 RX4 zone E).
17:29:03.236: [D0 00 28 00 01 06]  Transponder address 1 (short) absent at LR41 (Loop1-A) (BDL16x Board ID 3 RX4 zone E).
17:29:04.726: [B2 14 70 29]  Sensor LS42 (Loop1-B) is High.  (BDL16 # 3, DS10; DS54/DS64/SE8c # 6, SwiA/S1/DS02).
17:29:19.056: [D0 20 29 00 06 20]  Transponder address 6 (short) present at LR42 (Loop1-B) (BDL16x Board ID 3 RX4 zone (Unknown 9)).
17:29:19.536: [D0 00 29 00 06 00]  Transponder address 6 (short) absent at LR42 (Loop1-B) (BDL16x Board ID 3 RX4 zone (Unknown 9)).
17:29:22.156: [B2 14 40 19]  Sensor LS41 (Loop1-A) is Low.  (BDL16 # 3, DS9; DS54/DS64/SE8c # 6, AuxA/A1/DS01).
17:29:24.286: [D0 20 2A 00 01 24]  Transponder address 1 (short) present at LR43 (Loop1-C) (BDL16x Board ID 3 RX4 zone F).
17:29:24.376: [B2 15 50 08]  Sensor LS43 (Loop1-C) is High.  (BDL16 # 3, DS11; DS54/DS64/SE8c # 6, AuxB/A2/DS03).
17:29:24.946: [D0 00 2A 00 01 04]  Transponder address 1 (short) absent at LR43 (Loop1-C) (BDL16x Board ID 3 RX4 zone F).
17:29:37.676: [D0 20 2A 00 01 24]  Transponder address 1 (short) present at LR43 (Loop1-C) (BDL16x Board ID 3 RX4 zone F).
17:29:38.346: [D0 00 2A 00 01 04]  Transponder address 1 (short) absent at LR43 (Loop1-C) (BDL16x Board ID 3 RX4 zone F).`

ClockLoop, 2018-01-02:
Thats odd that the bxp88 can't be configured.
I do it the same way as the other devices, at least with the throttle.
The DS64 has a config page and its settings are all set with the throttle, and same with the BXP88
It must be related to how you get the bxp88 into the mode that will let you change the id or settings, because with the throttle one must first press a button on the bxp88.

They have released two firmware updates since the bxp88 was released.   
Its on rev 0.2

devel-bobm, 2018-01-02:
There are two different mechanisms at play here.  One, where the USER puts the board into an "opsw programming mode" and uses normal "switch control" LocoNet messages to read and set board OpSw settings, and another, intended for computer-based mechanissms, where the computer uses entirely different LocoNet messages to send an opsw write or opsw query to a specific board type with a specific board address.

The two are radically different.

The BXP88 firmware implements only the first type.  I infer from reports from the field that the first mechanism works just fine for the BXP88.

ClockLoop, 2018-01-04:
I am still looking into this and have rewired my layout for the first circle plus a few extra tracks to show the transponding problem.

Here is a video that shows the jmri desktop capture of loco 1 going around my layout with transponding working.   
The transponding dosen't work like a sensor, its on and then off.  Transponding messages are intermittent, thus i don't see how you can rely on the present absent messages to determine the enters or exits...
22:22:13.479: [D0 20 39 00 01 37]  Transponder address 1 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
22:22:13.706: [D0 00 39 00 01 17]  Transponder address 1 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
Perhaps the bxp88 works different or my layout is "all wrong" however, at some point make a simple circle of track with gaps and multiple bxp88s just to demonstrate that either it works differently or not how one thinks.   The distance from the bxp88 to the track has its limits to how close you can get it, but i have gotten mine with in 1 foot of the track, my wire is copper, the AB track wires are twisted together only until they reach the track and bxp88.    I am considering using coax cable..  :) (yea like catv)  But really for just 8 inches of wire?
So i made a video of the transponding.
http://www.youtube.com/watch?v=-k9y6BBt3do

ClockLoop, 2018-01-04:
I HAVE confirmed that this is not a crosstalk issue.

I did this by putting loco1 on loop1 and make it go around, watch for transponding messages for loco 1.

I then put loco7 on a track labeled crossover-5, 
"00:19:07.285: [B2 18 70 25]  Sensor LS50 () is High.  (BDL16 # 4, DS2; DS54/DS64/SE8c # 7, SwiA/S1/DS02)."

I then waited to see if loco1 took on loco7's name.
It did not, so then i took my dt500 throttle and selected loco7 with the secondary throttle spot.
00:23:34.886: [BF 00 07 47]  Request slot for loco address 7 (short).

The loco7 started transponding...
00:23:45.946: [D0 20 31 00 07 39]  Transponder address 7 (short) present at LR50 () (BDL16x Board ID 4 RX4 zone (Unknown 1)).
00:23:46.726: [D0 00 31 00 07 19]  Transponder address 7 (short) absent at LR50 () (BDL16x Board ID 4 RX4 zone (Unknown 1)).

..and soon enough the two locos were swapping their names.
"00:23:49.446: [D0 20 31 00 01 3F]  Transponder address 1 (short) present at LR50 () (BDL16x Board ID 4 RX4 zone (Unknown 1)).
00:23:50.016: [D0 00 31 00 01 1F]  Transponder address 1 (short) absent at LR50 () (BDL16x Board ID 4 RX4 zone (Unknown 1)).
00:23:57.046: [D0 20 31 00 07 39]  Transponder address 7 (short) present at LR50 () (BDL16x Board ID 4 RX4 zone (Unknown 1)).
00:23:57.566: [D0 00 31 00 07 19]  Transponder address 7 (short) absent at LR50 () (BDL16x Board ID 4 RX4 zone (Unknown 1))."

On both locos, the names are mixing up.  And remember, loco7 DOSENT EVER MOVE FROM CROSSOVER-5
00:25:22.276: [D0 20 39 00 07 31]  Transponder address 7 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:25:22.736: [D0 00 39 00 07 11]  Transponder address 7 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).

I then lifted loco7 totally off the track and let loco1 keep going.
"00:34:27.727: [B2 18 60 35]  Sensor LS50 (Crossover-5) is Low.  (BDL16 # 4, DS2; DS54/DS64/SE8c # 7, SwiA/S1/DS02)."

Loco1 reported being 7 right away...
00:34:44.247: [D0 20 39 00 07 31]  Transponder address 7 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:34:44.707: [D0 00 39 00 07 11]  Transponder address 7 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
and kept doing it... while loco1 went around the track.
00:37:09.587: [D0 20 39 00 07 31]  Transponder address 7 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:37:10.107: [D0 00 39 00 07 11]  Transponder address 7 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).

And it would say it was its self too.  In the same zone, while having never left it.

00:39:08.657: [D0 20 39 00 01 37]  Transponder address 1 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:39:09.177: [D0 00 39 00 01 17]  Transponder address 1 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:39:17.627: [D0 20 39 00 07 31]  Transponder address 7 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:39:18.117: [D0 00 39 00 07 11]  Transponder address 7 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).

00:40:12.597: [D0 20 2C 00 01 22]  Transponder address 1 (short) present at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).
00:40:13.287: [D0 00 2C 00 01 02]  Transponder address 1 (short) absent at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).
00:40:14.007: [D0 20 2C 00 07 24]  Transponder address 7 (short) present at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).
00:40:14.777: [D0 00 2C 00 07 04]  Transponder address 7 (short) absent at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).

I then deselected 7 on my dt500 throttle, and replaced it with loco0
"00:42:07.327: [B5 02 13 5B]  Write slot 2 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
00:42:09.107: [BF 00 00 40]  Request slot for loco address 0 (short).
00:42:09.117: [E7 0E 03 03 00 00 00 47 00 00 00 00 00 51]  Report of slot 3 information:"

It wouldn't report loco1 on any part of the track then, so i did a find.  It found it and transponded a few times by its self, but reported 1 properly, never 7, but i should have let the train go around the track here more before doing the next step.
00:46:59.798: [E5 09 40 7D 01 00 00 00 2F]  Transponding Find query for loco address 1 (short) (or long address 16001).
00:46:59.808: [E5 09 00 7D 01 00 39 00 56]  Transponder Find report: address 1 (short) (or long address 16001) present at LR58 (Loop1-A) (BDL16x Board 4 RX4 zone unknown (9)).
00:47:01.348: [D0 20 39 00 01 37]  Transponder address 1 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:47:01.688: [D0 00 39 00 01 17]  Transponder address 1 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:47:02.178: [D0 20 39 00 01 37]  Transponder address 1 (short) present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:47:02.548: [D0 00 39 00 01 17]  Transponder address 1 (short) absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:47:06.628: [EF 0E 7B 01 6D 78 48 47 76 03 70 11 0A 60]  Request Fast Clock is  Running, rate is 1:1. Day 3, 14:05. Last set by ID 0x0A 0x11 (1297).
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available.
00:47:06.628: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
[... cut ...]
00:47:48.218: [D0 00 2C 00 01 02]  Transponder address 1 (short) absent at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).
00:47:50.158: [A0 01 0D 53]  Set speed of loco in slot 1 to 13.
00:47:51.728: [B2 18 40 15]  Sensor LS49 (Loop1-B) is Low.  (BDL16 # 4, DS1; DS54/DS64/SE8c # 7, AuxA/A1/DS01).
00:47:55.458: [D0 20 2C 00 07 24]  Transponder address 7 (short) present at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).
00:47:56.028: [D0 00 2C 00 07 04]  Transponder address 7 (short) absent at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).

ClockLoop, 2018-01-04:
I then close virtual loco7 and open a totally new virtual loco307
And get this, the transponder reports loco1 as loco307.
"00:48:15.618: [BF 02 33 71]  Request slot for loco address 307.
00:48:15.788: [E7 0E 04 03 33 00 00 47 00 02 00 00 00 67]  Report of slot 4 information:
	Loco 307 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x00 0x00 (0).
00:48:15.798: [EF 0E 04 03 33 00 00 47 00 02 00 71 02 1C]  Write slot 4 information:
	Loco 307 is Not Consisted, Free, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
00:48:15.798: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
00:48:15.798: [BA 04 04 45]  Set status of slot 4 to IN_USE.
00:48:15.798: [E7 0E 04 33 33 00 00 47 00 02 00 71 02 24]  Report of slot 4 information:
	Loco 307 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
00:48:17.898: [A0 02 00 5D]  Set speed of loco in slot 2 to 0.
00:48:19.948: [A0 03 00 5C]  Set speed of loco in slot 3 to 0.
00:48:21.178: [D0 20 2C 00 01 22]  Transponder address 1 (short) present at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).
00:48:21.828: [D0 00 2C 00 01 02]  Transponder address 1 (short) absent at LR45 (Loop1-C) (BDL16x Board ID 3 RX4 zone G).
00:48:21.928: [A0 01 0D 53]  Set speed of loco in slot 1 to 13.
00:48:24.768: [B2 16 70 2B]  Sensor LS46 (Loop1-D-LeftWye) is High.  (BDL16 # 3, DS14; DS54/DS64/SE8c # 6, SwiC/S3/DS06).
00:48:27.318: [B2 17 70 2A]  Sensor LS48 (Loop1-E-Programming) is High.  (BDL16 # 3, DS16; DS54/DS64/SE8c # 6, SwiD/S4/DS08).
00:48:27.698: [D0 20 2F 00 01 21]  Transponder address 1 (short) present at LR48 (Loop1-E-Programming) (BDL16x Board ID 3 RX4 zone (Unknown 15)).
00:48:28.458: [D0 00 2F 00 01 01]  Transponder address 1 (short) absent at LR48 (Loop1-E-Programming) (BDL16x Board ID 3 RX4 zone (Unknown 15)).
00:48:29.848: [B2 16 40 1B]  Sensor LS45 (Loop1-C) is Low.  (BDL16 # 3, DS13; DS54/DS64/SE8c # 6, AuxC/A3/DS05).
00:48:32.088: [B2 16 60 3B]  Sensor LS46 (Loop1-D-LeftWye) is Low.  (BDL16 # 3, DS14; DS54/DS64/SE8c # 6, SwiC/S3/DS06).
00:48:50.038: [B2 1C 50 01]  Sensor LS57 (Loop1-F-RightWye) is High.  (BDL16 # 4, DS9; DS54/DS64/SE8c # 8, AuxA/A1/DS01).
00:48:50.208: [D0 20 38 00 01 36]  Transponder address 1 (short) present at LR57 (Loop1-F-RightWye) (BDL16x Board ID 4 RX4 zone E).
00:48:50.788: [D0 00 38 00 01 16]  Transponder address 1 (short) absent at LR57 (Loop1-F-RightWye) (BDL16x Board ID 4 RX4 zone E).
00:48:51.728: [A0 03 00 5C]  Set speed of loco in slot 3 to 0.
00:48:52.298: [B2 1C 70 21]  Sensor LS58 (Loop1-A) is High.  (BDL16 # 4, DS10; DS54/DS64/SE8c # 8, SwiA/S1/DS02).
00:48:53.098: [D0 20 39 02 33 07]  Transponder address 307 present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:48:53.708: [A0 01 0D 53]  Set speed of loco in slot 1 to 13.
00:48:53.738: [D0 00 39 02 33 27]  Transponder address 307 absent at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:48:55.008: [B2 17 60 3A]  Sensor LS48 (Loop1-E-Programming) is Low.  (BDL16 # 3, DS16; DS54/DS64/SE8c # 6, SwiD/S4/DS08).
00:48:57.688: [B2 1C 40 11]  Sensor LS57 (Loop1-F-RightWye) is Low.  (BDL16 # 4, DS9; DS54/DS64/SE8c # 8, AuxA/A1/DS01).
00:49:05.628: [A0 04 00 5B]  Set speed of loco in slot 4 to 0.
00:49:05.818: [B5 04 13 5D]  Write slot 4 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
00:49:05.818: [B5 04 23 6D]  Write slot 4 with status value 35 (0x23) - Loco is Not Consisted, Idle and operating in 128 speed step mode.
00:49:05.818: [D0 20 39 02 33 07]  Transponder address 307 present at LR58 (Loop1-A) (BDL16x Board ID 4 RX4 zone (Unknown 9)).
00:49:06.868: [EF 0E 7B 01 6D 78 4A 47 76 03 70 11 0A 62]  Request Fast Clock is  Running, rate is 1:1. Day 3, 14:07. Last set by ID 0x0A 0x11 (1297).
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available.
00:49:06.868: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
00:49:07.908: [A0 02 00 5D]  Set speed of loco in slot 2 to 0.
00:49:23.502: [A0 03 00 5C]  Set speed of loco in slot 3 to 0.
00:49:25.492: [A0 01 0D 53]  Set speed of loco in slot 1 to 13.
00:49:27.652: [B2 18 50 05]  Sensor LS49 (Loop1-B) is High.  (BDL16 # 4, DS1; DS54/DS64/SE8c # 7, AuxA/A1/DS01).
00:49:31.462: [B2 1C 60 31]  Sensor LS58 (Loop1-A) is Low.  (BDL16 # 4, DS10; DS54/DS64/SE8c # 8, SwiA/S1/DS02).
00:49:50.092: [BB 7F 00 3B]  Request Command Station OpSwitches (or DCS210/DCS240 check for multiple command stations on LocoNet).
00:49:51.312: [BB 7B 00 3F]  Request Fast Clock information.
00:49:51.322: [E7 0E 7B 01 55 72 4B 47 76 03 70 11 0A 59]  Response Fast Clock is  Running, rate is 1:1. Day 3, 14:08. Last set by ID 0x0A 0x11 (1297).
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available.
[... cut ...]
00:49:55.282: [A0 03 00 5C]  Set speed of loco in slot 3 to 0.
00:49:57.272: [A0 01 0D 53]  Set speed of loco in slot 1 to 13.
00:49:57.912: [A0 02 00 5D]  Set speed of loco in slot 2 to 0.
00:49:58.272: [B2 16 50 0B]  Sensor LS45 (Loop1-C) is High.  (BDL16 # 3, DS13; DS54/DS64/SE8c # 6, AuxC/A3/DS05).
00:50:04.022: [B2 18 40 15]  Sensor LS49 (Loop1-B) is Low.  (BDL16 # 4, DS1; DS54/DS64/SE8c # 7, AuxA/A1/DS01).
00:50:27.062: [A0 03 00 5C]  Set speed of loco in slot 3 to 0.
00:50:29.052: [A0 01 0D 53]  Set speed of loco in slot 1 to 13.
"

ClockLoop, 2018-01-04:
To reaffirm this is a digitrax bug, ...
I placed loco1 on the track as before, and did a find on a dt500 throttle..  it reports zone 0044

I then placed loco7 on the track as before and did a find on a dt500 throttle..  of LOCO1 (not 7), it reports zone 0044.

I then selected loco7 with the secondary spot on the dt500 throttle.
I then selected loco1 by pressing the speed knob and did a find of loco1.
Immediately it reports zone 0049 the wrong zone for loco1 so i press find again, and it shows both 0049 and 0044 randomly.

So i selected loco7 by pressing its speed knob and did a find.
It also reports zone 0049 and 0044 randomly if I repeatedly press the find button.

So I remove loco7 from the track and press find for loco7 and IT ONLY REPORTS ZONE 0044, the zone for loco1.

So I select loco1 and press find.  It also now only reports zone 0044.
So I select loco 0 with the secondary throttle.
And press find for loco0 for the hell of it, and it shows nothing, repeatedly.(like it should)
So I select loco1 and press find..  and reports zone 0044. like it should.

So I remove loco1 from the tracks and press find.  And nothing..
so I select loco7 again and press find, and nothing.
So I put loco7 back on the tracks in zone 0049 and press find... and it reports zone 0049.
So I reselect loco1 and press find and it reports...  ZONE 0049.  but its not on the tracks.

The digitrax hardware for transponding with multiple bxp88s is bugged at this point in time for the latest firmware available.

KenC57, 2018-01-04:
From all that tracing, I'm thinking the problem is the transponding mixing
it up since it reports the address. So the JMRI side is only reporting what
it is being told. I'm thinking Digitrax has a bug somewhere. Very nice
trace, I'd say talk to Digitrax and show them your findings.
-ken c

ClockLoop, 2018-01-04:
I ran a true test with two dz123 decoders running on two loops on the layout.
They work perfectly.

I then put two sdxn136ps decoders running on the same two loops, and all transponding is mixed up.
Yea., contact them and want have them tell me to rma my 4- BXP88 and 12-SDXN136PS ?
Who do I charge for the removal and re-install?  Seriously?  Nscale decoders are no fun to do.  Nor is disconnecting things like the bxp88s..   Mabee they should send a tech out to me, and do the work since I paid my part.

KenC57, 2018-01-04:
Ok, you say it worked with one pair of decoders, but failed with a different
pair. Did you try one of each? I'm thinking is it the second model decoder
that may be scrambling the messages. Granted the next beyond that is a third
or more different models of decoders to see if it follows the models or not.

While it may be unlikely, talking with Digitrax may give you some clues or
they may even know of some incompatible details. Past history says it is
less likely they will admit some of that or more likely the odds being low
of you talking to somebody who understands that sort of thing.
-ken c

ClockLoop, 2018-01-04:
I did try one dz123 with a sdxn136ps, and the dz123 reports improper address, while the sdxn136ps reports proper.
One would say its the dz123 then but no, two dz123s work together, but two sdxn136ps do not work together, I did talk about this, but granted I put alot of info here in a short time.

I am 100% certain that 2- sdxn136ps transponding on a bxp88 will introduce this digitrax bug.

ClockLoop, 2018-01-04:
I downgraded my bxp88s to file bxp88_9sep17 instead of the latest bxp88_21oct17, and the tool says I still have version 0.2 so it will NOT let me downgrade the firmware.  So I can't test if they introduced bugs since the last release version.  However the sep17 file is no longer available so it was probably bad anyway.

pabender, 2018-01-04:
On 01/04/2018 10:32 AM, ClockLoop wrote:
> I did try one dz123 with a sdxn136ps, and the dz123 reports improper
> address, while the sdxn136ps reports proper.
> One would say its the dz123 then but no, two dz123s work together, but
> two [...]

So what happens when you have the two dz123s and one sdxn136ps?

Also, this is pretty clearly a hardware issue.  JMRI is just reporting 
what it sees, so this isn't a JMRI issue.  You really need to report 
this to Digitrax.

ClockLoop, 2018-01-04:
I made tickets with digitrax on 4 issues, my dead bxp88, a dead sdxn136ps, the SDXN136PS transponding failure, and the dz123 transponding failure on its consist address.

pabender, 2018-01-04:
Since this isn't a JMRI issue, I am closing this.

Hopefully Digitrax will provide a satisfactory resolution.

closed 2018-01-04 by pabender

KenC57, 2018-01-04:
Historically Digitrax updated prevented back loading (prior version). On
some there is an old version that did not have that test. UR92's had that to
get around a bad update. You had to load a very old version, then update to
the one before the problem to recover. I don't know if the same trick is
available for the BXP88. My understanding isn't clear if the DCS51 might
also be playing into this issue. So we have a number of variables to deal
with. I suggest asking on the user group to find other users of the BXP and
see what they have found and which versions and combinations they are
running. I don't know if the Digitrax list might be better for this or not.
-ken c

ClockLoop, 2018-01-04:
I didn't have the dcs51 connected for any of the final tests.  So its a non-factor.
The user group might help but i can't stand yahoo,  sign up for yet another email address..
I will probably do it and post all this crap there too eventually and try crap people suggest too, like you said, "So what happens when you have the two dz123s and one sdxn136ps?"
Thats a good next test step... and the yahoo group would be the proper place to continue this logic deducing for this non jmri issue.
sign up for yet another email, on yet another platform...  gah, what ever happened to forums.
Companies don't like them cause their customers can eaisly figure out all the problems and get the word out using the company its self..  they did away with that stuff a while back.

devel-bobm, 2018-01-04:
@ClockLoop,

You wrote: 
`The loco7 started transponding...`
`00:23:45.946: [D0 20 31 00 07 39] Transponder address 7 (short) present at LR50 () (BDL16x Board ID 4 RX4 zone (Unknown 1)).`

The "Transponding Location" message you reported (repeated above) appears to differ from the message format used by the BDL168.  With the BDL168, byte two (0x31 in the message above) would always have the least-significant bit cleared (for any BDL168 "transponding location" message I've seen).  Your BXP88 message shows the least-significant bit of byte 2 as a "1".  As such, any "interpretation" done by JMRI is suspicious, and possibly wrong.

Either Digitrax has changed the LocoNet messaging format with the BXP88, or Digitrax has a bug in the firmware.

If the LocoNet message format has changed, JMRI will need an update.  But we would need clear information on how it has changed, and how to differentiate between BDL168-format and BXP88-format.

(The fact that some BXP88 messages for loco 1 and loco 7 appear to be identical other than address information when the locos are in different transponding zones is a different problem that cannot be resolved by software.)

ClockLoop, 2018-01-04:
Here is the BXP88 I/O breakdown starting with BXP88 id0.
These results are real results copied from the loconet monitor in jmri, as I shorted each zone.
BXP1 all the way to BXP10  After that the sequence should be logically deduced.

BXP88 ID1
15:33:22.852: [B2 00 50 1D]  Sensor LS1 () is High.  (BDL16 # 1, DS1; DS54/DS64/SE8c # 1, AuxA/A1/DS01).
15:33:24.582: [B2 00 40 0D]  Sensor LS1 () is Low.  (BDL16 # 1, DS1; DS54/DS64/SE8c # 1, AuxA/A1/DS01).

15:30:18.398: [B2 00 70 3D]  Sensor LS2 () is High.  (BDL16 # 1, DS2; DS54/DS64/SE8c # 1, SwiA/S1/DS02).
15:30:22.488: [B2 00 60 2D]  Sensor LS2 () is Low.  (BDL16 # 1, DS2; DS54/DS64/SE8c # 1, SwiA/S1/DS02).

15:30:23.628: [B2 01 50 1C]  Sensor LS3 () is High.  (BDL16 # 1, DS3; DS54/DS64/SE8c # 1, AuxB/A2/DS03).
15:30:27.078: [B2 01 40 0C]  Sensor LS3 () is Low.  (BDL16 # 1, DS3; DS54/DS64/SE8c # 1, AuxB/A2/DS03).

15:30:27.778: [B2 01 70 3C]  Sensor LS4 () is High.  (BDL16 # 1, DS4; DS54/DS64/SE8c # 1, SwiB/S2/DS04).
15:30:31.228: [B2 01 60 2C]  Sensor LS4 () is Low.  (BDL16 # 1, DS4; DS54/DS64/SE8c # 1, SwiB/S2/DS04).

15:30:37.538: [B2 02 50 1F]  Sensor LS5 () is High.  (BDL16 # 1, DS5; DS54/DS64/SE8c # 1, AuxC/A3/DS05).
15:30:44.868: [B2 02 40 0F]  Sensor LS5 () is Low.  (BDL16 # 1, DS5; DS54/DS64/SE8c # 1, AuxC/A3/DS05).

15:31:13.848: [B2 02 70 3F]  Sensor LS6 () is High.  (BDL16 # 1, DS6; DS54/DS64/SE8c # 1, SwiC/S3/DS06).
15:31:16.008: [B2 02 60 2F]  Sensor LS6 () is Low.  (BDL16 # 1, DS6; DS54/DS64/SE8c # 1, SwiC/S3/DS06).

15:31:16.058: [B2 03 50 1E]  Sensor LS7 () is High.  (BDL16 # 1, DS7; DS54/DS64/SE8c # 1, AuxD/A4/DS07).
15:31:18.648: [B2 03 40 0E]  Sensor LS7 () is Low.  (BDL16 # 1, DS7; DS54/DS64/SE8c # 1, AuxD/A4/DS07).

15:31:18.918: [B2 03 70 3E]  Sensor LS8 () is High.  (BDL16 # 1, DS8; DS54/DS64/SE8c # 1, SwiD/S4/DS08).
15:31:21.508: [B2 03 60 2E]  Sensor LS8 () is Low.  (BDL16 # 1, DS8; DS54/DS64/SE8c # 1, SwiD/S4/DS08).

BXP88 ID2
15:34:54.772: [B2 04 50 19]  Sensor LS9 () is High.  (BDL16 # 1, DS9; DS54/DS64/SE8c # 2, AuxA/A1/DS01).
15:34:57.802: [B2 04 40 09]  Sensor LS9 () is Low.  (BDL16 # 1, DS9; DS54/DS64/SE8c # 2, AuxA/A1/DS01).

15:34:58.502: [B2 04 70 39]  Sensor LS10 () is High.  (BDL16 # 1, DS10; DS54/DS64/SE8c # 2, SwiA/S1/DS02).
15:35:01.092: [B2 04 60 29]  Sensor LS10 () is Low.  (BDL16 # 1, DS10; DS54/DS64/SE8c # 2, SwiA/S1/DS02).

15:35:02.012: [B2 05 50 18]  Sensor LS11 () is High.  (BDL16 # 1, DS11; DS54/DS64/SE8c # 2, AuxB/A2/DS03).
15:35:04.812: [B2 05 40 08]  Sensor LS11 () is Low.  (BDL16 # 1, DS11; DS54/DS64/SE8c # 2, AuxB/A2/DS03).

15:35:05.952: [B2 05 70 38]  Sensor LS12 () is High.  (BDL16 # 1, DS12; DS54/DS64/SE8c # 2, SwiB/S2/DS04).
15:35:08.542: [B2 05 60 28]  Sensor LS12 () is Low.  (BDL16 # 1, DS12; DS54/DS64/SE8c # 2, SwiB/S2/DS04).

15:35:11.402: [B2 06 50 1B]  Sensor LS13 () is High.  (BDL16 # 1, DS13; DS54/DS64/SE8c # 2, AuxC/A3/DS05).
15:35:14.202: [B2 06 40 0B]  Sensor LS13 () is Low.  (BDL16 # 1, DS13; DS54/DS64/SE8c # 2, AuxC/A3/DS05).

15:35:14.912: [B2 06 70 3B]  Sensor LS14 () is High.  (BDL16 # 1, DS14; DS54/DS64/SE8c # 2, SwiC/S3/DS06).
15:35:16.852: [B2 06 60 2B]  Sensor LS14 () is Low.  (BDL16 # 1, DS14; DS54/DS64/SE8c # 2, SwiC/S3/DS06).

15:35:17.332: [B2 07 50 1A]  Sensor LS15 () is High.  (BDL16 # 1, DS15; DS54/DS64/SE8c # 2, AuxD/A4/DS07).
15:35:19.712: [B2 07 40 0A]  Sensor LS15 () is Low.  (BDL16 # 1, DS15; DS54/DS64/SE8c # 2, AuxD/A4/DS07).

15:35:20.622: [B2 07 70 3A]  Sensor LS16 () is High.  (BDL16 # 1, DS16; DS54/DS64/SE8c # 2, SwiD/S4/DS08).
15:35:23.002: [B2 07 60 2A]  Sensor LS16 () is Low.  (BDL16 # 1, DS16; DS54/DS64/SE8c # 2, SwiD/S4/DS08).

BXP88 ID3
15:37:45.232: [B2 08 50 15]  Sensor LS17 () is High.  (BDL16 # 2, DS1; DS54/DS64/SE8c # 3, AuxA/A1/DS01).
15:37:47.832: [B2 08 40 05]  Sensor LS17 () is Low.  (BDL16 # 2, DS1; DS54/DS64/SE8c # 3, AuxA/A1/DS01).

15:37:49.392: [B2 08 70 35]  Sensor LS18 () is High.  (BDL16 # 2, DS2; DS54/DS64/SE8c # 3, SwiA/S1/DS02).
15:37:51.542: [B2 08 60 25]  Sensor LS18 () is Low.  (BDL16 # 2, DS2; DS54/DS64/SE8c # 3, SwiA/S1/DS02).

15:37:52.462: [B2 09 50 14]  Sensor LS19 () is High.  (BDL16 # 2, DS3; DS54/DS64/SE8c # 3, AuxB/A2/DS03).
15:37:54.192: [B2 09 40 04]  Sensor LS19 () is Low.  (BDL16 # 2, DS3; DS54/DS64/SE8c # 3, AuxB/A2/DS03).

15:37:55.102: [B2 09 70 34]  Sensor LS20 () is High.  (BDL16 # 2, DS4; DS54/DS64/SE8c # 3, SwiB/S2/DS04).
15:37:56.832: [B2 09 60 24]  Sensor LS20 () is Low.  (BDL16 # 2, DS4; DS54/DS64/SE8c # 3, SwiB/S2/DS04).

15:38:04.862: [B2 0A 50 17]  Sensor LS21 () is High.  (BDL16 # 2, DS5; DS54/DS64/SE8c # 3, AuxC/A3/DS05).
15:38:07.022: [B2 0A 40 07]  Sensor LS21 () is Low.  (BDL16 # 2, DS5; DS54/DS64/SE8c # 3, AuxC/A3/DS05).

[... 158 lines cut ...]

BXP88 ID10
15:54:36.634: [B2 24 50 39]  Sensor LS73 () is High.  (BDL16 # 5, DS9; DS54/DS64/SE8c # 10, AuxA/A1/DS01).
15:54:40.084: [B2 24 40 29]  Sensor LS73 () is Low.  (BDL16 # 5, DS9; DS54/DS64/SE8c # 10, AuxA/A1/DS01).

15:54:47.904: [B2 24 70 19]  Sensor LS74 () is High.  (BDL16 # 5, DS10; DS54/DS64/SE8c # 10, SwiA/S1/DS02).
15:54:50.484: [B2 24 60 09]  Sensor LS74 () is Low.  (BDL16 # 5, DS10; DS54/DS64/SE8c # 10, SwiA/S1/DS02).

15:54:50.754: [B2 25 50 38]  Sensor LS75 () is High.  (BDL16 # 5, DS11; DS54/DS64/SE8c # 10, AuxB/A2/DS03).
15:54:53.134: [B2 25 40 28]  Sensor LS75 () is Low.  (BDL16 # 5, DS11; DS54/DS64/SE8c # 10, AuxB/A2/DS03).

[... cut ...]
15:55:04.514: [B2 26 60 0B]  Sensor LS78 () is Low.  (BDL16 # 5, DS14; DS54/DS64/SE8c # 10, SwiC/S3/DS06).

15:55:05.214: [B2 27 50 3A]  Sensor LS79 () is High.  (BDL16 # 5, DS15; DS54/DS64/SE8c # 10, AuxD/A4/DS07).
15:55:07.154: [B2 27 40 2A]  Sensor LS79 () is Low.  (BDL16 # 5, DS15; DS54/DS64/SE8c # 10, AuxD/A4/DS07).

15:55:07.854: [B2 27 70 1A]  Sensor LS80 () is High.  (BDL16 # 5, DS16; DS54/DS64/SE8c # 10, SwiD/S4/DS08).
15:55:10.224: [B2 27 60 0A]  Sensor LS80 () is Low.  (BDL16 # 5, DS16; DS54/DS64/SE8c # 10, SwiD/S4/DS08).

devel-bobm, 2018-01-04:
@ClockLoop,

The "detection" messages you report look "normal" to me - no surprises.  

Transponding "presence/absence" messages are entirely different.  The BXP88 ones you reported previously, however, look problematic, as they do not follow the pattern I've seen with the BDL168's "transponding presence/absence" messages.

ClockLoop, 2018-01-04:
Oh, I can make one of those arrays also.

devel-bobm, 2018-01-05:
@ClockLoop ,

The problem I saw from a previous BXP88 log is that BXP transponding presence/absence reports seem to encode the "zone" differently than for BDL168.  As such, varying the Board ID doesn't provide much help.  

Instead, it would be helpful to run a single engine thru all 8 zones for a given Board ID.

ClockLoop, 2018-01-05:
I can get a complete table, ill post here when I can.

This video shows locomotive 3780 and locomotive 307 going around two different tracks. The transponding addresses between them swap. This is not crosstalk due to the fact that you can remove either loco from the tracks without shutting down the loco in the throttle, and **the transponding will continue to swap and show the bad address of the loco that isn't even running anymore**, its in your hand. Nuff said, BUG.
http://www.youtube.com/watch?v=42gyUZdyp-Y

ClockLoop, 2018-01-05:
Ok, heres 3 bxp88s that loco 307 transpond on each of the bxp88s starting at bxp88 with ID1 in sequence all the way to the end of bxp88 with ID3.
I can see what you mean the naming is odd.

```

BXP88 ID1
23:29:22.409: [D0 20 00 02 33 3E]  Transponder address 307 present at LR1 () (BDL16x Board ID 1 RX4 zone A).
23:29:23.469: [D0 00 00 02 33 1E]  Transponder address 307 absent at LR1 () (BDL16x Board ID 1 RX4 zone A).

23:29:37.519: [D0 20 01 02 33 3F]  Transponder address 307 present at LR2 () (BDL16x Board ID 1 RX4 zone (Unknown 1)).
23:29:41.509: [D0 00 01 02 33 1F]  Transponder address 307 absent at LR2 () (BDL16x Board ID 1 RX4 zone (Unknown 1)).

23:29:59.619: [D0 20 02 02 33 3C]  Transponder address 307 present at LR3 () (BDL16x Board ID 1 RX4 zone B).
23:30:01.869: [D0 00 02 02 33 1C]  Transponder address 307 absent at LR3 () (BDL16x Board ID 1 RX4 zone B).

23:30:22.779: [D0 20 03 02 33 3D]  Transponder address 307 present at LR4 () (BDL16x Board ID 1 RX4 zone (Unknown 3)).
23:30:24.279: [D0 00 03 02 33 1D]  Transponder address 307 absent at LR4 () (BDL16x Board ID 1 RX4 zone (Unknown 3)).

23:30:36.879: [D0 20 04 02 33 3A]  Transponder address 307 present at LR5 () (BDL16x Board ID 1 RX4 zone C).
23:30:41.189: [D0 00 04 02 33 1A]  Transponder address 307 absent at LR5 () (BDL16x Board ID 1 RX4 zone C).

23:30:47.779: [D0 20 05 02 33 3B]  Transponder address 307 present at LR6 () (BDL16x Board ID 1 RX4 zone (Unknown 5)).
23:30:49.529: [D0 00 05 02 33 1B]  Transponder address 307 absent at LR6 () (BDL16x Board ID 1 RX4 zone (Unknown 5)).

23:30:58.899: [D0 20 06 02 33 38]  Transponder address 307 present at LR7 () (BDL16x Board ID 1 RX4 zone D).
23:31:00.289: [D0 00 06 02 33 18]  Transponder address 307 absent at LR7 () (BDL16x Board ID 1 RX4 zone D).

23:31:12.839: [D0 20 07 02 33 39]  Transponder address 307 present at LR8 () (BDL16x Board ID 1 RX4 zone (Unknown 7)).
23:31:14.549: [D0 00 07 02 33 19]  Transponder address 307 absent at LR8 () (BDL16x Board ID 1 RX4 zone (Unknown 7)).

BXP88 ID2
23:31:40.189: [D0 20 08 02 33 36]  Transponder address 307 present at LR9 () (BDL16x Board ID 1 RX4 zone E).
23:31:43.859: [D0 00 08 02 33 16]  Transponder address 307 absent at LR9 () (BDL16x Board ID 1 RX4 zone E).

23:31:56.079: [D0 20 09 02 33 37]  Transponder address 307 present at LR10 () (BDL16x Board ID 1 RX4 zone (Unknown 9)).
23:31:58.929: [D0 00 09 02 33 17]  Transponder address 307 absent at LR10 () (BDL16x Board ID 1 RX4 zone (Unknown 9)).

23:32:11.929: [D0 20 0A 02 33 34]  Transponder address 307 present at LR11 () (BDL16x Board ID 1 RX4 zone F).
23:32:13.009: [D0 00 0A 02 33 14]  Transponder address 307 absent at LR11 () (BDL16x Board ID 1 RX4 zone F).

23:32:21.579: [D0 20 0B 02 33 35]  Transponder address 307 present at LR12 () (BDL16x Board ID 1 RX4 zone (Unknown 11)).
23:32:29.589: [D0 00 0B 02 33 15]  Transponder address 307 absent at LR12 () (BDL16x Board ID 1 RX4 zone (Unknown 11)).

23:32:36.680: [D0 20 0C 02 33 32]  Transponder address 307 present at LR13 () (BDL16x Board ID 1 RX4 zone G).
23:32:41.210: [D0 00 0C 02 33 12]  Transponder address 307 absent at LR13 () (BDL16x Board ID 1 RX4 zone G).

23:32:47.620: [D0 20 0D 02 33 33]  Transponder address 307 present at LR14 () (BDL16x Board ID 1 RX4 zone (Unknown 13)).
23:32:52.050: [D0 00 0D 02 33 13]  Transponder address 307 absent at LR14 () (BDL16x Board ID 1 RX4 zone (Unknown 13)).

23:33:04.670: [D0 20 0E 02 33 30]  Transponder address 307 present at LR15 () (BDL16x Board ID 1 RX4 zone H).
23:33:06.410: [D0 00 0E 02 33 10]  Transponder address 307 absent at LR15 () (BDL16x Board ID 1 RX4 zone H).

23:33:16.910: [D0 20 0F 02 33 31]  Transponder address 307 present at LR16 () (BDL16x Board ID 1 RX4 zone (Unknown 15)).
23:33:19.060: [D0 00 0F 02 33 11]  Transponder address 307 absent at LR16 () (BDL16x Board ID 1 RX4 zone (Unknown 15)).

BXP88 ID3
23:33:35.940: [D0 20 10 02 33 2E]  Transponder address 307 present at LR17 () (BDL16x Board ID 2 RX4 zone A).
23:33:37.250: [D0 00 10 02 33 0E]  Transponder address 307 absent at LR17 () (BDL16x Board ID 2 RX4 zone A).

23:33:54.200: [D0 20 11 02 33 2F]  Transponder address 307 present at LR18 () (BDL16x Board ID 2 RX4 zone (Unknown 1)).
23:34:03.160: [D0 00 11 02 33 0F]  Transponder address 307 absent at LR18 () (BDL16x Board ID 2 RX4 zone (Unknown 1)).

23:34:26.250: [D0 20 12 02 33 2C]  Transponder address 307 present at LR19 () (BDL16x Board ID 2 RX4 zone B).
23:34:29.090: [D0 00 12 02 33 0C]  Transponder address 307 absent at LR19 () (BDL16x Board ID 2 RX4 zone B).

23:34:44.030: [D0 20 13 02 33 2D]  Transponder address 307 present at LR20 () (BDL16x Board ID 2 RX4 zone (Unknown 3)).
23:34:58.010: [D0 00 13 02 33 0D]  Transponder address 307 absent at LR20 () (BDL16x Board ID 2 RX4 zone (Unknown 3)).

23:35:07.890: [D0 20 14 02 33 2A]  Transponder address 307 present at LR21 () (BDL16x Board ID 2 RX4 zone C).
23:35:15.250: [D0 00 14 02 33 0A]  Transponder address 307 absent at LR21 () (BDL16x Board ID 2 RX4 zone C).

23:35:32.000: [D0 20 15 02 33 2B]  Transponder address 307 present at LR22 () (BDL16x Board ID 2 RX4 zone (Unknown 5)).
23:35:35.040: [D0 00 15 02 33 0B]  Transponder address 307 absent at LR22 () (BDL16x Board ID 2 RX4 zone (Unknown 5)).

23:35:46.470: [D0 20 16 02 33 28]  Transponder address 307 present at LR23 () (BDL16x Board ID 2 RX4 zone D).
23:35:52.250: [D0 00 16 02 33 08]  Transponder address 307 absent at LR23 () (BDL16x Board ID 2 RX4 zone D).

23:35:58.960: [D0 20 17 02 33 29]  Transponder address 307 present at LR24 () (BDL16x Board ID 2 RX4 zone (Unknown 7)).
23:36:04.500: [D0 00 17 02 33 09]  Transponder address 307 absent at LR24 () (BDL16x Board ID 2 RX4 zone (Unknown 7)).

```

ClockLoop, 2018-01-05:
SOLVED
The bxp88s must be on their own loconet cable connected to the command station and nothing else.
[image]

devel-bobm, 2018-01-05:
I think that you may be at the crux of the problem, because LocoNet is NOT supposed to care where/how your LocoNet cables are attached, so long as the LocoNet cables are all assembled "correctly" and so long as all LocoNet cables have proper connectivity between the plugs.  You should report your findings to Digitrax; I suspect that they will be curious about your findings.

The fiirst thing they are likely to ask is about quality of LocoNet cabling.  Are your LocoNet cables "assembled correctly" and have good continuity?  By "assembled correctly", I mean "is the blue wire on the same edge of the two connectors when the two connectors are held adjacent to each other and in the same orientation, with the cable next to each other?".

Historically, BDL16x transponding has been sensitive to "LocoNet cable reverses" (i.e. a jack crimped on opposite of the way it ought to be crimped on), but the problem seen with cable reverses was simple non-operation of transponding.  The little bits I know of how transponding works leads me to believe that simple "cable reversal" problems would not cause mis-identification of locos, regardless of which hardware was in-use.

bobjacobsen, 2018-01-05:
In addition to @devel-bobm's points, the LocoNet cable carries a ground wire that can inject _significant_ noise if there are power-supply issues on the attached board(s). Any ground loops, noisy supplies, etc can pull the ground-RailSync voltage enough to stop transponding detection and even BDL168 occupancy detection. You've got a fix, might be time to declare victory and move on, but if you want to investigate a bit more try attaching boards until you find the one that's a problem, then check it's power supply and grounding (including, depending on what kind of board it is, it's external connection power and grounding).

ClockLoop, 2018-01-07:
Unfortunately the problem isn't fixed, in the last video, you can see the swap happen once.

I took it all apart, placed my dcs240, ps2012, bxp88 and dt500 all on a piece of wood, and wired it all up to match the bxp88 manual page 7.  Still problematic.   I have 3 different working bxp88s and putting them all under individual test they operate identically.

I inspected the bxp88s pcb and found unsoldered resistors, and one of the bxp88 a smt capacitor fell out of the case.  My 4th bxp88 was DOA (worked for a month)
I have never gotten them to transpond properly.

Without any info how can i figure it out, i can't even tell what these bxp88 messages are saying, for all i know they could be valid data that is garbled, or non garbled valid data.

Im guessing its valid because lots of other data gets through fine.

**IT would just be nice to know if its legit cause that would tell me what might be the issue.
The personal copy of loconet is dated 1997 and has not info about these codes.
I think its safe to say that theres a bit of loconet info neglect going on here.**
```
11:35:10.936: [D0 40 6F 06 17 11]  Unable to parse LocoNet message.
contents: D0 40 6F 06 17 11
11:35:18.286: [D0 40 2F 06 17 51]  Unable to parse LocoNet message.
contents: D0 40 2F 06 17 51
11:36:04.187: [D0 40 5F 06 17 21]  Unable to parse LocoNet message.
contents: D0 40 5F 06 17 21
11:37:16.407: [D0 40 0E 06 17 70]  Unable to parse LocoNet message.
contents: D0 40 0E 06 17 70
11:38:07.727: [D0 40 34 06 17 4A]  Unable to parse LocoNet message.
contents: D0 40 34 06 17 4A
11:38:45.147: [D0 40 38 06 17 46]  Unable to parse LocoNet message.
contents: D0 40 38 06 17 46
11:38:51.977: [D0 40 6E 06 17 10]  Unable to parse LocoNet message.
contents: D0 40 6E 06 17 10
11:38:52.397: [D0 40 42 06 17 3C]  Unable to parse LocoNet message.
contents: D0 40 42 06 17 3C
11:38:56.767: [D0 40 32 06 17 4C]  Unable to parse LocoNet message.
contents: D0 40 32 06 17 4C
11:39:23.077: [D0 40 18 06 17 66]  Unable to parse LocoNet message.
contents: D0 40 18 06 17 66
11:39:48.207: [D0 40 79 06 17 07]  Unable to parse LocoNet message.
contents: D0 40 79 06 17 07
11:40:01.027: [D0 40 72 06 17 0C]  Unable to parse LocoNet message.
contents: D0 40 72 06 17 0C
11:40:19.977: [D0 40 56 06 17 28]  Unable to parse LocoNet message.
contents: D0 40 56 06 17 28
11:41:50.837: [D0 40 27 06 17 59]  Unable to parse LocoNet message.
contents: D0 40 27 06 17 59
11:41:58.277: [D0 40 3C 06 17 42]  Unable to parse LocoNet message.
contents: D0 40 3C 06 17 42
11:42:02.867: [D0 40 30 06 17 4E]  Unable to parse LocoNet message.
contents: D0 40 30 06 17 4E
11:42:13.457: [D0 40 33 06 17 4D]  Unable to parse LocoNet message.
contents: D0 40 33 06 17 4D
11:42:18.467: [D0 40 13 06 17 6D]  Unable to parse LocoNet message.
contents: D0 40 13 06 17 6D
11:42:40.217: [D0 40 1E 06 17 60]  Unable to parse LocoNet message.
contents: D0 40 1E 06 17 60
11:43:12.627: [D0 40 10 06 17 6E]  Unable to parse LocoNet message.
contents: D0 40 10 06 17 6E
11:43:37.807: [D0 40 6D 06 17 13]  Unable to parse LocoNet message.
contents: D0 40 6D 06 17 13
11:44:16.507: [D0 40 40 06 17 3E]  Unable to parse LocoNet message.
contents: D0 40 40 06 17 3E
11:44:22.497: [D0 40 68 06 17 16]  Unable to parse LocoNet message.
contents: D0 40 68 06 17 16
11:44:30.557: [D0 40 52 06 17 2C]  Unable to parse LocoNet message.
contents: D0 40 52 06 17 2C
11:44:35.767: [D0 40 10 06 17 6E]  Unable to parse LocoNet message.
contents: D0 40 10 06 17 6E
11:44:41.527: [D0 40 16 06 17 68]  Unable to parse LocoNet message.
contents: D0 40 16 06 17 68
11:44:46.237: [D0 40 22 06 17 5C]  Unable to parse LocoNet message.
contents: D0 40 22 06 17 5C
11:44:54.147: [D0 40 42 06 17 3C]  Unable to parse LocoNet message.
[... 150 lines cut ...]
contents: D0 40 6B 06 17 15
12:08:30.870: [D0 40 10 06 17 6E]  Unable to parse LocoNet message.
contents: D0 40 10 06 17 6E
12:08:35.689: [D0 40 58 06 17 26]  Unable to parse LocoNet message.
contents: D0 40 58 06 17 26
12:08:35.949: [D0 40 3F 06 17 41]  Unable to parse LocoNet message.
contents: D0 40 3F 06 17 41
12:08:52.230: [D0 40 3A 06 17 44]  Unable to parse LocoNet message.
contents: D0 40 3A 06 17 44
12:09:01.240: [D0 40 10 06 17 6E]  Unable to parse LocoNet message.
contents: D0 40 10 06 17 6E
12:09:33.010: [D0 40 59 06 17 27]  Unable to parse LocoNet message.
contents: D0 40 59 06 17 27
12:09:37.420: [D0 40 58 06 17 26]  Unable to parse LocoNet message.
contents: D0 40 58 06 17 26
12:09:38.290: [D0 40 49 06 17 37]  Unable to parse LocoNet message.
contents: D0 40 49 06 17 37
12:10:47.330: [D0 40 7A 06 17 04]  Unable to parse LocoNet message.
contents: D0 40 7A 06 17 04
12:11:56.780: [D0 40 18 06 17 66]  Unable to parse LocoNet message.
contents: D0 40 18 06 17 66
12:12:38.570: [D0 40 5E 06 17 20]  Unable to parse LocoNet message.
contents: D0 40 5E 06 17 20

```

devel-bobm, 2018-01-07:
Complaining about Digitrax here is pointless.  Asking here for Digitrax proprietary information is pointless.  

Remember - this is a Github repository for JMRI.  JMRI is not a product of Digitrax.  **JMRI'S DEVELOPERS ARE NOT DIGITRAX REPRESENTATIVES OR DIGITRAX EMPLOYEES.** We do not have the sorts of "Digitrax proprietary information" you are looking for.

Digitrax has NO requirement to publicly document any of its "intellectual property".  Digitrax, like most electronics companies, holds their "intellectual property" close.  And it is the environment in which we must live.

ClockLoop, 2018-01-07:
Dont worry, I am aware of all that.   Posting info here so others may benefit is what this is, but if you wanna call it complainin.. whatever...    Nah, just tellin it like it is.    

Asking here for Digitrax proprietary info, is not only pointless, its a big joke, thats like asking to live eternally.   Don't believe everything, and think everything you read on the internet is truth or even serious.   Sarcasm is clearly impossible to convey on the internet, its always miconstrewed as attacking, complaining, or something that implies a negative emotion...  People default to negative interpretation when they don't have facial, body and emotional indications of intent.

Again, try not to misinterpret my statements as anything targeted at jmri or you guys, ... take the internets a bit less personally, and it will work out.

The only reason any of this is here is because it was revealed using jmri, and others will probably find such things using the same methods.

Digitrax has NO requirement to publicly document........ laws..  can ... be ... changed...

devel-bobm, 2018-01-07:
Please take your rant elsewhere.  It does not belong in the Github repository for JMRI software.

ClockLoop, 2018-01-08:
I tried to do just that, and delete this and bring it elsewhere, but i see no way to delete this thread, i can only edit my comments to null.