JMRI/JMRI #4676: Unable to parse LocoNet message.  (related to system wide track short circuit)

ClockLoop, 2018-01-02:
I gathered a bunch of logs and found some data that might be useful.
Many times data appears on loconet that jmri can't parse which is from a track short.

Is this data that tells loconet about the shorts?

Some of these might be redundant.

In all of them, my steam engine either caused a short and stopped, and i had to fix it, or it temporarily shorted and recovered right away.

16:33:34.607: [D0 62 05 20 24 4C]  Unable to parse LocoNet message.
contents: D0 62 05 20 24 4C
16:33:34.697: [D0 62 05 20 20 48]  Unable to parse LocoNet message.
contents: D0 62 05 20 20 48

16:39:58.308: [D0 62 07 20 21 4B]  Unable to parse LocoNet message.
contents: D0 62 07 20 21 4B
16:39:58.528: [D0 62 07 20 20 4A]  Unable to parse LocoNet message.
contents: D0 62 07 20 20 4A

18:22:45.081: [D0 62 06 28 20 43]  Unable to parse LocoNet message.
contents: D0 62 06 28 20 43

18:22:45.171: [D0 62 06 20 20 4B]  Unable to parse LocoNet message.
contents: D0 62 06 20 20 4B
18:22:46.701: [D0 62 06 28 20 43]  Unable to parse LocoNet message.
contents: D0 62 06 28 20 43
18:22:47.761: [D0 62 06 20 20 4B]  Unable to parse LocoNet message.
contents: D0 62 06 20 20 4B

18:22:57.161: [D0 62 06 28 20 43]  Unable to parse LocoNet message.
contents: D0 62 06 28 20 43
18:22:57.251: [D0 62 06 20 20 4B]  Unable to parse LocoNet message.
contents: D0 62 06 20 20 4B

18:22:58.751: [D0 62 06 28 20 43]  Unable to parse LocoNet message.
contents: D0 62 06 28 20 43
18:23:00.121: [D0 62 06 20 20 4B]  Unable to parse LocoNet message.
contents: D0 62 06 20 20 4B

18:23:00.211: [D0 62 06 28 20 43]  Unable to parse LocoNet message.
contents: D0 62 06 28 20 43
18:23:00.301: [D0 62 06 20 20 4B]  Unable to parse LocoNet message.
contents: D0 62 06 20 20 4B

18:23:00.431: [D0 62 06 48 40 43]  Unable to parse LocoNet message.
contents: D0 62 06 48 40 43
18:23:02.351: [D0 62 06 40 40 4B]  Unable to parse LocoNet message.
contents: D0 62 06 40 40 4B

18:26:39.423: [D0 62 05 24 27 4B]  Unable to parse LocoNet message.
contents: D0 62 05 24 27 4B

05:17:58.287: [D0 62 05 20 20 48]  Unable to parse LocoNet message.
contents: D0 62 05 20 20 48
05:17:58.297: [D0 62 07 20 20 4A]  Unable to parse LocoNet message.
contents: D0 62 07 20 20 4A

05:18:05.447: [D0 62 05 20 21 49]  Unable to parse LocoNet message.
contents: D0 62 05 20 21 49
05:18:05.537: [D0 62 05 20 20 48]  Unable to parse LocoNet message.
contents: D0 62 05 20 20 48
05:18:05.687: [D0 62 05 20 21 49]  Unable to parse LocoNet message.
contents: D0 62 05 20 21 49
05:18:05.777: [D0 62 05 20 20 48]  Unable to parse LocoNet message.
contents: D0 62 05 20 20 48

05:18:26.917: [D0 62 07 20 22 48]  Unable to parse LocoNet message.
contents: D0 62 07 20 22 48
05:18:26.927: [D0 62 07 20 2A 40]  Unable to parse LocoNet message.
contents: D0 62 07 20 2A 40
05:18:27.007: [D0 62 07 20 28 42]  Unable to parse LocoNet message.
contents: D0 62 07 20 28 42
05:18:27.017: [D0 62 07 20 20 4A]  Unable to parse LocoNet message.
contents: D0 62 07 20 20 4A

05:18:28.617: [D0 62 07 20 28 42]  Unable to parse LocoNet message.
contents: D0 62 07 20 28 42
05:18:28.627: [D0 62 07 20 2A 40]  Unable to parse LocoNet message.
contents: D0 62 07 20 2A 40
05:18:28.707: [D0 62 07 20 22 48]  Unable to parse LocoNet message.
contents: D0 62 07 20 22 48
05:18:28.717: [D0 62 07 20 20 4A]  Unable to parse LocoNet message.
contents: D0 62 07 20 20 4A
05:18:28.747: [D0 62 07 20 2A 40]  Unable to parse LocoNet message.
contents: D0 62 07 20 2A 40
05:18:28.967: [D0 62 07 20 28 42]  Unable to parse LocoNet message.
contents: D0 62 07 20 28 42
05:18:28.977: [D0 62 07 20 20 4A]  Unable to parse LocoNet message.
contents: D0 62 07 20 20 4A
05:18:29.097: [D0 62 07 20 22 48]  Unable to parse LocoNet message.
contents: D0 62 07 20 22 48
05:18:29.107: [D0 62 07 20 2A 40]  Unable to parse LocoNet message.
contents: D0 62 07 20 2A 40
05:18:29.317: [D0 62 07 20 28 42]  Unable to parse LocoNet message.
contents: D0 62 07 20 28 42
05:18:29.327: [D0 62 07 20 20 4A]  Unable to parse LocoNet message.
contents: D0 62 07 20 20 4A
05:18:29.337: [D0 62 07 20 2A 40]  Unable to parse LocoNet message.
contents: D0 62 07 20 2A 40
05:18:29.497: [D0 62 07 20 28 42]  Unable to parse LocoNet message.
contents: D0 62 07 20 28 42
05:18:29.507: [D0 62 07 20 20 4A]  Unable to parse LocoNet message.
contents: D0 62 07 20 20 4A
05:18:29.517: [D0 62 07 20 2A 40]  Unable to parse LocoNet message.
contents: D0 62 07 20 2A 40
05:18:29.607: [D0 62 07 20 28 42]  Unable to parse LocoNet message.
contents: D0 62 07 20 28 42
05:18:29.627: [D0 62 07 20 2A 40]  Unable to parse LocoNet message.
contents: D0 62 07 20 2A 40
05:18:29.677: [D0 62 05 20 28 40]  Unable to parse LocoNet message.
contents: D0 62 05 20 28 40

devel-bobm, 2018-01-02:
Bug introduced in #3206 or #3207 - improper differentiation between autoreversing and short-circuit messages.

ClockLoop, 2018-01-05:
Ok, just so you know these messages were generated on a dcs240 that dosen't have any autoreversing, on or any extra autoreversing.  

I have bxp88s doing all track service, so these messages were all track short messages either from the dcs240 or the bxp88.

closed 2021-04-14 by stale[bot]