JMRI/JMRI #4678: Adding an audio buffer on a raspberrypi makes empty buffer.

ClockLoop, 2018-01-02:
Adding an audio buffer on a raspberrypi results in a "Empty Buffer" on the raspberrypi.

I just can't have my automated sounds and jmri!!!! (the decoder function for transits is also broken!)

ClockLoop, 2018-01-04:
I would like to add that playing an audio wav file using logix works to play audio on the pi, but this only lets you play it once for the action.   The audio buffer allows so much more.  So the sound system does work, just the buffer is broken.

klk32003, 2018-01-05:
Does this script work for you?
```
import jmri
buffer = audio.provideAudio("IAB1")
buffer.setURL("program:resources/sounds/Signal-lever.wav")
buffer.setUserName("Buffer Signal-lever")
source = audio.provideAudio("IAS1")
source.setUserName("Source Signal-Lever")
source.setAssignedBuffer("IAB1")
source.play()
```

In Logix, you then use ```Add Action > Audio > Action Type > Control Audio object``` and choose the appropriate action (play, loop, stop, etc.)

closed 2021-04-14 by stale[bot]