JMRI/JMRI #4684: Dispatcher InterruptedException Error

dsand47, 2018-01-02:
The **Dispatcher Auto Trains** thread on Yahoo discusses this problem.

When a train is added, direction sensors are inserted into the signal head SSL.  This requires that the current Siglet thread be removed, the SSL updated and the Siglet thread re-created.

The remove process is triggered by the thread stop() method issuing an **interrupt()**.  This triggers the thread **catch (InterruptedException e)** to cleanup and die.

If the thread is currently running setOutput() using ThreadingUtil.runOnLayout()  [[ actually runOnGui ]] when the stop process occurs, runOnGui catches the interrupt, not the thread.  This hangs JMRI and it must be killed at the OS level.

This is ultimately a timing problem.  The same Dispatcher setup may work on one machine, but not another.   Or be dependent on the current state of single machine.

I see two solutions:  Remove the runOnLayout or have runOnGui propagate the interrupt.  Given my Java rookie understanding, there may be others.

Dave Sand

bobjacobsen, 2018-01-03:
I can't recreate this, so can't test it directly, but I think I have a fix.  What's the best way to get it checked?

dsand47, 2018-01-03:
If you attach a diff file I can apply it and test it.

bobjacobsen, 2018-01-03:
It's the three files at the tip of this branch: https://github.com/bobjacobsen/JMRI/tree/thread-defer-interrupt  so you can just pick that up directly, or use the attached diff from HEAD of master.
[patch.txt](https://github.com/JMRI/JMRI/files/1599418/patch.txt)

dsand47, 2018-01-03:
I have applied the patch to a new branch from my master that was updated earlier today.

I temporarily disabled the interrupt propagation in runOnGui to make sure the failure still occurrs, which it did.  Enabling propagation work properly.

I ran the jmri.jmrit.blockboss.PackageTest and the jmri.jmrit.PackageTest.  Both were successful.

The attached file has **_before_** and **_after_** trace snippets.

Dave Sand

[test results.txt](https://github.com/JMRI/JMRI/files/1599483/test.results.txt)

bobjacobsen, 2018-01-04:
Merged #4686 which should be the fix for this.

closed 2018-01-04 by bobjacobsen

dsand47, 2018-01-04:
Final testing using Jenkins build 2162.  Works as expected.  I did have 15 propagated interrupts (out of 305) but the simulation train runs worked.

PR #4686 (2018-01-04): Fix for Dispatcher intermittent error
Hopefully, this fixes #4684, but in any case it fixes a race condition when interrupting a process that has used ThreadingUtil.runOnGui or runOnLayout