JMRI/JMRI #4696: Pressing the "LOCO reset" on the front of a DCS240 results in unable to parse message.

ClockLoop, 2018-01-05:
Pressing the "LOCO reset" button on the front of a DCS240 results in this message.

15:00:06.271: [8A 75]  Unable to parse LocoNet message.
contents: 8A 75

devel-bobm, 2018-01-05:
@ClockLoop,
You wrote
'15:00:06.271: [8A 75] Unable to parse LocoNet message.'
'contents: 8A 75'

What does that message do/mean?  Without that knowledge, we cannot display a sensible interpretation.

ClockLoop, 2018-01-05:
The manual of the DCS240 states:
```
4.14  LOCO RESET Button
The LOCO RESET button allows you to clear all decoder info and consists 
from your DCS240.  Pressing this button preforms the same function as closing 
OpSw 36.  This button is useful in clearing all locos from the system after a 
large operating session or if you are getting a slot full error.

OpSw 36*  c =  Clears all mobile decoder info & consists
```

So 8A 75 is some kinda acknowledge?  Or

devel-bobm, 2018-01-05:
WIthout an accurate description, it is best to leave the message unparsed.

ClockLoop, 2018-01-05:
Heh, im kinda confused, if one cannot find out what these loconet messages mean, then whats the use to making all this hardware if no software can use it?
So if its not info widely available, what software "has the inside digitrax scoop" on what the heck all this is, so we can use it?
Or am i supposed to buy hardware from digitrax and never connect it to anything?

Why use software with your pc, its bios runs just fine all by its self, an OS improves a pc almost none at all, so, lets just do away with it.

bobjacobsen, 2018-01-05:
A few responses to all that:

- Having the results of pushing a button appear as interpreted text in a monitor isn't really "using" anything. You know you pushed the button, you get to see the message.  Yes, it's nice to have the monitor display things instead of having to look them up in Digitrax's documentation, but that's all it is.

- If you want to make this decoding better, convince Digitrax to provide more documentation. People around here tend to be enthused about implementing that kind of thing correctly.

- If you can't get Digitrax to do that, and you want to work it out yourself and code it in JMRI, go ahead.  The starting point is [here](http://jmri.org/help/en/html/doc/Technical/gitdeveloper.shtml).  Lots of people have put effort into making possible for you do do that. The piece of code you want to change is probably [here](https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/loconet/locomon/Llnmon.java) with associated test suite [here](https://github.com/JMRI/JMRI/blob/master/java/test/jmri/jmrix/loconet/locomon/LlnmonTest.java).

But here's by far the most important part of this reply: if you want to complain, take it elsewhere. You don't get to decide what people long-term contributors should be doing; you don't get to cast aspersions on what they choose to do or not do.  Don't do it again.

ClockLoop, 2018-01-05:
I was actually not complaning at all to anyone here, please point me to where?
If anything it was to the makers of the hardware for not releasing info like you said, but the nets are full of misinterpretation, and for some reason everyone runs as far as they can with the first slight sign of it..  why not just ask if that was my intent. 

I assure you sir, it was not.

Clearly I have over stayed my welcome here, at the jmri issues..  I bow out sir.

closed 2020-07-25 by bobjacobsen