JMRI/JMRI #4697: Setting switches with a dt500 throttle results in unable to parse message.

ClockLoop, 2018-01-05:
Setting switches with a dt500 throttle results in unable to parse messages.
```
15:26:19.977: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:42.517: [BC 04 00 47]  Request status of switch LT5 (Cross5).
15:26:42.517: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:45.337: [BC 05 30 76]  Unable to parse LocoNet message.
contents: BC 05 30 76
15:26:45.337: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:46.997: [BC 05 00 46]  Request status of switch LT6 (Cross6).
15:26:46.997: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:48.797: [BC 06 30 75]  Unable to parse LocoNet message.
contents: BC 06 30 75
15:26:48.797: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:49.207: [BC 06 00 45]  Request status of switch LT7 (Cross7).
15:26:49.207: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:49.517: [BC 07 30 74]  Unable to parse LocoNet message.
contents: BC 07 30 74
15:26:49.517: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:49.807: [BC 07 00 44]  Request status of switch LT8 (Cross8).
15:26:49.807: [B4 3C 34 43]  LONG_ACK: Command station response to switch state request 0x34 (Closed).
15:26:50.077: [BC 08 30 7B]  Unable to parse LocoNet message.
```

Do you guys want to know about these?
I tried to search to see if anyone has already reported them.

devel-bobm, 2018-01-05:
A fix is identified and undergoing the full suite of integration/reversion tests.  I hope that this fix will get into the next test release.

devel-bobm, 2020-08-06:
Closing this old issue, which was closed by P/R #4986 .

closed 2020-08-06 by devel-bobm