JMRI/JMRI #4702: Labels in pane definition are not overridden

AlanUS, 2018-01-05:
When a label is defined in a pane for a given CV, this label is NOT overridden by the label given on the CV declaration for this CV.
This is not consistent with the behavior of tooltips, where the tooltip defined at CV declaration overrides the tooltip defined in pane definition.
One option is to clean all panes containing label definitions, the other is to implement the same mechanism as for tooltips (preferred).
What do you think and where should I look in the code for this?

bobjacobsen, 2018-01-05:
I think this is a bug, but I'm not sure when it would have been introduced. The original logic was based on the idea that labels in the variable definition were optional.  Then:
- define a label in the pane, and that's the generic label
- only if you have to, because something really needs to be said, do you define "label" in the variable definition; if you do, it overrides the generic

What may have happened is that people were putting "label" on lots of things they were specifically writing, not knowing about the idea of the general names on the pane definitions (or not wanting to spend to time for it).

The "nameFmt" attribute might also be part of this.

This is probably being handled in java/src/jmri/jmrit/symbolicprog//tabbedframe/PaneProgPane.java

dheap, 2018-01-05:
> @AlanUS When a label is defined in a pane for a given CV, this label is NOT overridden by the label given on the CV declaration for this CV.
> This is not consistent with the behavior of tooltips, where the tooltip defined at CV declaration overrides the tooltip defined in pane definition.
> One option is to clean all panes containing label definitions, the other is to implement the same mechanism as for tooltips (preferred).

**Do not do do either of these.** A  lot of panes rely on this behaviour for correct operation. 

For example, every pane that displays an hslider with an alternate text entry field beside it will be broken. A blank label in the pane is used to suppress the otherwise-duplicate label above the text field.

> What do you think and where should I look in the code for this?

If you were to bring the tooltip behaviour into line with the label behaviour, there may the places where this will break existing definitions that rely on this behaviour. I'd be reluctant to do that.

dheap, 2018-01-05:
It's not a bug. See for example `~/jmri/xml/programmers/parts/SoundPane.xml`:
```
    <row>
      <display item="Sound Group 3 Option 1" layout="above" label=""/>
      <label>
        <text> </text>
      </label>
      <display item="Sound Group 3 Option 1" layout="above" format="hslider"/>
    </row>
```

Change the behaviour and you break that (and thousands of others).

bobjacobsen, 2018-01-05:
Sorry, I don’t understand. That’s a blank label in the first element. How is it bad to allow that to be overridden?

Bob

> On Jan 5, 2018, at 12:13 PM, Dave Heap <notifications@github.com> wrote:
> 
> It's not a bug. See for example ~/jmri/xml/programmers/parts/SoundPane.xml:
> 
>     <row>
>       <display item="Sound Group 3 Option 1" layout="above" label=""/>
>       <label>
>         <text> </text>
>       </label>
>       <display item="Sound Group 3 Option 1" layout="above" format="hslider"/>
>     </row>
> Change the behaviour and you break that (and thousands of others).
>

dheap, 2018-01-05:
Look more carefully. The two elements refer to the same variable. The current behaviour is critical. I'll make an example...

dheap, 2018-01-05:
Take for example `~/jmri/xml/programmers/parts/SoundLevelsPane.xml`:
```
      <row>
        <display item="Sound Setting nn" layout="above" label=""/>
        <label>
          <text> </text>
        </label>
        <display item="Sound Setting nn" layout="above" format="hslider"/>
      </row>
```
Consider the ECO100
[image]

Allow the label="" to be overridden and you get 

[image]

Considering `SoundLevelsPane.xml` alome is 2,640 occurrences in 174 files. That's just the tip of the iceberg.

dheap, 2018-01-05:
Gotta go have breakfast...

dheap, 2018-01-05:
Allow the label="" to be overridden and you affect 13,909 occurrences in 227 files.

AlanUS, 2018-01-05:
@dheap So the real issue is with label="" in pane that means 'do not show a label' for this display item.
I did not think about this case that is used extensively with sliders, I agree.
But I guess this can be set as a rule; If a non void label is set, then it can be overridden. 
But there could be other side effects that I'm not aware of.

dheap, 2018-01-05:
18,846 occurrences in 237 files to check (of which 13,909 are null).

A further 1,769 occurrences of label=" +" (grep for one or more spaces). Some are single space (probably just suppression), some are multiple spaces, probably spacing and suppression combined.

That leaves 3,168  places where appearance will change...

AlanUS, 2018-01-05:
Killing me with numbers...
Anyway I fixed manually the case that led me to raise the issue, by removing labels from a pane that was too specific, and moving them to the CV variables definition. Took me a couple of hours though.
Looking at the generic panes (programmers/parts), they are all mostly without label, in order to be generic enough.
To me, in the last 4 years that I've been contributing to JMRI decoder definitions, I have always though that the proper practice was to define a label for each variable, considering there was no label in panes.
But I understand that changing the current behavior will trigger some unwanted result.
I was just asking ....

dheap, 2018-01-05:
[Create a Custom Programmer](http://jmri.org/help/en/html/apps/DecoderPro/Programmer.shtml) (actually about Pane definitions) states under the "display" element:

- _label_ - If present, this will be used to label the variable on the pane instead of the name or standard name of the variable.
..
- _tooltip_ - Allows you to specify a tooltip to be presented when the cursor is left over the variable. If not present, the tooltip defined in the decoder file is used.

So it already takes into account _item_ and _label_ attributes/elements of the variable and pledges to unconditionally overwrite them.

This is further explained [here](http://jmri.local/help/en/html/apps/DecoderPro/Programmer.shtml#display):

- If a label is specified, it will be used. Use a label if you want absolute control over what a variable should be called, regardless of what the decoder file calls it, and you don't like the standard name.

Definition authors have used this page as their rulebook. I don't think changing it (and hence pane behaviour) retrospectively is appropriate.

I think its unfortunate that the _tooltip_ code doesn't behave as specified. There's a case for bringing that into line with both the documentation and  _label_ behaviour but even that could have side effects...

dheap, 2018-01-05:
See my last comment about the help pages (while you were closing this). Should the _tooltip_ code be changed?

AlanUS, 2018-01-05:
> Should the tooltip code be changed?

Well no. I could use the very same arguments and statistics as you the other way round. So unless someone wants to spend considerable time checking (and fixing) every occurrence, leave it as is.

dheap, 2018-01-05:
Interestingly, we used the "pane _label_ overrides all" rule some years ago to enforce consistent terminology for address types in `~/jmri/xml/programmers/parts/BasicPane.xml`. There was a mix of short/long and 2-digit/4-digit terminology. Since  2-digit/4-digit terminology is actually incorrect per NMRA Standards, we were able to eliminate it.

It also worked in translators' favour as there is now only one place to translate address terminology.

dheap, 2018-01-05:
I'm still concerned that the tooltip code does not behave as specified in [Create a Custom Programmer](http://jmri.org/help/en/html/apps/DecoderPro/Programmer.shtml) and I think there may be a case for fixing that.

@bobjacobsen Do you think we should bring the _tooltip_ code into line with both the documentation and the _label_ behaviour?

bobjacobsen, 2018-01-05:
Part of the problem here is that we have too many panes inside the decoder definitions.  Some of that is necessary, but some of it is historical. In the long run, working on the common panes so they're handling more cases would be good.

I think it would also be good to add a suppressLabelDisplay="yes/no/default" or something similar instead of relying on the side-effects of label="".  

As to changing the tooltip logic vs documentation:  Definitely agree they should be consistent.  My memories of how this is all supposed to work and where the special cases lie are clearly faulty, sorry, so I have no opinion on which way is the right one.

dheap, 2018-01-05:
I've raised a new Issue #4709 to change _tooltip_ behaviour.

dheap, 2018-01-06:
Decided to keep existing behaviour and update documentation. See #4710.

closed 2018-01-06 by dheap