JMRI/JMRI #4703: Travis CI runtime

bobjacobsen, 2018-01-05:
If I've understood the logs properly, Travis CI graphical runs the JavaDoc step, but Travis CI headless does not.

In the interests of getting the graphical run done quicker, should we move the javadoc step over to the headless task?

(Although we might want to wait a bit, to see if there's any significant change in runtime due to the current InstanceManager work)

pabender, 2018-01-05:
Would it be helpful to pull some of the tasks currently done by the two Travis jobs into a third job?

I could see splitting off the javadoc, line endings, and findbugs tests.  This would give us something like:
Headless unit tests
Headed unit tests
Static tests.

I am pretty sure line endings and findbugs are still run in the headless job.

There may be other things we are grouping in with unit tests that would fall into the static tests category.

pabender, 2018-01-05:
Btw, the current logs for headless are usually too long to read without going to the raw logs.

rhwood, 2018-01-05:
Keep in mind that we are only allowed to run three Travis CI jobs simultaneously, so having three jobs per push fills that capacity.

bobjacobsen, 2018-01-23:
OK, coming back to the original question:  Travis graphical takes longer than Travis headless. If we move javadoc to Travis headless, the testing will get done quicker because Travis graphical, the rate limiting step, will get done quicker.  Any objections to doing this?

pabender, 2018-01-23:
I would say go ahead with that move

bobjacobsen, 2018-01-24:
PR #4817 addresses this, though perhaps not in an elegant way. (I don't have much Maven experience).  It overshoots a bit in balancing the time:  It went from about 29min/39min to 36/33. Still, perhaps that will give the graphical CI a bit more headroom.

closed 2018-01-24 by bobjacobsen