JMRI/JMRI #4709: Bring tooltip code into line with both the documentation and label behaviour

dheap, 2018-01-05:
For prior discussion, see #4702.

### Problem
The _tooltip_ code does not behave as specified in [Create a Custom Programmer](http://jmri.org/help/en/html/apps/DecoderPro/Programmer.shtml)

### Background
[Create a Custom Programmer](http://jmri.org/help/en/html/apps/DecoderPro/Programmer.shtml) (actually about Pane definitions) states under the "display" element:

- _label_ - If present, this will be used to label the variable on the pane instead of the name or standard name of the variable.
..
- _tooltip_ - Allows you to specify a tooltip to be presented when the cursor is left over the variable. If not present, the tooltip defined in the decoder file is used.

This is further explained [here](http://jmri.local/help/en/html/apps/DecoderPro/Programmer.shtml#display):

- If a label is specified, it will be used. Use a label if you want absolute control over what a variable should be called, regardless of what the decoder file calls it, and you don't like the standard name.

### Comment
Behaviour should be as documented (match _label_ behaviour).

 ### Suggested Action
- [ ] Fix _tooltip_ code to behave as documented (override any decoder file _tooltip_)

AlanUS, 2018-01-05:
This will cause side effects.
e.g. I have created specific tooltips for SoundTraxx CV105/106 (#4681) that override the pane tooltips of the the generic Basic pane.
If the tooltip on pane has precedence, then in order to get the same effect, it will mean creating a specific Basic pane for SoundTraxx, which is not a desirable solution.
There might be similar side effects elsewhere.
Alternative implementation suggestion welcome (for the case above, could be additional placeholders in the Basic pane, without tooltips).

dheap, 2018-01-06:
Since raising this and breaking for coffeeI had thought about just the situation you mentioned and your comment reinforces my thinking. Maybe we should keep the existing behaviour and just change the _tooltip_ documentation.

dheap, 2018-01-06:
That way we keep the existing suppression behaviour, enforce standard names where appropriate but provide an avenue for manufacturer-specific explanatory comments.

AlanUS, 2018-01-06:
> Maybe we should keep the existing behaviour and just change the tooltip documentation.

@dheap Agreed.

dheap, 2018-01-06:
Decided to keep existing behaviour and update documentation to match. See #4710.

closed 2018-01-06 by dheap