JMRI/JMRI #4714: Layout Block Routing is broken in 4.11.2ish

dsand47, 2018-01-06:
The path information is not being built for turnouts.

4.11.1:
[image]

4.11.2ish:
[image]

I was able to get around the issue on my development system using the following:
[image]

Here are a couple of forced log statements:
4.11.1ish INFO  - LayoutTurnout TO0 != LayoutTurnout TO0: will quit [AWT-EventQueue-0]
4.11.2ish INFO  - jmri.jmrit.display.layoutEditor.LayoutTrackExpectedState@3c690908 != jmri.jmrit.display.layoutEditor.LayoutTrackExpectedState@62d5f266: will quit [AWT-EventQueue-0]

There might be more issues like this, but for now I can resume my NX development.

Dave Sand

geowar1, 2018-01-06:
@rhwood did a recent PR #4646 that replaces parallel data structs with a single results class (object)… I suspect that you want "if (!dtos.get(i).equals(stod.get(i))) {" or "if (!dtos.get(i).getObject().equals(stod.get(i).getObject())) {" there.

JLang1951, 2018-01-07:
I'm basically seeing the same problem with my panels when I tried 4.11.2ish. Something is definitely wrong with the block routing paths through turnouts and crossovers. When there are no turnouts or crossovers between source and destination signalmasts you can see valid paths. The signalmast logics between any source and destination mast that was setup through the discovery process when this panel was setup awhile back in a previous version still show up but none of them will activate because the routing is not working. When I delete one of the logics for one of the source and destination masts then try and recreate it using layout editor blocks and turnout boxes checked it says the destination mast is unreachable. Block routing paths is definitely not working. Here is the panel that I am playing with.
[JDL_NYC_LEPanel_SIM.zip](https://github.com/JMRI/JMRI/files/1610188/JDL_NYC_LEPanel_SIM.zip)
NCE Simulated
CMRI Simulated
I hope this can help in some way to fix things.

John Lang

dsand47, 2018-01-07:
PR #4731 has been submitted to resolve this issue.

JLang1951, 2018-01-07:
Thanks Dave. Let me know when the new build will be available.

dsand47, 2018-01-11:
Jenkins build 2188 resolves this problem.

closed 2018-01-11 by dsand47