JMRI/JMRI #4715: Multiple Index value in CV31 - Märklin mLD/3-mSD/3

AlanUS, 2018-01-06:
[Request on JMRI User Group](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/144521) to add [Märklin mLD/3-mSD/3](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwjEkI6D78LYAhXPhrQKHc6-BzwQFggnMAA&url=https%3A%2F%2Fwww.maerklin.de%2Ffileadmin%2Fmedia%2Fservice%2Fdecoder-updates%2FmLD3_mSD3_Zusatzanleitung_0716.pdf&usg=AOvVaw0Lmkypn1pyFKCTjbYjXaNy)

Interestingly, Märklin is using 3 different possible values for CV31: the usual 16 value, and two others 17 and 255. This is allowed from the NMRA perspective.

However how would it work in JMRI? The simple index declaration (e.g. in SoundTraxx Tsunami 2) implies that CV31=16.
Would a CV declaration like CV="17.1.297" work ?

Besides Märklin Manufacturer number 8 is not (yet) in JMRI and not referenced either in the latest published NMRA list of Manufacturers dated Nov. 4, 20216

dheap, 2018-01-06:
@AlanUS Not a problem. Take a look at ESU decoders. They use all sorts of values for both CVs 31 & 32:
```
            <capability>
                <name>Indexed CV access</name>
                <parameter name="PI">31</parameter>
                <parameter name="SI">32</parameter>
                <parameter name="cvFirst">false</parameter>
            </capability>
```
Typical CVs are "0.255.285", , "16.0.375", "16.1.259", "16.4.381". Other CVs not yet used (except in my development code) are "1.1.259", "0.252.310"...

dheap, 2018-01-06:
The SoundTraxx definitions:
```
            <capability>
                <name>Indexed CV access</name>
                <parameter name="PI">32</parameter>
                <parameter name="SI">31</parameter>
                <parameter name="cvFirst">false</parameter>
            </capability>
```
are a special case where only CV32 is specified (and hence designated the PI) under the assumption that the CV31 value in the decoder is readonly and fixed at 16.

My view is that these definitions are incorrect to make that assumption.

From the Tsunami 2 Diesel reference Manual:
**Note:** _To access Indexed CV Page 1 with CVs 257-512, CV 31 (CV Index 1) must be set
to a value of 16 (default) and CV 32 (CV Index 2) must be set to a value of 1 (default)._
Again:
**Note:** _To access Indexed CV Page 2 with CVs 257-512, CV 31 (CV Index 1) must be set
to a value of 16 (default) and CV 32 (CV Index 2) must be set to a value of 2._

These definitions never do that and hence could fail to read/write indexed CVs.

AlanUS, 2018-01-06:
> Besides Märklin Manufacturer number 8 is not (yet) in JMRI and not referenced either in the latest published NMRA list of Manufacturers dated Nov. 4, 20216

This looks like a documentation error.
[Older version m/LD-m/SD](https://www.maerklin.de/fileadmin/media/service/technische_informationen/Umruestdecoder_mLD-mSD_CV-Liste_DCC.pdf) mentions 131 (Trix) as manufacturer. Value 8 probably refers to the (usual) reset value.

AlanUS, 2018-01-06:
@dheap  Thanks for confirmation on syntax for Indexed variables

closed 2018-01-06 by AlanUS