JMRI/JMRI #4716: Possible issue with SoundTraxx Indexed CVs

AlanUS, 2018-01-06:
From discussion initiated on #4715

@dheap wrote

> 
> The SoundTraxx definitions:
> 

>             <capability>
>                 <name>Indexed CV access</name>
>                 <parameter name="PI">32</parameter>
>                 <parameter name="SI">31</parameter>
>                 <parameter name="cvFirst">false</parameter>
>             </capability>

> 
> are a special case where only CV32 is specified (and hence designated the PI) under the assumption that the CV31 value in the decoder is readonly and fixed at 16.
> 
> My view is that these definitions are incorrect to make that assumption.
> 
> From the Tsunami 2 Diesel reference Manual:
> Note: To access Indexed CV Page 1 with CVs 257-512, CV 31 (CV Index 1) must be set
> to a value of 16 (default) and CV 32 (CV Index 2) must be set to a value of 1 (default).
> Again:
> Note: To access Indexed CV Page 2 with CVs 257-512, CV 31 (CV Index 1) must be set
> to a value of 16 (default) and CV 32 (CV Index 2) must be set to a value of 2.
> 
> These definitions never set the value of CV31 to 16 and hence could fail to read/write the specified indexed CVs.

AlanUS, 2018-01-06:
As the Tsunami 2 and Econami have been around for a couple of years, and users never reported issue with Indexed variables; it is assumed that CV31=16 all the time is implied.
Indexed CVs in SoundTraxx TSu2 ane Eco are declared like "1.297", i.e. CV32=1 (PI) and CV31 (SI) not specified. 

CV31 is defaulted to 16, and if not overwritten, will keep 16 as value all the time, therefore does not need to be written at each access. However CV31 is NOT read-only so could be changed accidentally by a user, thus disabling access to indexed variables.

Should this be modified in JMRI and CV31=16 enforced in a systematic way, i.e. CVs declared as "16.1.297" and PI=31 / SI=32?

dheap, 2021-03-31:
@AlanUS > As the Tsunami 2 and Econami have been around for a couple of years, and users never reported issue with Indexed variables; it is assumed that CV31=16 all the time is implied.
> Indexed CVs in SoundTraxx TSu2 ane Eco are declared like "1.297", i.e. CV32=1 (PI) and CV31 (SI) not specified.
>
> CV31 is defaulted to 16, and if not overwritten, will keep 16 as value all the time, [...]

I'll check whether this is still not done.

dheap, 2021-05-15:
This still hasn't been done, but needs doing to prevent problems.

dheap, 2021-06-29:
#9925 will resolve.

closed 2021-07-09 by danielb987