JMRI/JMRI #4717: No label display in pane

AlanUS, 2018-01-06:
Follow-up of #4702

Currently in order not to display a label, e.g. for a slider when the value is also displayed in digital form, the display variable has attribute label="". This works properly as pane labels are not overridden by labels defined at each individual variable declaration in decoder files.

@bobjacobsen suggested

> I think it would also be good to add a suppressLabelDisplay="yes/no/default" or something similar instead of relying on the side-effects of label="".

I also think that an explicit attribute for no label display would be better, should we change one day the order of precedence for labels between panes and decoder files.

I suggest to add one value to the existing attribute "layout" of `\xml\schema\types\panetype.xsd`
```
    <xs:attribute name="layout">
        <xs:simpleType>
            <xs:restriction base="xs:string">
                <xs:enumeration value="left" />
                <xs:enumeration value="right" />
                <xs:enumeration value="below" />
                <xs:enumeration value="above" />
            </xs:restriction>
        </xs:simpleType>
    </xs:attribute>
```
Could be anything like "noDisplay" or "noLabel"

dheap, 2018-01-06:
Bedtime now but setting it as a value of `layout` rather than a separate attribute will create alignment problems.

Consider the hslider case. The text field will no longer be aligned with the slider because there is no way to specify the alignment to match hslider with the label above (or below).

dheap, 2018-01-06:
Update:
Consider the hslider case. The text field will no longer be aligned with the slider because there is no way to specify the text field alignment to match hslider with the label above (or below).

The PaneProgPane code has to know where to put a dummy text field so the two items in the same row have matching alignment.

We end up with this:
[image]

Instead of this:
[image]

dheap, 2018-01-06:
Must go back to bed...

AlanUS, 2018-01-06:
Then the attribute could be 'noLabelAbove' to take this case into consideration. But maybe a separate attribute would be easier to handle.
BTW the use of grid could resolve a number of these alignment problems. The sound volume pane very often looks ugly - like in your example - when label' length exceed the slider length.

dheap, 2018-01-06:
I think this is a "solution" in search of a problem. We are adding complexity to the task of writing a decoder definition.

The main justification seems to be "should we change one day the order of precedence for labels between panes and decoder files".

We already have explicit rules on this in [the help file](http://jmri.local/help/en/html/apps/DecoderPro/Programmer.shtml#display) which clearly state _"If a label is specified, it will be used. Use a label if you want absolute control over what a variable should be called, regardless of what the decoder file calls it, and you don't like the standard name."_

These are the rules under which the existing >1,000 decoder definitions have been written and we now suggest we _may_ want to retrospectively change those rules, with the risk of breaking an unknown number of definitions. I don't think we should work that way.

I know that the use of "grid" can solve many difficult alignment problems (the "grid" feature and the associated code were entirely my idea and my work) but the reality is that not many definition writers want to (or have) come to grips with the concepts, complexities and nuances of [GridBagLayout](http://docs.oracle.com/javase/tutorial/uiswing/layout/gridbag.html) and neither should they be pushed to do so in order to change the existing rules of precedence and hence invalidate an existing very simple solution for a common task; the alternative of being able to type a value into a variable and not be compelled to use a mouse and slider (a task requiring finer motor control on the part of a user than many may have) to get an exact value (in retrospect it maybe should have been an integral feature of the `DecVarSlider` code).

I can't see the benefit of replacing:
- `layout="left|above|below|right"  label="some overriding label"` and `layout="left|above|below|right"  label=""` 
with something like
- `layout="left|above|below|right"  label="some overriding label"` and `noLabel="left|above|below|right"`

Others may disagree and that is fine. But I wouldn't wish the task of retrospectively trying to deprecate the existing usage by updating existing definitions and deciding how to handle cases like `label="    "` where the embedded spaces may or may not be influencing alignment.

However I very strongly oppose the idea of changing the existing clear rules of precedence in [Create a Custom Programmer](http://jmri.local/help/en/html/apps/DecoderPro/Programmer.shtml#display) and applying the new rules retrospectively to existing definitions, most likely breaking quite a few of them. If we have a new problem we come up with a new solution that doesn't invalidate existing work.

@AlanUS please don't take this as a personal criticism of you. You have already made (and are still making) a great contribution to a very difficult task and we will do all we can to make it easier for you. But we always need to tread lightly in the footsteps of the many others before either of us and not compromise existing good work.

In a project like this we will inevitably have many with strongly held views on various topics and there will inevitably be times when discussion will be robust. I've been on both the giving and receiving end of this and I know it is not an easy path to tread.

AlanUS, 2018-01-06:
@dheap No offense. Just trying to make suggestions. Without changing the current behaviour, it might be better programming legibility for future definitions to have an explicit attribute as proposed by @bobjacobsen 

Yes the "grid" feature is a great one and thank you for this. For complex tables, it might be easier to program and debug than a complex entanglement of `row `and `column`.

The generic sound volume pane could be easily converted to grid. I can do that if you think it worth.

dheap, 2018-01-06:
> @dheap No offense. Just trying to make suggestions.

I didn't take offence at your suggestions. I just didn't agree.
> Without changing the current behaviour, it might be better programming legibility for future definitions to have an explicit attribute as proposed by @bobjacobsen

That's where I have a difference of opinion with both you and @bobjacobsen. My view was that it increased complexity without improving legibility. But that's just a difference of opinion and not a show-stopper.
> Yes the "grid" feature is a great one and thank you for this. For complex tables, it might be easier to program and debug than a complex entanglement of row and column.
> 
> The generic sound volume pane could be easily converted to grid. I can do that if you think it's worth.

The ESU Sound pane is already multiple grids (as do a number of other ESU panes) but it doesn't address all the problems that concern you.
```
        <grid gridx="NEXT" gridy="CURRENT" ipadx="10" ipady="10">
            <!-- default settings for all grid items -->
            <griditem gridx="0" gridy="NEXT" weightx="1" anchor="LINE_END">
                <!-- weightx expands column to right-justify row -->
                <soundlabel num="1"/>
            </griditem>
            <griditem>
                <display item="Sound Volume 1" layout="above" label=""/>
                <display item="Sound Volume 1" format="hslider" layout="above"/>
            </griditem>
            <griditem>
                <display item="Sound Drive 1" format="checkbox" layout="above"/>
            </griditem>
            <griditem>
                <display item="Sound Min Speed 1" layout="above" label=""/>
                <display item="Sound Min Speed 1" format="hslider" layout="above"/>
            </griditem>
            <griditem>
                <display item="Sound Max Speed 1" layout="above" label=""/>
                <display item="Sound Max Speed 1" format="hslider" layout="above"/>
            </griditem>
            ...
```
It's not just a case of splitting the two into separate grid items. I tried it and you would need a complex combination of anchors and weights to replace the `layout="above"` and avoid upsetting the overall row justification. Way beyond the average user to understand and we're looking at at least 1,400 occurrences in 39 files to address, convert to grid if needed and redo.

It doesn't fix the label over-width issue either. It's not even possible to split the label off from  the variable into a separate row above so you can make it span two columns (unless it was possible to make a display element with just the label and a hidden field!) . Even then the grid layout would be horrendously complex.

In Australia we have a saying; "If it ain't broke don't try to fix it!". I don't see any currently-broken behaviour or any net gain from the proposed changes.

Dave

dheap, 2018-01-06:
Time for breakfast...

dheap, 2018-01-06:
I like the idea of a `silderTextBox="left|above|below|right|none"` attribute (probably with a default of `"none"`) applicable to the "slider" formats. It would remove much of the need for `label=""`, address the label overflow issue and enable cleanup of pane XML files. Shouldn't be too hard to put into `PaneProgPane`.

AlanUS, 2018-01-07:
@dheap 
Going further on your idea of `silderTextBox="left|above|below|right|none" attribute ` we could also imagine a more sophisticated 'combo' slider object that would have an attribute for displaying (or not) the digital value on either side of the slider.
As you know today the construct is to display twice the same item/variable, one with label and the other without. That could be a more elegant way to reach the same or better result and taking care of the alignment programmatically in the combo itself. For new users, I think also that it would be easier to manipulate and understand.

dheap, 2018-01-07:
@AlanUS I'm not sure I follow your 'combo' slider object concept (maybe you have an illustration). 

But I think we can do better than at present. ESU uses a Combo textbox/Spinner beside each slider in its LokProgrammer software but I think that's overkill...

No promises but I'll take a closer look at the PaneProPane code when I catch up with some of the rest of life. Don't expect it too soon...

dheap, 2018-01-07:
My idea is that this:
```
  <row>
    <display item="Sound Volume 1" layout="above" label=""/>
    <display item="Sound Volume 1" format="hslider" layout="above"/>
  </row>
```
would become simply:
```
    <display item="Sound Volume 1" format="hslider" layout="above" silderTextBox="left"/>
```
with the label spanning above both the slider and the text box in this  particular example.

AlanUS, 2018-01-07:
@dheap 
Just taking the example above in this thread - generic sound volumes. The digital value is displayed on the left-hand side of the slider. This is being achieved with 2 times displaying the same value, the digital one without label and the slider with label above.
So if we have two attributes:
`sliderTextBox="left|above|below|right|none"`      --> above in this case
`sliderDigitalValue= "left|above|below|right|none"` --> left in this case
and we use only ONE display variable.
The 'none' value is the default to remain compatible. If `sliderTextBox is different than `none,`then same effect as with current 'layout' attribute. Need to think a bit more about precedence rules between the various attributes if we go that way, to remain compatible with current syntax.

Second example, basic speed control pane, Vstart/Vmid/Vmax:
label in on the left-hand side, then digital value, then slider, on the same row.
`sliderTextBox="left|above|below|right|none"`      --> left in this case, i.e. left of the combo
`sliderDigitalValue= "left|above|below|right|none"` --> left in this case

AlanUS, 2018-01-07:
`    <display item="Sound Volume 1" format="hslider" layout="above" silderTextBox="left"/>`
I was writing in parallel. I did not get the concept with `sliderTextBox`. You proposal is much simpler indeed.

AlanUS, 2018-01-07:
And adding attribute `none` to the layout `attribute` allows to get rid of the `label=''` implicit syntax. OK I'm a bit stubborn on that one....

dheap, 2018-01-07:
```
    <display item="Sound Volume 1" format="hslider" layout="above" label=""/>
   <display item="Sound Volume 1" format="hslider" layout="below" label=""/>
   <display item="Sound Volume 1" format="hslider" layout="none"/>

```
Actually produces three different results during pane construction. The first two put a standard-height (one line of text high) space either above or below the slider , the third has no space around the slider. Makes a big difference to the size of the element and hence the space between pane elements and hence overall pane appearance. Users may end up having to insert 
```
      <label>
        <text> </text>
      </label>
```
in order to get desired spacing.

AlanUS, 2018-01-07:
> Users may end up having to insert
> ```
>       <label>
>         <text> </text>
>       </label>
> ```
So not really a simplification in this case of use. So the `label=""` form has to be kept first for compatibility and second to allow spacing in a simpler way..

dheap, 2018-01-07:
@AlanUS Agreed, that's my feeling. But I will look into the other simplification.

dheap, 2018-01-07:
Time for bed now...

closed 2021-04-14 by stale[bot]