JMRI/JMRI #4722: Digitrax Steal and WiThrottle

JohnSL, 2018-01-06:
I've had a short conversation with @mstevetodd via email about the behavior of the WiThrottle protocol and JMRI. From that conversation, it sounds like the behavior between JMRI and Digitrax LNWI is different. Here is the WiThrottle sequence he described for LNWI (here x is usually T):

| Phone | LNWI
|---|---
| Mx+ | MxS
| Say Yes in dialog |
| MxS | Mx+ and rest of acquire reply

And here is the sequence I've confirmed with the iPhone WiThrottle app and JMRI:

| Phone | JMRI
|---|---
| Mx+ | MxS
| Say Yes in dialog |
| MxS | no reply
| Mx- |
| Mx+ | Mx+ and rest of acquire reply

If the Mx- is not present, JMRI won't complete the add. So the above sequence appears to be the required sequence. I've also verified this with my own implementation.

Some background. Several years ago, while working on a new throttle, I wrote most of the information on the WiThrottle Protocol page. I want to update this information, as I never finished that project. Therefore, I also want to confirm a few things:

* Is JMRI 4.10 the first version with support for steal?
* Is the above behavior for JMRI correct, or a bug?

Second, I'm also working with TCS on a set of throttles that support both LCC and WiThrottle: http://model-railroad-hobbyist.com/blog/20164

devel-bobm, 2018-01-07:
Given the state of LocoNet "acquisition of throttles" being very much "in flux", please do not make any assumptions about WiThrottle Server or its behavior when working with LocoNet-based hardware.  The LocoNet throttle acquisition needs to be fixed first!  One (of many) refs is #4726; see other refs in that pull-request.

JohnSL, 2018-01-09:
OK, so it sounds like this is a bug in JMRI, correct?

pabender, 2018-01-09:
@JohnSL,

Can you see what happens with one of the daily uilds from master ( http://jmri.tagadab.com/jenkins/job/Development/job/Packages/ )

PR#4726 which was merged into master earlier today is included in the latest one, and it fixes many of the issues with throttle acquisition on Loconet.

JohnSL, 2018-01-09:
I installed and tried version JMRI.4.11.2ish+jenkins+20180109T0146Z+Rf12811d.exe, and that still requires two presses of the Set button in the iPhone WiThrottle app. EngineDriver, on the other hand, isn't able to steal the loco. It asks to steal the first time, but then after that, it doesn't select the loco, and doesn't ask to steal again. However, if I exit EngineDriver and start it up again, selecting the loco works this time without asking about steal.

Do you need more information? I can run Wireshark traces to see what's going on. I'm also in the process of getting this machine (the one attached to the LocoBuffer) setup so I can download and debug the code. I've mostly looked at the WiThrottle code, so I'd need pointers if you need me to look elsewhere.

devel-bobm, 2018-01-09:
We need to know what command station and LocoNet throttles (if any) you are using.

We will need LocoNet logs taken during your attempts to acquire a WiThrottle.  Recent changes to LocoNet throttle acquisition were thought to address this sort of issue.  Apparantly there is some boundary-condition which we haven't addressed.

We need LocoNet logs from "Monitor LocoNet" which " show raw data".  It would be helpful to have some comments injected into the log to describe what you "just did" and/or what you "will be doing".  This can be done via the "add comment" feature in LocoNet Monitor, while you are experimenting.

And it would be helpful to have any "System Console" text that is emitted during your testing.  Occasionally, such logs can give a vital clue to the failure mode, saving much guesswork and trial-and-error debugging for the user and the developer.

JohnSL, 2018-01-10:
OK, here is what I think you need.

Command Station: **Digitrax Zephry DCS 50**
Connection: **LocoBuffer USB**

Here are lines from the console:

```text
2018-01-09 20:39:15,970 withrottle.UserInterface              INFO  - Creating new WiThrottle DeviceServer(socket) on port 12090, waiting for incoming connection... [WiThrottleServer]
2018-01-09 20:39:15,970 withrottle.DeviceServer               INFO  - Received Name: John’s iPhone [WiThrottleUIDeviceServer]
2018-01-09 20:39:53,519 jmrix.AbstractMonPane                 WARN  - File selection dialog box did not provide a path to the specified file.  Log will be saved to C:\Users\john_\JMRI\My_JMRI_Railroad\monitorLog.txt [AWT-EventQueue-0]
2018-01-09 20:39:53,519 jmrix.AbstractMonPane                 WARN  - startLogButtonActionPerformed: getSelectedFile() returns monitorLog.txt monitorLog.txt [AWT-EventQueue-0]
2018-01-09 20:39:53,519 jmrix.AbstractMonPane                 WARN  - startLogButtonActionPerformed: is attempting to use returned file path and file name C:\Users\john_\JMRI\My_JMRI_Railroad\monitorLog.txt [AWT-EventQueue-0]
2018-01-09 20:40:10,522 loconet.LnThrottleManager             WARN  - slot 10 address 346 is  already in-use. [AWT-EventQueue-0]
2018-01-09 20:40:10,522 withrottle.ThrottleController         WARN  - Throttle request failed for 346(L) because Steal Required. [AWT-EventQueue-0]
2018-01-09 20:40:28,149 loconet.LnThrottleManager             WARN  - user agreed to steal address 346, but no code is in-place to handle the 'steal' (yet) [WiThrottleUIDeviceServer]
```

And here is from the LocoNet Monitor:

```text
[A0 0A 00 55]  Set speed of loco in slot 10 to 0.
About to select loco
[BF 02 5A 18]  Request slot for loco address 346.
[81 7E]  Master is busy.
[E7 0E 0A 33 5A 00 04 07 00 02 04 00 55 25]  Report of slot 10 information:
	Loco 346 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=On, F4=Off, F5=Off, F6=Off, F7=On, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x55 0x00 (10880).
About to say yes to steal
[BA 0A 0A 45]  Set status of slot 10 to IN_USE.
[B4 3A 00 71]  LONG_ACK: The Move Slots command was rejected.
[EF 0E 0A 33 5A 00 04 07 00 02 04 71 02 0B]  Write slot 10 information:
	Loco 346 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=On, F4=Off, F5=Off, F6=Off, F7=On, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
[B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
Said yes
About to set again
[A0 0A 00 55]  Set speed of loco in slot 10 to 0.
Done set
```
Is that enough? Or do you need something more?

Pugwash1, 2018-01-10:
On the steal the write  slot happens right after the null move, no pause. In a no steal the write slot follows after about 14ms later.
RX: BF 0F 53 1C  <- ask for slot
RX: E7 0E 01 33 53 00 00 07 00 0F 00 00 42 3D <- status in-use
RX: BA 01 01 45 <-null move
RX: EF 0E 01 33 53 00 00 07 00 0F 00 71 02 04 <- write slot, no pause 
RX: B4 3A 00 71 <- null move rejected.

Pugwash1, 2018-01-10:
@bobjacobsen @devel-bobm 
When a DT402 steals from JMRI according to the log it does not issue the null move, it just writes its throttle ID in. You just have to love this stuff and laugh. I did double check on the other sniffer and it saw the same thing.

> 11:35:45.797: jmri allocates throttle ----------------------------------------------------------------
11:35:47.389: [BF 0F 53 1C]  Request slot for loco address 2003.
11:35:47.418: [E7 0E 01 23 53 00 00 07 00 0F 00 00 42 2D]  Report of slot 1 information:
	Loco 2003 is Not Consisted, Idle, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x42 0x00 (8448).
11:35:47.494: [BA 01 01 45]  Set status of slot 1 to IN_USE.
11:35:47.496: [E7 0E 01 33 53 00 00 07 00 0F 00 00 42 3D]  Report of slot 1 information:
	Loco 2003 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x42 0x00 (8448).
11:35:47.509: [EF 0E 01 33 53 00 00 07 00 0F 00 71 02 04]  Write slot 1 information:
	Loco 2003 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
11:35:47.510: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
11:35:52.359: [A0 01 00 5E]  Set speed of loco in slot 1 to 0.
11:35:53.450: dt402 steals --------------------------------------------------------------------------
11:35:57.441: [BF 0F 53 1C]  Request slot for loco address 2003.
11:35:57.452: [E7 0E 01 33 53 00 00 07 00 0F 00 71 02 0C]  Report of slot 1 information:
	Loco 2003 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x02 0x71 (369).
	----------------------NB no null move --------------------
11:36:00.123: [EF 0E 01 33 53 00 00 07 00 0F 00 00 42 35]  Write slot 1 information:
	Loco 2003 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x42 0x00 (8448).
11:36:00.124: [B4 6F 7F 5B]  LONG_ACK: Function not implemented, no reply will follow.
11:36:00.366: [A0 01 00 5E]  Set speed of loco in slot 1 to 0.

devel-bobm, 2018-01-11:
I was able to try an experiment (with the help of a smart-phone-equipped club member).  

After a very cursory review of console logs, it _seems_ like WiThrottleServer may be seeing the "do you want to steal" information from LocoNet, and then sending a message to the throttle while simultaneously automatically responding "steal it" to LocoNet.  But then WiThrottle doesn't do anything with the "stolen" throttle.  Meanwhile, the LocoNet code appears to be "stealing" the slot.

Using EngineDriver 2.16, attempted to acquire loco 701 when it was dialed-up on a DT400 throttle.  First time Engine Driver asked "steal" and user said "yes", but Engine Driver did not take control.

Tried again, ED did NOT ask "steal?", just plain got the loco.

Logs follow:
```
Some directly-related messages:
20:29:33,563 withrottle.DeviceServer               INFO  - Received Name: Droid Turbo Throttle [WiThrottleUIDeviceServer]
20:29:35,576 loconet.LnThrottleManager             DEBUG - requestThrottleSetup: address 701(L), control true [WiThrottleUIDeviceServer]

20:29:35.591: [BF 05 3D 78]  Request slot for loco address 701.
20:29:35.607: [E7 0E 0B 33 3D 00 00 07 00 05 00 33 44 66]  Report of slot 11 information:
	Loco 701 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x44 0x33 (8755).

20:29:35,607 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 11 [AWT-EventQueue-0]
20:29:35,607 loconet.SlotManager                   DEBUG - LOCO_ADR resp is slot 11 for addr 701 [AWT-EventQueue-0]
20:29:35,607 loconet.SlotManager                   DEBUG - notify listener [AWT-EventQueue-0]
20:29:35,607 loconet.LnThrottleManager             DEBUG - notifyChangedSlot - slot 11, slotStatus 30 [AWT-EventQueue-0]
20:29:35,607 loconet.LnThrottleManager             WARN  - slot 11 address 701 is  already in-use. [AWT-EventQueue-0]
20:29:35,607 withrottle.ThrottleController         WARN  - Throttle request failed for 701(L) because Steal Required. [AWT-EventQueue-0]
20:29:35,607 loconet.LnThrottleManager             DEBUG - cancelThrottleRequest - address 701 [AWT-EventQueue-0]

20:29:39.148: [A0 0B 00 54]  Set speed of loco in slot 11 to 0.
20:29:39,148 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 11 [AWT-EventQueue-0]

20:29:39,585 withrottle.DeviceServer               INFO  - Received Name: Droid Turbo Throttle [WiThrottleUIDeviceServer]
20:29:42,767 loconet.SlotManager                   DEBUG - LACK in state 0 message: B4 3A 00 71 [AWT-EventQueue-0]
20:29:42,783 loconet.LocoNetThrottle               DEBUG - notifyChangedSlot executing for slot 11, slotStatus 30 [AWT-EventQueue-0]
20:29:42,783 loconet.LocoNetThrottle               DEBUG - speed (int) is 0 [AWT-EventQueue-0]
20:29:42,783 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 11 [AWT-EventQueue-0]
20:29:42,814 loconet.SlotManager                   DEBUG - LACK in state 0 message: B4 6F 7F 5B [AWT-EventQueue-0]
20:29:45,575 withrottle.DeviceServer               INFO  - Received Name: Droid Turbo Throttle [WiThrottleUIDeviceServer]

(Some unrelated traffic):
20:29:41,473 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 14 [AWT-EventQueue-0]
20:29:41,473 loconet.SlotManager                   DEBUG - LOCO_ADR resp is slot 14 for addr 130 [AWT-EventQueue-0]
20:29:41,473 loconet.SlotManager                   DEBUG - no request for addr 130 [AWT-EventQueue-0]
20:29:41,504 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 15 [AWT-EventQueue-0]
20:29:41,504 loconet.SlotManager                   DEBUG - LOCO_ADR resp is slot 15 for addr 28 [AWT-EventQueue-0]
20:29:41,504 loconet.SlotManager                   DEBUG - no request for addr 28 [AWT-EventQueue-0]
20:29:41.161: [A0 0A 00 55]  Set speed of loco in slot 10 to 0.
20:29:41,161 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 10 [AWT-EventQueue-0]
20:29:41.395: [BB 0C 00 48]  Request data/status for slot 12.
20:29:41.426: [E7 0E 0C 33 05 00 21 07 00 00 00 34 75 4B]  Report of slot 12 information:
	Loco 5 (short) is Not Consisted, In-Use, operating in 128 SS mode, and is moving Reverse at speed 0,
	F0=Off, F1=On, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x75 0x34 (15028).
20:29:41,426 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 12 [AWT-EventQueue-0]
20:29:41,426 loconet.SlotManager                   DEBUG - LOCO_ADR resp is slot 12 for addr 5 [AWT-EventQueue-0]
20:29:41,426 loconet.SlotManager                   DEBUG - no request for addr 5 [AWT-EventQueue-0]
20:29:41.457: [BB 0E 00 4A]  Request data/status for slot 14.
20:29:41.473: [E7 0E 0E 33 02 00 20 07 00 01 00 07 74 7C]  Report of slot 14 information:
	Loco 130 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Reverse at speed 0,
	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; STAT2=0x00, ThrottleID=0x74 0x07 (14855).
20:29:41.488: [BB 0F 00 4B]  Request data/status for slot 15.
[... cut ...]
20:30:03,874 loconet.LocoNetThrottle               DEBUG - Releasing loco address 701 slot 11 during dispose [WiThrottleUIDeviceServer]
20:30:03,874 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 11 [AWT-EventQueue-0]

20:30:03.905: [B5 0B 23 62]  Write slot 11 with status value 35 (0x23) - Loco is Not Consisted, Idle and operating in 128 speed step mode.
20:30:03,905 loconet.SlotManager                   DEBUG - notify 2 SlotListeners about slot 11 [AWT-EventQueue-0]
```

devel-bobm, 2018-01-11:
The sequence of events for the **first** Loco acquire attempt appears to be:

EngineDriver requests a loco from `withrottle`(Server).  This is passed to the default ThrottleManager.  In this case, it filters down to `LnThrottleManager`.  `LnThrottleManager` requests that `SlotManager` attempt to acquire the loco.  `SlotManager` returns a slot to `LnThrottleManager`, which sees it as "IN_USE" by a throttle with a different ID.  `LnThrottleManager` informs of the need to "steal" via `notifyStealRequest()`.  This passes back via `AbstractThrottleManager.notifyStealRequest()` to `WiThrottle.ThrottleController()`, which is:
```
    @Override
    public void notifyStealThrottleRequired(LocoAddress address) {
        notifyFailedThrottleRequest(address, "Steal Required");

        // this is an automatically stealing impelementation.
//        InstanceManager.throttleManagerInstance().stealThrottleRequest(address, this, true);
    }
```

This invokes `wiThrottle.ThrottleController.notifyFailedThrottleRequest()`, which is:
```
    @Override
    public void notifyFailedThrottleRequest(LocoAddress address, String reason) {
        log.warn("Throttle request failed for {} because {}.", address, reason);
        for (ThrottleControllerListener l : listeners) {
            l.notifyControllerAddressDeclined(this, (DccLocoAddress) address);
            log.debug("Notify TCListener address declined in-use: {}", l.getClass());
        }
    }
```

This invokes `withrottle.DeviceServer.notifyControllerAddressDeclined()`, which is:

```
    @Override
    public void notifyControllerAddressDeclined(ThrottleController tc, DccLocoAddress address) {
        log.debug("notifyControllerAddressDeclined");
        if (multiThrottles != null) {   //  Should exist by this point
            jmri.InstanceManager.throttleManagerInstance().cancelThrottleRequest(address.getNumber(), address.isLongAddress(), tc);
            multiThrottles.get(tc.whichThrottle).canceledThrottleRequest(tc.locoKey);
        }
    }
```
That code invokes `loconet.LnThrottleManager.cancelThrottleRequest()`.

`LnThrottleManager` removes the locomotive address from the list of addresses which throttles have attempted to acquire.

Game over for the first attempt.

It seems to me that at least part of the problem is how `withrottle.ThrottleController.notifyStealThrottleRequired()` immediately returns "failed acquire", rather than posting the question to the phone.

I do not understand how the phone gets a "do you want to steal" message; it is not obvious to me how the code sequence I traced would do that.  And it seems that the answer to that question, posed only for the first acquisition attempt, seems to be "saved" and used during the second acquisition attempt.

Pugwash1, 2018-01-11:
@devel-bobm @bobjacobsen 
There are 3 separate problems here.
1 - The underlying steal process doesn't work for a DCS50. It does work for a DCS51. In both cases though the null move is rejected, other command stations may react differently.
2 - The WifFi steal (on a DCS51) correctly updates the slot info, replaces the ID with the JMRI ID, but that gets lost somewhere.
3 - If a loco on a WiFi throttle is then aquired by another WiFi throttle there is no steal message. Both throttles control it. When one throttle releases it the loco stops (it was set to IDLE and zero speed on release, not COMMON, so packets stop going to the track until the still connected throttle  changes speed. (mysterious stopping locos.)
4 - Setting the throttle to IDLE on dispose, even after a dispatch, is causing grief with Digitrax Consists and any stolen throttle being returned to the pool, as the loco in that slot will stop until the next speed refresh from a throttle. Digitrax command stations on the whole do not like any throttle apart from the owner of the Top loco getting hold of a Sub or Mid. (mysterious slot MAX messages)

Number one is easy to fix, just bypass the null move if the slot is already in-use. 
Numbers two, I cant see what happens, the throttle gets set up correctly but the WiFi, server?, side does seem to process it.
Number three requires that each JMRI PC throttle, WiFi throttle, device has a seperate throttle ID (which it should have), so stealing can be detected.

devel-bobm, 2018-01-11:
@Pugwash1 ,

This LocoNet acquire problem is like an onion.  We are peeling it one step at a time.

Items numbered 1 and 2 can likely be resolved "timing" issues.  I would guess that the DCS50 would work just fine if the "null move" were delayed a bit (the DCS50 is beastly slow at some things!).  The "Throttle ID" update has previously shown some timing sensitivity; a slight delay is likely to resolve problem #2; this needs investigation.
 
Your problem #3 issue has a direct relationship to the basic infrastructure changes that caused all this trouble to begin with - JMRI didn't provide a "throttle ID".  Now it does.  We are struggling thru the growing pains of that addition.

But the mechanism currently uses an identical "throttle ID" for EVERY throttle JMRI creates.  That means that no one WiFi-assigned JMRI LocoNet-based throttle can be differentiated from any other WiFi-assigned JMRI LocoNet-based throttle.  So one WiFi-assigned LocoNet-based throttle cannot "steal" from another.  But even when we update the code to provide pseudo-unique throttle ID numbers for every LocoNet-based throttle, there is still the problem that "steal" doesn't mean "steal" in LocoNet parlance.  It means "share".  Sort of.  Lots of technobabble could follow that last set of statements; I will omit.

Problem #4 issue needs investigation.  I believe that it is based in poor terminology in the definition of what JMRI throttles "do" when a throttle is "released" or "disposed".  To my way of thinking, a "release" is a soft action, while "dispose" is "throw it away with extreme prejudice". So that's what LocoNet code I worked on did.  But there may have been coders working on it since then...

The big-picture problems with LocoNet are not going to be resoved overnight...

devel-bobm, 2018-01-11:
I figured out how the Wi-device got the "need to steal" message - in `withrottle.MultiThrottleController.notifyStealThrottleRequired`:
```
    /**
     * Send a steal required message to the connected device prior to disposing
     * of this MTC
     *
     * @param address of DCC locomotive involved in the steal
     */
    @Override
    public void notifyStealThrottleRequired(LocoAddress address) {
        sendStealAddress();
        notifyFailedThrottleRequest(address, "Steal Required");
    }
```
I think that withrottle should not be sending an immediate rejection to `notifyFailedThrottleRequest()`.- instead it should wait for a user reply.  Or perhaps implement a reply time-out (perhaps 10 seconds), and if the user responds, kill the timer, but if the timer times-out, automatically send a "don't steal" reply back to the (LocoNet) throttle manager.

devel-bobm, 2018-01-11:
I've edited my long log to remove a cut-and-paste error which resulted in duplicated, out-of-time-sequence, messages

Pugwash1, 2018-01-11:
@devel-bobm 
We the exception of the null move on steals I totally agree. I believe we should remove the null move on a steal as the slot is already in-use and the adrSel bit is unlikely set. I cannot find a throttle that does a null move on a steal, but I haven't stopped looking. Number four is not just poor terminology but also there are discrepancies on when digitrax throttles use one or other, I believe the DCS51 and throttles without a "dispatch" always do COMMON.

devel-bobm, 2018-01-11:
@Pugwash1, are you saying that, with respect to your Problem #4, current JMRI behavior matches some Digitrax equipment behavior?

Pugwash1, 2018-01-11:
@devel-bobm
Only in that certain throttles always set to one state, "COMMON", none always set "IDLE", some can do both.
Oh, and the "dispose" I was referring to was what we did on object disposal in the code. The Dispatch button in a throttle window sets "COMMON", but then closing the throttle window sets its "IDLE" and speed zero in the throttle object dispose code.

devel-bobm, 2018-01-11:
@pugwash1, this new JMRI "feature" during "dispose()" of a withrottle device connection (and presumably disposal of a JMRI-native GUI throttle) behavior requires study.  

It is made difficult because one JMRI-or-withrottle-based throttle may "dispose()" its GUI, and that gets propagated down to the JMRI LocoNet Throttle object.  What effect should it have on any other JMRI-or-withrottle-based throttle?

If the goal is to retain control by any other JMRI-or-withrottle-based throttle, that will be difficult.  if I recall correctly, the LocoNet Throttle object doesn't know how many withrottle or JMRI-based-GUI throttles are in-use.  If I recall correctly, that fact is "hidden" from the LocoNet throttle implementation.  So JMRI doesn't know what to do with the slot when a "throttle dispose" occurs.

devel-bobm, 2018-01-11:
@BHoffman351 and @mstevetodd , please be aware of this "issue".  While it starts out as a LocoNet-based set of problems, there appears (in my estimation) an incorrect implementation of "stealing" in jmrit.withrottle, resulting in "odd" behavior with Digitrax-based systems.

The specific problem is described about six comments before this one - look for a code snippet which includes `withrottle.MultiThrottleController.notifyStealThrottleRequired` code.  Symptoms and code analysis leads me to believe that this routine should not be telling the LocoNet code that the steal was rejected; instead the code presumably should defer responding until the user replies to the GUI pop-up question.

Note that this Github issue also has brought up several other issues which are likely LocoNet-specific issues that do not necessarily imply changes to jmri.withrottle.

greenwizard88, 2018-01-24:
Another byproduct of this issue is that the withrottle status pane usually displays the engine as a long address, despite it being a short address being used. The engine is still able to be operated, it's just a ui issue, AFAIK. I suspect the underlying issue goes deeper though.

devel-bobm, 2018-01-24:
@greenwizard88,

Digitrax makes the split at 128, always.  Other systems use other mechanisms.  I cannot claim to know how WiThrottle decides whether an address is "long" or "short".

greenwizard88, 2018-01-24:
@devel-bobm 

Sorry, this was a horrible report. Here's what I observed, using a PR3 connected to a DCS51, and selecting an address via a `MT+S83<;>S83` command (In this case the first "S" represents a short address, +L83 would set a long address, presumably with leading 0's, although I have not tested that).

JMRI 4.8: The WiThrottle window lists the engine as (S)83, responds to WiThrottle commands sent from my app or WiThrottle.app
JMRI 4.10: The WiThrottle window lists the engine as (L)83, only once was I able to control an engine with my own app, but it does work consistently with WiThrottle.app
JMRI 4.11.2, The WiThrottle window lists the engine as (L)83, responds to WiThrottle commands sent sent from my own app or WiThrottle.app

As I have not changed my app, updated WiThrottle, or updated the DCS51 firmware, I can only surmise that the difference is within JMRI.

bobjacobsen, 2018-01-24:
> On Jan 24, 2018, at 10:00 AM, Drew <notifications@github.com> wrote:
> 
> Another byproduct of this issue is that the withrottle status pane usually displays the engine as a long address, despite it being a short address being used. The engine is still able to be operated, it's just a ui issue, AFAIK. I suspect the underlying issue goes deeper though.

JMRI itself knows where each system draws its long/short boundary.  In some cases (NCE, perhaps others) where some addresses can be either long or short, JMRI will display a checkbox or selection box so the user can pick.

But I don’t know how that’s reflected in the phone apps.

Bob

--
Bob Jacobsen
jacobsen@mac.com +1-510-708-5988 AIM, Skype JacobsenRG

dheap, 2018-01-25:
When you create a JMRI throttle, it uses methods such as canBeLongAddress(address), canBeShortAddress(address) & addressTypeUnique(address) from the system-specific Throttle Manager to determine what options to present for "address".

greenwizard88, 2018-01-25:
@dheap This makes sense- but there's an issue with the address length being displayed in the Withrottle applet in JMRI. Since 4.10 it (almost) always states that it's a long address that the physical phone has control of, when in fact the throttle has acquired a short address.

I'm assuming that there's a bug related to how JMRI throttles interact with the WiThrottle package, and how they behave with a (in my case) DCS51 device. Glancing through the changelog, it looks like it's the only relevant thing that changed was swapping DccLocoAddress to LocoAddress. Perhaps this is more relevant to #4726 ?

mstevetodd, 2018-01-27:
@greenwizard88, 
I get rather different results with my DCS240 and JMRI 4.11.3xxx. In my case, requesting a long address 53 looks to be ignored by the command station. Examples, with WiThrottle DEBUG enabled:
EngineDriver sends: `MT+L53<;>L53` and there are no further messages, and nothing changes in the window. 
Conversely, if ED sends `MT+S53<;>S53` withrottle.ThrottleController logs `DccThrottle found for: 53(L)`, which seems odd, but then withrottle.DeviceServer sends `MT+S53<;>` to ED, which is correct. The WiThrottle window is updated to show `53(S)`, which is also correct.
So I cannot acquire a long address < 128. This seems normal to me.
I also note that the JMRI Throttle window doesn't ask for Long or Short, it only asks for address. If I ask for 53, I get 53(S).

dheap, 2018-01-27:
> I also note that the JMRI Throttle window doesn't ask for Long or Short, it only asks for address. If I ask for 53, I get 53(S).

Correct, because the JMRI throttle code asks the Digitrax throttle manager what address types Digitrax considers valid for address 53 and it says only short is valid for that system.

Pugwash1, 2018-01-27:
Engine Driver 2.16. JMRI 4.11.3ish
Ask for 123 (defaults to long) select short. All good. 
That is:  Ask for 12 still says short add 3 for 123 short changes to long.
Steve G.

devel-bobm, 2018-01-31:
This is probably fixed by #4830, which should be in the ongoing "Development" builds at tagadab, and should be in the next test release.

devel-bobm, 2018-01-31:
Let me rephrase my previous comment: #4830 probably fixes the incorrect reporting of address type as "long" when it should have been "short".  That pull request does not address the original question about proper WiThrottleServer behavior which was posed by @JohnSL .

mstevetodd, 2018-11-14:
We continue to get issues regarding the partially-implemented loconet steal mechanism. I checked the latest JMRI master and the "steal not yet implemented" warning is still there.
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/jmrix/loconet/LnThrottleManager.java#L539
@devel-bobm are you still working on the loconet steal?

devel-bobm, 2018-11-14:
@mstevetodd, I am not actively working on "LocoNet throttle acquisition" or "LocoNet throttle steal" behavior.  I have not been following those developments, or even working much on JMRI, because of personal issues.  I do not expect to have any significant time for JMRI development for several months (at best), so it would be best if someone else took up this issue.

pabender, 2018-11-24:
Closed via #6154 and #6168

closed 2018-11-24 by pabender