JMRI/JMRI #4724: trainicon gremlin returns

Cliff-Anderson, 2018-01-07:
Yahoo User reports 4.11.1 problem in message 144569 that duplicates symptoms reported in #3927 and fixed by @bobjacobsen in #3928 

Duplicated the reported symptom using ```PanelPro version 4.11.2ish+cliffaa-rr+20180107T0351Z+Rc4c93f6e96 starts under Java 1.8.0_112 on Windows 10 amd64 v10.0 at Sat Jan 06 19:54:05 PST 2018``` 

BUT, was not able to duplicate the symptom using ```PanelPro version 4.11.2ish+jenkins+20180106T1833Z+R134bdee starts under Java 1.8.0_151 on Windows 10 amd64 v10.0 at Sat Jan 06 20:14:50 PST 2018```

Panel file with "xml" replaced by "txt" is attached:
[NTraK DPE.txt](https://github.com/JMRI/JMRI/files/1609493/NTraK.DPE.txt)

Attached panel also represents the future work that will be needed to declutter our club Layout Panel for Ops when Layout Editor Improvements go into production, but I digress.

bobjacobsen, 2018-01-07:
Thanks, will take a look.

In the meantime, running jython/TurnOffXmlValidation.py before loading the file should succeed.

bobjacobsen, 2018-01-08:
I'm not seeing a "trainicon" problem with that file.  Instead, I'm seeing a different issue:

NTraK.DPE.txt:3343: element layoutTrackDrawingOptions: Schemas validity error : Element 'layoutTrackDrawingOptions': This element is not expected. Expected is one of ( positionablelabel, positionablepoint, trainicon, turnouticon, sensoricon, memoryicon, memoryInputIcon, signalheadicon, signalmasticon, reportericon ).
NTraK.DPE.txt fails to validate

Working on that.

bobjacobsen, 2018-01-08:
The problem mentioned above (`layoutTrackDrawingOptions` element unexpected) was because this file was written by the LayoutEditor-development builds, and requires that build (or later) to be read.

Once I started using that code, I don't see a validation problem here any more. Is this the right file?

Cliff-Anderson, 2018-01-08:
It appears that I have made an embarrassing beginner's mistake.  As you say, I allowed the LayoutEditor-development build to overwrite the panel file.  Backing up my panel and log files got overlooked too???

My messages.log file is not helping me recover what was done.  

In  spite of my mistake, the session.log text that appears in Yahoo note 144574 does include:
```
2018-01-06 19:17:04,025 configurexml.ErrorHandler ERROR - Parse error while parsing file /C:/Users/Optiplex/JMRI/ERD2017/panels/ERD%20CTC.xml Exception: 
org.jdom2.input.JDOMParseException: Error on line 6587: cvc-complex-type.2.4.a: Invalid content was found starting with element 'trainicon'. One of '{icons, positionablelabel, linkinglabel, positionablepoint, positionableRectangle, positionableRoundRect, positionableCircle, positionableEllipse, positionablePolygon, turnouticon, sensoricon, memoryicon, memoryInputIcon, memoryComboIcon, signalheadicon, signalmasticon, reportericon, layoutturnout, layoutSlip, multisensoricon, fastclock, locoicon, LocoLabel, indicatortrackicon, slipturnouticon, indicatorturnouticon, PortalIcon, LightIcon}' is expected. 
```
I hope that my mistakes have not buried what was causing his problem.  Indeed, he had several problems and that message thread got very long.

My apologies to all.

dsand47, 2018-01-08:
Cliff,

I was able to revert a test panel:

1. Carefully edit the panel XML file and remove the **layoutTrackDrawingOptions** element and its components from each **LayoutEditor** block.
2. After starting JMRI, use LayoutEditor **Options >> Track Options >> Set Track Line Width...** to set the track widths back to reasonable values for each LE panel.  They have been changed to 1.

Dave Sand

Cliff-Anderson, 2018-01-10:
Mike Carr has sent me a copy of his panel that instigated this Issue.  This time around, I have verified with version 4.10 that there really is a strange event related to a trainicon object description in his Panel file:

```
2018-01-10 09:40:56,844 configurexml.ErrorHandler             ERROR - Parse error while parsing file /C:/Users/cliffaa-rr/JMRI/Config%20Profiles/Mike_Carr/panels/ERD%20CTC.xml Exception: org.jdom2.input.JDOMParseException: Error on line 6587: cvc-complex-type.2.4.a: Invalid content was found starting with element 'trainicon'. One of '{icons, positionablelabel, linkinglabel, positionablepoint, positionableRectangle, positionableRoundRect, positionableCircle, positionableEllipse, positionablePolygon, turnouticon, sensoricon, memoryicon, memoryInputIcon, memoryComboIcon, signalheadicon, signalmasticon, reportericon, layoutturnout, layoutSlip, multisensoricon, fastclock, locoicon, LocoLabel, indicatortrackicon, slipturnouticon, indicatorturnouticon, PortalIcon, LightIcon}' is expected.
```
But, there are several other problems with that panel that I am not qualified to answer.  It uses both Lenz and Acela interfaces and JMRI version 4.10 does not even supply a simulator for Acela so I am at a dead end once I hit the trainicon error.  None of those problems are known to be related to this issue.

Mike's panel file, as a text file, and my session.log file are attached here.

Again my apologies for not being careful enough.  

[ERD CTC,txt](https://github.com/JMRI/JMRI/files/1619985/ERD.CTC.txt)
[session.log](https://github.com/JMRI/JMRI/files/1619987/session.log)

dsand47, 2018-01-10:
The `<trainicon>` element is not included in the **paneleditor.xsd** file.  It is in the **layouteditor.xsd** file.

After renaming the Acela system names to LocoNet system names and changing an icon reference from home:... to an arbitrary icon in program:, it loads.

Dave Sand

Cliff-Anderson, 2018-01-10:
Thank you @dsand47.

All of my attempts to understand what was going wrong were focused on the internal details in the description of the trainicon object and not on the context of which Panel Editor had been selected.  

The panel file was constructed using version 4.4 and it seems unprofitable to search for the reasons that the Operations were allowed to connect to the wrong kind of panel back then.  Mike has assured me that things were working OK until last weekend when he updated to 4.11.1.

If there is now some checking going on when the panel is chosen in the Operations - Setup window it seems minimal.  My limited testing in the last few minutes has allowed me to enter a Control Panel name or even a name of a Panel that does not exist.  Thus far, I have been unable to locate the source code.  

Perhaps a pulldown list of defined LayoutPanels might be worth the effort, but with only one error per decade, maybe it is not worth the effort.  Maybe such a change would entail less effort that has been expended on this goosefeather chase?

Cliff-Anderson, 2018-01-15:
It may be possible that long since obsolete versions of JMRI could have caused the faulty panel xml definition.  Strictly based on the lack of any more reports of such faulty panel, it is not prudent to attempt to retrofit outdated versions.   

Lack of response to my suggestion for a verification of the existence of the named Layout Panel that is entered in the textbox of the Operations - Setup window for the Panel indicates that most users seem not to need such a verification.

It is now past the time to close this Issue.

Thank you for the great program.

Cliff

closed 2018-01-15 by Cliff-Anderson