JMRI/JMRI #4730: Layout Editor: Connecting up edge connector does not show all connectors on other panels

pnbgit, 2018-01-07:
When you select edit link for the first time, the drop down list for blocks to list to only shows edge connectors that are on the opposite end of other panels. IE: If your EC is on the right of the panel, you can only connect it to ECs on the left side of other panels, and vice versa.

This assumption is often incorrect, because if the other panel is to be displayed on a screen the opposite side of the layout, it is quite likely you may want to connetct the right edge of one panel to the right edge of another panel.

The current workaround is to temporarily drag the edge connector to the opposite side of hte panel it is on, connect it up, then drag it back to the right place.

[image]

[image]

pnbgit, 2018-01-11:
This may be related to #4729

pnbgit, 2018-01-16:
Dave has now had a look at this, it seems that it is not the position on the panel that is affecting the choice of what you can link to, but the direction in which the edge connector is facing, so when I dragged it left, it also had the side effect of making it face the opossite dircfdtion.  Now I'm hoping that the direction an edge connector faces does not matter (otherwise other things I have done might well be broken!), so hopefully we can just disable that direction check before offering the user which blocks to connect to.

JDWR, 2018-12-18:
It is now December 18 2018 and I am about to start linking EC's between eight panels where both Single Line working takes place as well as UP/DOWN Main lines. When I set out the EC's originally I started at one Terminus and worked outwards from there in order to maintain the "one direction of travel' - in this case the UP line direction.
 
The EC's numbered automatically sequentially until I got to the next Terminus. I now had to reverse direction until I got to an intermediate Junction where the line split from Single line working to both UP and DOWN line working. 

Again, I kept to one direction only when creating the EC's up through another major Junction to the main station and along the UP Main Line all around the layout. When completed, I then followed the DOWN line all the way back around the layout to the original start point.

I tried to Edit both halves of each EC as I went from Board to Board, Panel to Panel but ran into the 'One Direction Only' problem that has already been highlighted. I found this article whilst trying to find a solution and it appears that I have to complete all EC's in one direction until all are edited. However, this is not possible where we are Single Line working in both directions. My thought is to ignore the connections already made and jump to the new ones that maintain the original UP direction until all EC's in that direction have been connected before starting on the DOWN EC's.

Has any progress in overcoming this problem been made? Guidance would be very much appreciated at present.

Dave

geowar1, 2018-12-19:
Trivial fix for this: change <PositionalPoint.java> line 1271 from:
`                        if (p.getConnect1().getLayoutBlock() != getConnect1().getLayoutBlock() && ourDir != p.getConnect1Dir()) {
`
to: `                        if (p.getConnect1().getLayoutBlock() != getConnect1().getLayoutBlock()) {
`
If someone wants to assign this to me I can fix this now.

geowar1, 2018-12-19:
See PR #6326

geowar1, 2018-12-21:
Maybe I’m missing something… but why do Edge Connectors attach to blocks? Shouldn’t they connect to other Edge Connectors? Blocks may have multiple edge connectors… How does LE know which EC in the block this EC connects to?

geowar1, 2018-12-21:
In any case PR #6326 removes the code that limits the directions allowed for EC connections.

JDWR, 2018-12-21:
A valid point George especially if the Blocks go from one end of a board to the other.

This whole EC thing is being looked at as I type this so hopefully we will get a resolution very soon. 

In an attempt to avoid the situation you mention I ensure that all EC’s are part of a different block name, even if it a short one. It is the ability to handle multiple EC’s coming into or leaving a Board from any of its sides that is causing a problem at the moment.

Dave

KenC57, 2018-12-21:
My problem with EC's has been that it required being a block boundary.
Generally you liked to know the status coming into or out of a pane. But the
worst case was the classic passing track configuration going into a single
track, which is a single block, to the next passing track configuration.
Ideal would be each pane shows its passing track with the turnout at the end
leading off the pane. But last time I tried to us EC's, it would only work
with the turnout being on one pane or the other.

If you had a track with three towns, each town has single track coming in,
double track through town, then single track leaving for the next town, how
could you do this with EC's and have each town as a separate pane showing
the meaningful track for that town?

To me, it had always seems more sensible that an EC would be inside a block
so both panes can show it 'coming and going' if you get my drift. But I've
not worked with it much because I couldn't get anywhere with my simple test
above. I suspect somebody else can explain how to pull this off, but my try
failed.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

dsand47, 2018-12-22:
I have been working with the panels uploaded by Dave Roberts.

If a track segment is split and a segment is deleted, the anchor point at the split may be yellow (valid) or red (not valid).  If a red anchor point is converted to an edge connector, it will not be included in the link selection list.

Start:  ap------ap
Two splits:  ap--ap--ap--ap
Remove center:  ap--ap(y)   (r)ap--ap
Convert to EC:  ap--ec(y)  (r)ec--ap

 I have not determined the actual issue yet but it appears that if **connection1** remains, the anchor point is yellow but if **connection2** remains, it is red.

To fix the red ap/ec, delete and add the track segment.  That changes the connection type to **connection1**.

geowar1, 2018-12-22:
What should happen when connection1 is deleted and connection2 isn’t null… swap connections 1 & 2 so connection 1 is the connection and 2 is null.

geowar1, 2018-12-22:
On Dec 22, 2018, at 10:22 AM, Dave Sand <notifications@github.com> wrote:
> The connection swap might be the easiest. There might be other situations where a null connection1 has an adverse impact similar to the edge connector link list.

See PR #6345

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

dsand47, 2018-12-22:
Preliminary testing of #6345 looks promising.

JDWR, 2018-12-22:
Dave,

Success! After a fashion. I have followed your suggestions to fix the RED connectors and edited all of the 38 EC’s on the individual location panels that will be used by operators and all appear to be working.

However, on the Whole Layout Panel, all of the EC’s between the Power Districts/Blocks remain Yellow as though defining the extent of the location Panels. Is this what you would normally expect?

It looks ok to me since the operators will normally be looking after their own location. It is only those of us who choose to follow a train on its journey around the layout that will need to look at the whole layout plan.

I think that a NEW Tool would come in very handy that will allow the user to Double Click on any Anchor Point and allow it to Disconnect at that Anchor Point to allow us to split the Track Segment at that location to insert something else - like EC’s - and avoid the problems we create by Splitting Blocks and removing the wrong bit!

Dave

dsand47, 2018-12-22:
Since the **South Devon Railways** and the **SDR Main Lines** panels do not participate in the edge connections, I would change the edge connectors on those two panels to anchor points, connect each pair with a track segment and set the block.  If you want a visual indication of the location panel boundaries, set the new track segment as hidden.

JDWR, 2018-12-23:
Dave and George,

Thanks for your insight on the EC's. I will change them later on today but then there appears to be another problem that has shown itself! 

The Power Districts within a layout may cross over several boundaries as in my case with the SDR Main Lines shown as a single panel for convenience. In this case we will have two PD’s - UP Line and DOWN Line all around the layout. A similar breakdown occurs within every panel where the Main Running Lines are separated out from the Sidings on that panel to form another sub-PD. The idea being that we do not want the Main Lines affected by operations in the sidings where derailments are more likely to occur.

I think that George has spotted this problem too. If my suggested ‘Disconnect' at Anchor Points is technically a non-starter and we opt for the method you suggest of automatically creating two breaks and delete the middle segment and convert the remaining AP’s into EC’s that will work but how do we address the PD problem which may span several Panels as it does in my case?

The Whole Layout Plan is primarily there for viewers to look at the whole picture as operations unfold.

Don’t you just love it when a plan falls apart!

Merry Christmas everybody!

Dave

JDWR, 2018-12-23:
Dave,

I have successfully changed the "SDR Main Lines” panel to replace the EC’s with AP’s and a new, hidden Track Segment, to show the break, as you suggested. I have saved it as "SDR Main Lines Connected” so as to retain the original format.

Several errors occurred whilst changing over every EC because the New connections are no longer as those in the Whole Layout panel "South Devon Railways” and the program was telling me so.

The Location Panels are all correct and working correctly but I am concerned that if I go ahead and make the changes to the Whole Layout Panel this will corrupt all of the location panels that were originally created from this one panel?

I would appreciate your thoughts on this matter before I proceed, please.

Dave

dsand47, 2018-12-24:
Dave,

Do you remember what the actual errors were?  If you upload the latest panel XML file, I can take a look at it.

Dave Sand

JDWR, 2018-12-24:
Dave,
The error messages were shown as “Inconsistencies” between the South Devon Railways whole layout panel, the SDR Main Lines panel and the appropriate location panel as I changed the connections around the layout.
I was expecting something like that between the SDR Main Lines panel as you suggested because I had not linked to that panel.
However, I was not expecting errors between the location panels and the whole layout panel. All location panels came from the one so any changes to any of the location should amend the whole layout panel as well because the share the same xml file.
The various location panels show the EC connections as GREEN made correctly whereas the whole layout panel show them YELLOW but they are consistent.
Dave

JDWR, 2018-12-24:
Dave,

I was out and about when your email came. I am home now and have opened up the South Devon Railways file. It is showing “Connectivities are not compatible" for the various Blocks in that panel as well as the Location Panels.

I am still working my way through the know errors between the various Location Panels and the whole layout panel, one at a time. I expect this to remove many small errors as I go through the process but it is these EC’s that are causing the major problems at present because they are Inconsistent - one is an EC on the location Panels and the South Devon Railways Whole layout Panel but they are an AP and Track Segment on the SDR Main Lines panel.

That is why I am reluctant to alter the South Devon Railways whole layout panel because these are showing as EC’s but in YELLOW instead of GREEN. The connectors should have been updated the xml file as I created/changed the EC’s in the location panels and the Whole Layout Panel when linking them up since they are all the same xml file?

It would appear that your suggestion has created the inconsistencies?

I am going to plough on and remove all of the other errors showing between the Location Panels and the Whole Layout Panel first, one by one and then come back to this EC problem after the break.

Dave

dsand47, 2018-12-24:
Dave,

The bottom line is that edge connectors for the location panels was the wrong solution.  I was focused on the edge connector issues and did not understand the actual requirements.  I am sorry for leading you down the wrong path.

What you have is a master panel, South Devon Railways, and sub-panels consisting of the SDR Main Lines panel and the location panels.

It appears that the sub panels where creating by copy the main panel definition in the XML.  This can lead to confusion by assuming that the panel components, such as anchor points, track segments and edge connectors are shared.  They are not.  Some items, such as sensor icons, turnout icons, etc., point to the same table items, but the icon definition itself is unique.

The important part of creating sub-panels is that the block assignments correspond to the block assignments in the master panel.  This includes having the same turnouts included in the blocks.  The block topology is independent of the panels but requires that the panels are consistent.

Dave Sand

JDWR, 2018-12-25:
Dave,

I have deleted the SDR Main Lines with the connections and reverted back to the original file. Error correction is going well between the Master Panel and the Sub-Panels. Most have been caused by me renaming odd Blocks for future use.

I know why the confusion arose. Originally, it is because of the insistance that there can only be ONE xml file. Back then, I had no clear idea of how the layout would develop so I followed normal build procedure starting at one location and created the panel for that location. As the layout expanded so new panels were added under that locations name - I ended up with more that one xml file so back to the drawing board!

It soon became clear that what was expected was the ‘Crystal Ball’ look into the future to develop the whole layout - the Master Panel and then Re-Title the Master Panel into the Sub-Panels used on the layout. The track not required on that sub-panel was deleted from the plan but the underlying ONE xml File for the whole layout kept, as you say.

The size of the Master Panel quickly became unweildy and difficult to work with and my thought was that any subsequent changes to anything on any sub-panel would be saved to the one file which described the whole layout and this appears to be working with the Turnouts because I have both the Master Panel and the one Sub-Panel that I am working on side by side. Any changes to a turnout on either panel is immediately shown on the other panel.

When I have completed this careful look at any errors that show up in the “Check” sub-panels I will then need to address why the EC’s on the Sub-Panels are correctly shown as being connected and GREEN but the same EC’s on the Master Panel are shown in YELLOW but show as connected and that is weird! I am hoping that as I clear all the errors they will revert to GREEN on the Master Panel too. We shall see.

Dave

geowar1, 2019-11-04:
What's the status of this? Can it be closed?

dsand47, 2019-11-04:
The original issue has been resolved.  The edge connector orientation is no longer relevant.

Removing the block boundary requirement requires a complete redesign of the edge connector capability and is outside the scope of this issue.

I recommend that it be closed.

closed 2019-11-05 by bobjacobsen