JMRI/JMRI #4738: AccessoryOpsModeProgrammer fails for some systems due no packet repeat.

dheap, 2018-01-09:


dheap, 2018-01-09:
For background to this and detailed findings, see this [jmriusers conversation](https://groups.yahoo.com/neo/groups/jmriusers/conversations/topics/144483).

devel-bobm, 2018-01-09:
This reply contains both "fact" and "opinion, and is a LocoNet-centric take on the issue.  Fact first, then opinion.

In JMRI, `loconet.SlotManager.sendPacket()` supports a "repeat count" of 0 to 7, and issues an error if the repeat count is greater than 7, without sending the packet to the DCC track signal.

This is likely because LocoNet messaging allows for a single LocoNet message to send a single NMRA packet to the DCC track signal between one and 8 times (if I interpret the LocoNet PE Spec correctly).  The LocoNet message (`OPC_IMM_PACKET`) has a 3-bit parameter (`CNT`) which  defines the "repeat count".  Presumably, `CNT` = 000b to indicate "send the packet a total of one time", and `CNT`=111b to indicate "send the packet a total of 8 times".

Making `loconet.SlotManger.sendPacket` allow for a maximum repeat count that is greater than 7 may not be easy, depending on how the command station firmware is implemented, and has an awkward and annoying implication.

It is known that the command station "buffers" any LocoNet operation which has an impact on the DCC track signals, and this buffer is finite.  "Standard" Mobile-decoder operations are buffered by the "slots".  Accessory Decoder and `OPC_IMM_PACKET` messages are separately buffered.  The buffer has limited size, and can overflow easily.  This is why "flashing" signals can be awkward with the SE8C or any "turnout-controlled" signal head when using a Digitrax (or LocoNet-based?) command station.  The "flash rate" will vary considerably depending on how many signal heads are flashing and how much the "playable whistle" feature is used by the various locomotives. 

When the command station buffer overflows, it signals to the sending device that the buffer is full.  Most LocoNet devices seem not to pay attention to this overflow condition.  So it falls to JMRI to provide the "overflow-protection".  And thus we get a flood of LocoNet traffic generated by JMRI which attempts to retry the offending turnout control message, which prevents other LocoNet devices from sending anything until the offending turnout control message gets thru.

Adding "OPC_IMM_PACKET" messages to the list of things JMRI must protect is going to increase the incidence of JMRI flooding LocoNet.  Especially when a user is working with the "playable whistle" feature, which also uses "OPC_IMM_PACKET".

It might be better to simply implement a system-specific "max" for the retry count, and emit a log.warn(), when appropriate, saying that the requested retry count was truncated to 7 retries max (for a total of 8 sends).  

Perhaps this last bit ought to be done instead of the log.error() which SlotManager.sendPacket() currently implements, regardless of the outcome of this issue.

Opinions follow:  
If an accessory decoder is "more reliable" with 8 or 9 retries, rather than 7 retries, then there is a basic problem in the delivery mechanism or the accessory decoder implementaiton.  Requiring the protocol to "make up for the deficiencies" of the delivery mechanism or accessory decoder implementation is a poor workaround.

It has always been my opinion that DCC-track-signal-connected accessory decoders aren't practical with Digitrax systems, and probably most other systems.  A short to the track signal prevents the signal getting thru to the track-connected accessory decoder (true of any DCC system).  That's REALLY bad for a signalling system, which ought to be treated as a fault-hardened safety appliance, rather than an implementation that might haphazardly miss signal aspect changes.

n3ix, 2018-01-09:
As a follow-on to Bob’s comments, would like to add that there are Loconet-connected stationary decoders for signaling that directly decode the DCC Signal Mast messages from the Loconet OPC_IMM_PACKET message.  When using those products (such as my QuadLN) for signaling the users likely have no DCC-connected devices listening for signaling messages and thus there would be no reason to require the command station send multiple copies of the signaling-related OPC_IMM_PACKET messages on DCC.  Wouldn’t want to up with multiple DCC copies of these messages by default when only Loconet devices are listening to them!  Thanks.

 

Robin

 

Robin Becker

San Diego, CA

pabender, 2018-01-09:
Sent from my iPad

> On Jan 9, 2018, at 12:31 PM, Bob Milhaupt <notifications@github.com> wrote:
> It might be better to simply implement a system-specific "max" for the retry count, and emit a log.warn(), when appropriate, saying that the requested retry count was truncated to 7 retries max (for a total of 8 sends).
> Perhaps this last bit ought to be done instead of the log.error() which SlotManager.sendPacket() currently implements, regardless of the outcome of this issue.
> 

It might be best to have the method throw an IllegalArgumentException, and then let the calling code deal with it.  

That let's the UI deal with this when the user is using the send packet tool, but other code can use that to adjust the parameters and resend.

> It has always been my opinion that DCC-track-signal-connected accessory decoders aren't practical with Digitrax systems, and probably most other systems. A short to the track signal prevents the [...]
Yes, it is a bad idea to connect accessory decoders to track blocks for all the reasons you say.  That is also the reason most of us who use "track connected" accessory decoders use a separate accessory bus for just that purpose.

I have two circuit breakers feed by the booster.  One circuit breaker feeds the track, the other feeds the accessory bus.  If this were a static home layout, I would could use one circuit breaker, with the accessory bus wyed out of the booster directly between the booster and the breaker.  ( you can overload the accessory bus, but that just means you need to break it into smaller pieces... )

dheap, 2018-01-09:
I'm not worried about the cases that set a reasonable upper limit on the repeat count.

My main concern is the systems that always only send one packet. This prevents accessory POM because two packets are required by S9.2.1. I assume these systems follow a different path (not raw packets) for mobile POM.

bobjacobsen, 2018-01-09:
- I have copies of the internal code for Accessory Decoders from three DCC manufacturers. None insist on a second packet. 
- The decoder reporting the issue is homemade. 
- AFAIK, there have been no reports of problems with commercial decoders. 
- The NMRA documents are, at best, ambiguous. They’re not really a guide to what’s done in practice. 

How big an issue is this?

dheap, 2018-01-10:
I'll do some tests with the real world decoders I have, but I need to do some real-life responsibilities first.

I see the decision by any manufacturer (home made or commercial) of interpreting the NMRA Standards as requiring a send confirming identical packet before making a CV change as reasonable and we shouldn't lock them out.

dheap, 2018-01-10:
I've taken a closer look at the code and I can't see a LocoNet problem.

If I change line 173 of AccessoryOpsModeProgrammerFacade from:
`        InstanceManager.getDefault(CommandStation.class).sendPacket(b, 1); // send packet`
to:
`        InstanceManager.getDefault(CommandStation.class).sendPacket(b, 2); // send packet`

nothing significant will change in LocoNet.

The only thing that will happen is that when a LocoNet system user Programs an accessory/signal CV value on the Main (using Single CV Programmer or a decoder definition) two raw DCC packet will be sent for each CV changed. This is exactly what already happens with mobile(loco) CV Programming on Main.

bobjacobsen, 2018-01-10:
> I see the decision by any manufacturer (home made or commercial) of interpreting the NMRA Standards as requiring a send confirming identical packet before making a CV change as reasonable and we shouldn't lock them out.

Perhaps.  I’m certainly not going to argue that the NMRA documents are unambiguous. But DCC bandwidth is very limited.  Two packets, no problem.  Lots more out a sense of caution might have more cost than benefit.

devel-bobm, 2018-01-10:
I have found contradictory behavior of LocoNet.

This was confirmed by monitoring the DCC track signal using a PRICOM "DCC Pocket Tester" with a DCS240, a DCS100, and a DB150.

I have confirmed that with the DCS100 and DCS240, the LocoNet message "repeat" coding is number of "additional" sends of the "immediate packet" after the initial attempt requested in the LocoNet message.  In other words, if <RPT> is 0, then the packet will show up on the DCC track signal exactly once.  If <RPT> is 7, then the packet will show up on the DCC track signal exactly 8 times.

I have confirmed that with the DB150, the LocoNet message "repeat" coding is odd.  Sometimes, the PRICOMM shows that the DCC packet shows up on LocoNet only once, regardless of "repeat" value.  Sometimes the DB150 seems to honor the repeat count by repeating the DCC packet the requested number of times (i.e. original send plus repeat count in range 0 to 7).  It is posisble that this is a Procomm Pocket Tester issue, and it is possible that the tester is just fine and is exposing a randomly variable behavior of the DB150.

(The usual caveat about firmware applies - there are many firmware versions for each Digitrax product, and a given firmware version can behave differently than other versions.)

dheap, 2018-01-10:
Given that changing it from one to two packets only increases traffic by one packet per accessory CV change (not an everyday occurrence) and nothing else, I see no harm in increasing the packet count in AccessoryOpsModeProgrammerFacade to two.

dheap, 2018-01-10:
> @devel-bobm I have found contradictory behavior of LocoNet...

How do you see this impacting on the current discussion?

Are you able to test what happens if you use Single CV Programmer to write one CV value using  Ops Accessory Byte mode after changing ?

Are you able to also test what happens if you use Single CV Programmer to write one CV value using  Ops Accessory Byte mode after line 173 of AccessoryOpsModeProgrammerFacade from 
`sendPacket(b, 1)` to `sendPacket(b, 2)`?

devel-bobm, 2018-01-10:
Is the second parameter for `sendPacket()` intended to be a "retry" count or a "try" count? Passing "2" to `loconet.SlotManager.sendPacket()` results in three packets on the track signal on (see possible DB150 exception I've noted previously).

devel-bobm, 2018-01-10:
@dheap,

With `sendPacket(b, 1);` in an un-modified `AccessoryOpsModeProgrammerFacade.java` :

When I use the Single CV programmer to write one CV value using Ops Accessory Byte mode, 
the DCS100 gives two packets every time and the DCS240 gives two packets every time.  it appears that the DB150 sometimes sends one write and sometimes sends two writes.

devel-bobm, 2018-01-10:
@dheap,

With sendPacket(b, 2); in an locally-modified AccessoryOpsModeProgrammerFacade.java :

When I use the Single CV programmer to write one CV value using Ops Accessory Byte mode,
the DCS100 gives three packets every time and the DCS240 gives three packets every time. it appears that the DB150 sometimes sends one write and sometimes sends three writes, randomly.

dheap, 2018-01-10:
I've always interpreted the JavaDoc for `CommandStation.sendPacket(byte[] packet, int repeats)` to be a "send" count. Lenz, DCC++ and EasyDCC (and possibly others) interpret it this way.

But I could see an argument for `send = repeats + 1.`

I've never though of interpreting "repeats" as "retries".

dheap, 2018-01-10:
NCE Power Pro ignores "repeats" and always sends two packets.

SPROG Command Station ignores "repeats" and always sends one packet.

KenC57, 2018-01-10:
Bob M,

Another test would be the NCE packet monitor. It would hopefully match what
the Pricom shows. We've used that for a number of track packet decoding
issues. But it is now something many people have. I would also say that when
we have done any testing, we didn't have locos on the tracks in case that
might mess up any of the output signal.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

devel-bobm, 2018-01-10:
@KenC57,  I happen to have one of those NCE packet monitors.  I just don't have a suitable handy RS232-to-USB converter...  I'll add it to the list of things to try as verification of the DB150 variability.

devel-bobm, 2018-01-10:
I have two branches in my github repository which address a couple of options for LocoNet mentioned in earlier comments:

- dbm-OpsAccessoryPacket - when repeat count is "out of range" logs a warning and forces the repeat count to be "in range"; when the packet length is "out of range", logs an error.

and

- dbm-OpsAccessPktIllegalArg - throw an illegal argument exception when the "packet" length is out of range or when the "repeat" count is out of range.

I see these as mutually-exclusive solutions.

Both would need to be re-worked if it is agreed that the second parameter of sendPacket() is a "packets count" rather than the currently-implemented "retries count beyond the initial try".

dheap, 2018-01-11:
> @KenC57 Another test would be the NCE packet monitor. It would hopefully match what the Pricom shows. We've used that for a number of track packet decoding issues.

Unfortunately bot these products are no longer available. I have the NCE one and use it extensively.

dheap, 2018-01-11:
> @devel-bobm Both would need to be re-worked if it is agreed that the second parameter of sendPacket() is a "packets count" rather than the currently-implemented "retries count beyond the initial try".

Either of these obviously make sense for a LocoNet system. Some systems have no failure detection mechanism so have to either interpret the parameter as a "packets count" or make their own judgement about the number that should be sent.

Lenz, DCC++ and EasyDCC currently interpret "repeats" as a "packets count".
NCE and SPROG currently ignore "repeats" and make their own decision.

devel-bobm, 2018-01-11:
@dheap - I guess for "consistency's sake", the LocoNet implementation will change to match Lenz, DCC++, and EasyDCC.  I'll work on that when we get a decision on whether to implement exceptions or to simply log warnings and "make it work as best as possible, or do nothing".

dheap, 2018-01-11:
@devel-bobm - My personal preference (I'm not a Loconet user) would be a third option:

"Make it work as best as possible (with a warning - packet count out of range), or throw an exception ("packet" length unacceptable)."

dheap, 2018-01-11:
As an FYI, sniffing  the NCE Power Pro system when using Pro Cab throttle keypad:
- Ops Prog Mobile CV sends 8 packets.
- Ops Prog Accessory CV sends 8 packets (4 Legacy and 4 Basic Accessory).
- Ops Prog Signal CV sends 4 packets (Extended Accessory).

When we use the JMRI interface to implement the above operations, we need to use NCE's  "Send Raw Packet to TEMP_Q" commands and explicitly specify a send count (up to 254 max).

closed 2020-02-10 by dheap