JMRI/JMRI #4756: Version 4.10 in tools cannot open any table.

1ngwpower, 2018-01-13:
Computer crashed and now cannot get into the table functions to enter new turnouts, sensors ... none of the listed operations will open? Have uninstalled JMRI reinstalled from web site and still cannot use tables.
[image]

rhwood, 2018-01-14:
This is a known issue fixed in JMRI version 4.11.1. If you do not want to upgrade to that version, search the JMRI Users group for “jmri.conf” and “JOAL” for a manual workaround.

rhwood, 2018-01-14:
See #4599 for the workaround for version 4.10.

1ngwpower, 2018-01-15:
Thank You rhwood!

1ngwpower, 2018-01-15:
Sure grateful you good people know about these things!

closed 2018-01-15 by 1ngwpower

1ngwpower, 2018-01-15:
Thank You! Have a safe week!
RJ