JMRI/JMRI #4766: Default visible columns in Roster Tables are not persistently hidden

rhwood, 2018-01-15:
Although users can set a column that is visible by default in the Roster Table to be hidden, that setting is not persistent across application restarts. All other table settings (column (re)ordering, sorting, showing columns hidden by default) are persistent.

rhwood, 2018-01-15:
The problem turns out to be that hidden column states are not persisted at all for any table.

bobjacobsen, 2018-01-15:
I think there might be something subtle going on, because it seems to work for some people and not others.  

For example, I created a new "Fascinating Fact-1" column a long time ago. It's not displaying. I tell it to display and move it to the middle, then quit. Comes back present, but at the right side.

So here the column's visibility is persisting, but not it's position.

Open LocoNet Monitor (this is a LocoNet simulator connection), move window, quit, restart

rhwood, 2018-02-04:
I think I have found a problem in the general persistence model. The model only works reliably if every table column has a unique identity, which is used to store the column data (width, position, etc), but we have not been requiring uniqueness of identities. If identities are not explicitly set for each column, the identity is matched to the column's header, but not all column headers are set or, if set, have unique titles, so we need someway of populating the identities early.

bobjacobsen, 2018-02-04:
From the above, it sounds like we need to do some work on the tables to ensure that columns have a unique identity.  Is there somewhere that I can find more information on what that means?  I.e. what's the "identity", what uniqueness scope is needed (in table, in table type, globally?), etc.

rhwood, 2018-02-04:
The "identity" is an arbitrary object provided [TableColumn.getIdentifier()](https://docs.oracle.com/javase/8/docs/api/javax/swing/table/TableColumn.html#getIdentifier--). I am working on something that at least ensures uniqueness in the BeanTableDataModel subclasses to see if my understanding is correct or not.

PR #4767 (2018-02-11): Hide Table Columns Persistently
Fixes #4766
closed 2018-02-11 by rhwood