JMRI/JMRI #4776: Excessive Error Logging in UsbUtil

dheap, 2018-01-17:
Simply selecting Debug->USB Devices produces this [session.log](https://github.com/JMRI/JMRI/files/1637872/session.log).

Great tool but too noisy.

rhwood, 2018-01-23:
@dheap What is the device that the USB utilities can't read? I ask, because I don't get those logs (with three devices connected via a USB 3 hub), but your logs indicate some USB device is not working correctly.

dheap, 2018-01-23:
I'll take another look. I'm pretty sure I tried on more than one Mac (with completely different USB devices).

dheap, 2018-01-23:
@rhwood Two different Macs, with non-overlapping USB peripherals:

[Mac Pro session.log](https://github.com/JMRI/JMRI/files/1657295/Mac.Pro.session.log)
[Mac Pro USB Sys Info.pdf](https://github.com/JMRI/JMRI/files/1657298/Mac.Pro.USB.Sys.Info.pdf)

[iMac session.log](https://github.com/JMRI/JMRI/files/1657300/iMac.session.log)
[iMac USB Sys Info.pdf](https://github.com/JMRI/JMRI/files/1657301/iMac.USB.Sys.Info.pdf)

dheap, 2018-01-23:
@rhwood Another set:

[Win7 VM session.log](https://github.com/JMRI/JMRI/files/1657384/Win7.VM.session.log)
[Win7 VM USBDeview.txt](https://github.com/JMRI/JMRI/files/1657385/Win7.VM.USBDeview.txt)

dheap, 2018-01-24:
@rhwood With an FTDI cable connected:

[Win7 VM session.log](https://github.com/JMRI/JMRI/files/1658312/Win7.VM.session.log)
[Win7 VM USBDeview.txt](https://github.com/JMRI/JMRI/files/1658313/Win7.VM.USBDeview.txt)

geowar1, 2018-01-24:
On Jan 16, 2018, at 10:04 PM, Dave Heap <notifications@github.com> wrote:
> Simply selecting Debug->USB Devices produces this session.log <https://github.com/JMRI/JMRI/files/1637872/session.log>.
> Great tool but too noisy.
> 
The relevant error is (repeated many time with stack track):

javax.usb.UsbPlatformException: USB error 99: Unable to get string descriptor languages: Other error

The issue is that usblib can't read USB device descriptor strings unless it can open the device… and Window's won't let it.

The UsbPlatformException should be caught and a null or empty string returned for the descriptor strings.

(As in the UsbUtil.java methods to retrieve the device descriptor strings.)

-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

dheap, 2018-01-24:
> @geowar1 The UsbPlatformException should be caught and a null or empty string returned for the descriptor strings.

Catching the UsbPlatformException and returning an empty string would have been my course of action but I didn't wrote that code so wanted to give the original author(s) the chance to comment/run with it.

rhwood, 2018-01-24:
@dheap Do you have some other application that directly accesses the USB running, or is the user you ran these as not an administrator on your systems? (At this point, I'm trying to understand what is preventing JMRI on your Macs from querying any USB device.)

geowar1, 2018-01-24:
UsbPlatformException should be added to the exceptions caught in UsbUtil.getFullProductName. (As the author I should fix that… tomorrow...). It should also be added to the javadocs for the API that’s throwing it.
Beyond that I’d also like to know why it’s being thrown.

dheap, 2018-01-24:
> @rhwood  Do you have some other application that directly accesses the USB running, or is the user you ran these as not an administrator on your systems? (At this point, I'm trying to understand what is preventing JMRI on your Macs from querying any USB device.)

Nothing else that I am aware of. Obviously some drives are mounted, CrashPlan service is running, likewise cupsd etc.

 User is certainly an administrator.

The the USB Devices result Window is not empty. All computers off now (storms) but can get results tomorrow morning.

geowar1, 2018-01-24:
On Jan 24, 2018, at 2:49 AM, George Warner <notifications@github.com> wrote:
> UsbPlatformException should be added to the exceptions caught in UsbUtil.getFullProductName. (As the author I should fix that… tomorrow...). It should also be added to the javadocs for the API that’s throwing it.

When I tried to add UsbPlatformException NetBeans reported that it was a subclass of UsbException which was already being caught...

        } catch (UsbException | UnsupportedEncodingException ex) {
            log.error("Unable to read data from {}", usbDevice, ex);

So it's the log.error line that's being noisy. Should that be changed to log.debug instead?

And should this routine return null for missing strings? Or an empty string?

Inquiring minds want to know…
-- 
Enjoy,
George Warner
eMail, Message, or FaceTime: <geowar1@mac.com>

dheap, 2018-01-24:
Some devices are readable:
[iMac USB Sys Info.pdf](https://github.com/JMRI/JMRI/files/1661815/iMac.USB.Sys.Info.pdf)
[image]

geowar1, 2018-01-24:
Which ones aren't?

dheap, 2018-01-24:
@geowar1 Which ones aren't?

Compare what shows in the PDF output from System Information.

dheap, 2018-01-24:
Likewise from another Mac:

[Mac Pro USB Sys Info.pdf](https://github.com/JMRI/JMRI/files/1661924/Mac.Pro.USB.Sys.Info.pdf)

[image]

rhwood, 2020-01-12:
I noticed this logging also occurs when the device is removed.

closed 2021-04-14 by stale[bot]