JMRI/JMRI #4788: releasing loconet throttle causes NoSuchMethodError

SteveRawlinson, 2018-01-19:
I've just done a pull from master. Loconet throttle acquisition looks to be improved. However I am now seeing this every time I release a throttle:

UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NoSuchMethodError: jmri.ThrottleManager.removeListener(Ljmri/DccLocoAddress;Ljava/beans/PropertyChangeListener;)V
	at jmri.jmrit.throttle.ThrottlesTableModel.notifyAddressReleased(ThrottlesTableModel.java:55)
	at jmri.jmrit.throttle.AddressPanel.notifyListenersOfThrottleRelease(AddressPanel.java:566)
	at jmri.jmrit.throttle.AddressPanel.notifyThrottleDisposed(AddressPanel.java:279)
	at jmri.jmrit.throttle.AddressPanel.releaseAddress(AddressPanel.java:556)
	at jmri.jmrit.throttle.AddressPanel$5.actionPerformed(AddressPanel.java:399)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	
(I'm sure I'm teaching my granny to suck eggs here but I thought I'd do a bit of investigation before submitting a new issue. Humour me.)

By modifying the code slightly:

    ThrottleManager tm = jmri.InstanceManager.throttleManagerInstance();
    tm.removeListener(la, this);

I've established that tm is a LnThrottleManager instance. LnThrottleManager extends AbstractThrottleManager which implements removeListener() however the argument signature doesn't look right. 'la' is an address but 'this' appears to be a ThrottlesTableModel, not a listener.

That might all be gibberish, sorry.

I'm using a DCS50 connected via LocoBufferUSB . Let me know what more information you need.

bobjacobsen, 2018-01-20:
NoSuchMethodError is often due to partial compilation.  Please try a "clean build". If you're working from the command line, do:
```
ant clean
```
and then compile and build.  If using an IDE, look for a button with a broom or similar symbol on it.

SteveRawlinson, 2018-01-20:
That fixed it. Thanks.

closed 2018-01-20 by SteveRawlinson