JMRI/JMRI #4789: Expected behavior of SystemConnectionMemo.get() when memo doesn't provide a type?

pabender, 2018-01-19:
This is a question that comes about because of the tests I added in PR #4785 to the base test class for SystemConnectionMemo objects.

What is the expected behavior of SystemConnectionMemo.get(type) when SystemConnectionMemo.provides(type) returns false?  

My expectation going into this was that the get call would return a null object, but that isn't the case for all systems, and I'm even to blame for some of that (though not all of it...).  

The issue really comes about for systems that either are 1) disabled or 2) sometimes provide an object and other times don't (e.g, Lenz Compact and Atlas Commander systems don't support programming through the XPressnet, so in that case, so we don't provide a programmer when those systems are connected as the command station, but do with other command stations..)

What I have as the test right now is:
```
public void getTest(Class t){
      Assume.assumeTrue(scm.provides(t));
      // if the manager reports providing the class, make sure it exists.
      Assert.assertNotNull("Provides Class " + t.getName() ,scm.get(t));
}

```

but I would like to change that to:

```
public void getTest(Class t){
      if(scm.provides(t)) {
        // if the manager reports providing the class, make sure it exists.
        Assert.assertNotNull("Provides Class " + t.getName() ,scm.get(t));
     } else {
        Assert.assertNull("Provides Class " + t.getName() ,scm.get(t));
     }
}
```

The else part fails on multiple systems currently however, so the behavior of those systems would have to change.

Does anyone have any thoughts on this before i proceed?

rhwood, 2018-01-19:
I would expect get(T) to be null if provides(T) is false, but it also appears to be completely undefined, so I would first define the expected behavior (in the Javadocs for SystemConnectionMemo) and then change any tests that need to be to adhere to that. Absent changing the defined behavior in documentation, I would make no assumptions in the tests.

Randall Wood
JMRI Developer - http://jmri.org
Alexandria Software - http://alexandriasoftware.com
Northern Virginia NTRAK - http://nvntrak.org

bobjacobsen, 2018-01-20:
I mostly agree with Randall on this.

The provides(T) call is somewhat redundant:  if get(T) reliably returned null when a _working_ T can’t be provided, then provides(T) can be replaced with get(T)!=null and we’d have less code to keep consistent.

In the meantime, I think defining the behavior as your second case is the right approach.

Bob

> 
> I would expect get(T) to be null if provides(T) is false, but it also appears to be completely undefined, so I would first define the expected behavior (in the Javadocs for SystemConnectionMemo) and then change any tests that need to be to adhere to that. Absent changing the defined behavior in documentation, I would make no assumptions in the tests.
> 
> Randall Wood
> JMRI Developer - http://jmri.org
> Alexandria Software - http://alexandriasoftware.com
> Northern Virginia NTRAK - http://nvntrak.org
> 
> 
> 
> 
> > 
> > This is a question that comes about because of the tests I added in PR #4785 <https://github.com/JMRI/JMRI/pull/4785> to the base test class for SystemConnectionMemo objects.
> > 
> > What is the expected behavior of SystemConnectionMemo.get(type) when SystemConnectionMemo.provides(type) returns false?
> > 
> > My expectation going into this was that the get call would return a null object, but that isn't the case for all systems, and I'm even to blame for some of that (though not all of it...).
> > 
> > The issue really comes about for systems that either are 1) disabled or 2) sometimes provide an object and other times don't (e.g, Lenz Compact and Atlas Commander systems don't support programming through the XPressnet, so in that case, so we don't provide a programmer when those systems are connected as the command station, but do with other command stations..)
> > 
> > What I have as the test right now is:
> > 
> > public void getTest(Class t){
> > Assume.assumeTrue(scm.provides(t));
> > // if the manager reports providing the class, make sure it exists.
> > Assert.assertNotNull("Provides Class " + t.getName() ,scm.get(t));
> > }
> > 
> > but I would like to change that to:
> > 
> > public void getTest(Class t){
> > if(scm.provides(t)) {
> > // if the manager reports providing the class, make sure it exists.
> > Assert.assertNotNull("Provides Class " + t.getName() ,scm.get(t));
> > } else {
> > Assert.assertNull("Provides Class " + t.getName() ,scm.get(t));
> > }
> > }
> > The else part fails on multiple systems currently however, so the behavior of those systems would have to change.
> > 
> > Does anyone have any thoughts on this before i proceed?

pabender, 2018-01-23:
Closed via #4809

closed 2018-01-23 by pabender