JMRI/JMRI #4803: Need help in new decoder definition

moba-klytos, 2018-01-21:
Hi,

I have defined a new decoder based on an existing one. For some reason I have a problem introducing a new parameter. The parameter is defined as

`<variable CV="124" default="3" item="EMF Optimization">
        <decVal max="255"/>
        <label>BEMF Optimization (1..15)</label>
        <label xml:lang="de">Lastregeloptimierung (1..15)</label>
      </variable>`

in my branch moba-klytos/JMRI
JMRI/xml/decoders/TAMS_LD-G-32.2.xml

The parameter is not shown in Decoder Pro and I am lost in how to get further. Anyone can toss me in the right direction?

moba-klytos(@)flying solder.net

/Uwe

silverailscolo, 2018-01-21:
Hi Uwe,
I notice you are working in the ``master`` branch. That's not advised. Better create a working branch from master, name it ``mk-tamsldg322`` or whatever works for you (but keep it short), then switch to that branch in your app and create and work on your xml file in that branch. In case things get messed up, you still have your master intact!
As a next step, run the file through the ``Validate XML file...`` tool in DecoderPro and report on the results here.

silverailscolo, 2018-01-21:
One more note: Because your new item "EMF Optimization" is not included in the template for the Motor tab sheet ("pane"), it will not display. The TAMS decoder definitions don't draw their own ``<pane>`` elements but - at least this one - use a standard template (look under ``xml.programmers.parts.MotorPane``). You might fix it by using a name that is in the template, or someone else might jump in and tell you how to add it somewhere else.

bobjacobsen, 2018-01-21:
@silverailscolo's comment goes to the heart of the problem.  There are two pieces involved:  

- The "variable" definition, which tells DecoderPro what's in the decoder
- The definition for the programmer panes, which tells DecoderPro what to display where.

If you look at the standard [motor pane definition](https://github.com/JMRI/JMRI/blob/master/xml/programmers/parts/MotorPane.xml), you'll see what DecoderPro is looking for to display there.  If you say `item="EMF Option 1"` (if that's not already used), it should display.

For more info, see [this writeup](http://jmri.org/help/en/html/apps/DecoderPro/Programmer.shtml#id)

moba-klytos, 2018-01-21:
Thanks for showing the MotorPane part, using an option did the trick.
I will close this issue now.
/Uwe

closed 2018-01-21 by moba-klytos