JMRI/JMRI #4805: Missing LocoNet messages

bobjacobsen, 2018-01-21:
In PR #4793, @Pugwash1 said:

> My loconet monitor shows packets that jmri does not show. 

There have been multiple report of JMRI missing messages from the LocoNet, but so far it hasn't been understood what's going on, let alone getting it fixed.

So I think we should track this down if we can.

Is there any pattern to what's being lost?  What type of messages, what's happening around them, what else is happening in JMRI, anything else?

devel-bobm, 2018-01-22:
Recently, this has been Digitrax "expanded slot" messaging.  A while back, there were a couple of reports of CML's signalling messages.  Neither of these have been defined by the manufacturer.

Pugwash1, 2018-01-23:
I have been poring over logs and it looks like whenever a packet goes "Missing" the next packet occurs twice. Which means that the first packet is overwritten by the second packet, either in the code or the PR3?
so on the Arduino LocoNet Monitor we see:   
 20328 | RX | A0 10 00 4F 
         4 | RX | B5 10 03 59  <--- The 4 indicates 4 mSec after the previous packet.
In RcvHandler and LocoNet Slot Mon we see
18:53:47.464: [B5 10 03 59]  ........
18:53:47.469: [B5 10 03 59]  ..........

bobjacobsen, 2018-01-23:
I'm checking to see whether that's an artifact in the JMRI LocoNet Monitor itself (i.e. the overwriting is in the display), but so far I don't see it.  Hard to imagine a way that could happen in JMRI's receive code, but of course it's possible.  

Worst case, this is somehow in the USB transmission from the PR3 into the OS USB support; we probably won't be able to find that. 

@Pugwash1 Do you have a LocoBuffer? It would be interesting to see if it behaves the same way.

Pugwash1, 2018-01-23:
In LnPacketizer the correct packets go out on the ostream, but the second packet comes back twice on the istream. 
Driving the pr3 thru a terminal, direct drive, all good packets are echoed back by the pr3, not the software driver (which doesnt know a good loconet packet from a bad one).
Its starting to look a lot like LnPacketizer or PureJavaCom or both. ps LnPacketizer seems to have redundant code.

Pugwash1, 2018-01-23:
No I don't have a LocoBuffer. But give me a day or two and I could get an arduino to look like a pr3.

devel-bobm, 2018-01-23:
@Pugwash1 , you wrote:
```
I have been poring over logs and it looks like whenever a packet goes "Missing" the next packet occurs twice. Which means that the first packet is overwritten by the second packet, either in the code or the PR3?
so on the Arduino LocoNet Monitor we see:
20328 | RX | A0 10 00 4F
4 | RX | B5 10 03 59 <--- The 4 indicates 4 mSec after the previous packet.
In RcvHandler and LocoNet Slot Mon we see
18:53:47.464: [B5 10 03 59] ........
18:53:47.469: [B5 10 03 59] ..........```

What are you reporting here?  Is this LocoNet monitor information or info from the console log with defaults.lcf changed to DEBUG or TRACE for LnPacketizer and SlotMon?  If so, be aware that there can be non-trivial delays between when a LocoNet packet is received by the hardware, when it is forwarded by USB to the computer, when the packetizer "sees" it and when a given listener is notified of its arrival.  This is especially true of "underpowered" computers and those computers which are particularly slow because of malware and virae and those actively running `Facebook` and other web-based tomfoolery.

It would help to see more context here, as it is impossible to tell what is happening.  Are you really looking at `LocoNet Slot Monitor`, and not `LocoNet Monitor`?  What is the full "context" of the messages, including

If you are not reporting `LocoNet Monitor`, what does `LocoNet Monitor` show for the same LocoNet traffic as being reported?

It would help to have a longer string of messages, so that a one-to-one correspondence could be established!

Note that it is possible for `LocoNet Monitor` to display "nothing at all" for a LocoNet message - no timestamp and no raw data and no "explanation".  This happens in a poorly-coded case for a given OpCode or a sub-set of an OpCode.  Several cases have been found in the past, and there may be other cases which have been missed so far.  A longer string of LocoNet messages, and an independently-logged (i.e. Arduino?) LocoNet message traffic may be helpful.

devel-bobm, 2018-02-03:
Another thing that may be happening - some LocoNet interface hardware has difficulty forwarding LocoNet traffic to the computer if LocoNet spec timings are not honored.  

When one LocoNet message begins "too soon" after the end of a previous LocoNet message, the LocoNet interface hardware may neglect to "see" the second message, so the computer never "sees" the second message.

Some 3rd-party and Digitrax hardware devices are known to violate the "back-to-back" LocoNet spec timings.  In some cases, the out-of-spec messages are not forwarded to the computer by the LocoNet interface device.

In these cases, it is best to fix the hardware which is producing the offending LocoNet timings.

Pugwash1, 2018-02-10:
Hi
Its not perfect yet but https://github.com/Pugwash1/JMRI.git branch LocoNetStuff has been working for me for a while now. 
It doesn't send the next loconet message until the "echo" from the PR3 et. al. has been received for the previous message. It doesn't send the next loconet message until it receives the LACK for a previous message that requires one. It takes a tea break on BUSY or IMM Busy messages. It re-transmits IMM rejects. Running 50 NMRA signal heads, and 6 autotrains for 3+ hours without a problem.
As an aside it doesn't send throttles active until they go in-use, and doesn't null move slots for a steal or re-aquires. It loads a 20 throttle layout no problem. It loads the slot monitor at 50 per second. And the loconet debug send does actual send at 0 intervals, really stacking the queue.
Its ugly, but appears to work.

devel-bobm, 2018-02-10:
Why is it necessary to put sequencing and delays and holds in JMRI code when the LocoNet interface is supposed to be able to handle them and has been able to handle them for more than a decade?  It seems to me that the level of change that you describe may not be addressing the root cause of your problem.  And it places a non-trivial burden upon those of us who don't have "lost message" problems!

I haven't seen any log evidence of "missing" messages, so I am skeptical that these code changes are needed.

What is the LocoNet hardware environment? Are the user-updatable devices updated to latest firmware?  What messages are being sent when the problem is seen?  How do you know a message is missing?  How do you know that a message which appears twice?  

How do you know that your Arduino code isn't missing messages or duplicating messages?  How is your Arduino hardware interfacing to LocoNet?  Is it using a UART receive implementation or a bit-banged (ICP interrupt pin) receive implementation?  Are you using a roll-your-own LocoNet implementation or are you using the MRRWA library?  If MRRWA, which version and how have you modified it?  Are you using any sleep or wait methods in the sketch?  Do you have any other functionality implemented in the Arduino sketch that could use processor cycles such that the LocoNet receive code would be blocked when the critical "Start bit" and "Opcode indication bit" would be received?  Do you have any Serial.* method invocations in interrupt code or LocoNet receive code?

And, back to your log snippet:
```
20328 | RX | A0 10 00 4F
4 | RX | B5 10 03 59 <--- The 4 indicates 4 mSec after the previous packet.
In RcvHandler and LocoNet Slot Mon we see
18:53:47.464: [B5 10 03 59] ........
18:53:47.469: [B5 10 03 59] ..........
```

As I asked before, what are you reporting here?  If one line is reported by `LnPacketizer` and one is from `LocoNet.slotmon`, this is entirely reasonable.  If both lines are being reported by `LnPacketizer`, then I would want more information, to include what hardware interface is being used to generate the log - the Arduino or some other interface?

Do you have any Oscilloscope traces of the LocoNet data line that show whether there are any out-of-spec LocoNet timings that may be influencing the situation?

Pugwash1, 2018-02-11:
Good morning Bob, another lovely day in paradise.
"I haven't seen any log evidence of "missing" messages, so I am skeptical that these code changes are needed."  Well, do you have a log of the messages that were supposed to be there? And how many times have someone question when a signal or switch isn't set right by Dispatcher? How many times has a train not stop? A train go backwards instead of forwards? Blame the WiFi throttle? Chalk it up to a miss pressed button? The scripts that have to send the speed twice?

I would suggest that you don't use autotrains much. Its a slightly different game. We have to rely on messages going thru, there is no one  standing there to make a correction, another push of a button or twitch of a slider, to cause a resend, we just crash. 

With out JMRI running you can run the echotest script against a PR3 and against a DCS240 internal connection and see the differences. Its linux, I don't know how to do a similar thing in Windows.

[echotest.txt](https://github.com/JMRI/JMRI/files/1714458/echotest.txt)

The PR3 echos back the last message twice, the DCS240 echos the 2 separate messages.

Both these messages are processed by the command station but Slots in JMRI are now incorrect as they are updated by the echo. On the next keep alive refresh the previous speed will be set. Waiting for the echo of a message before proceeding validates the message, an invalid message is not echoed.

The way messages are ignored by the CS is that we do not process BUSY and LACK messages.
If a LACK is expected then wait for it, because if it doesn't arrive something has to be done. If a BUSY arrives don't send another message for a while, if you do it is ignored, not missing but not received. If the LACK says it was rejected it then resend it after giving some time for the CS to recover. I don't think Digitrax invented these messages for fun.

As to your response to the IMM Reject problem, that was not correct. The failure to process the message and resend is a JMRI problem not a Digitrax problem, the CS correctly told us it was busy, try later.  And telling them to change there hardware to fix it, change the software would be a better fix. As for clogging the system, the DCS240 can keep 400 locos going, so a few NMRA packets aren't going to be a problem. The North Coast boys are quite smart, I doubt they would but out a device that is that flawed. I expect they are laughing there guts out right now.

As for what is being reported, its 2 good packets seen by one monitor and the sending JMRI monitor showing one packet twice.

devel-bobm, 2018-02-11:
@Pugwash1 ,

You wrote:
```
With out JMRI running you can run the echotest script against a PR3 and against a DCS240 internal connection and see the differences. Its linux, I don't know how to do a similar thing in Windows.

echotest.txt

The PR3 echos back the last message twice, the DCS240 echos the 2 separate messages.
```

Since I don't have Linux, I cannot run the same thing.  And since you haven't provided a log of PR3 or DCS240 activity from your Linux tests, I cannot do anything but guess at how to interpret your statement of results.

@Pugwash1 also wrote:
`The failure to process the message and resend is a JMRI problem not a Digitrax problem, ...`

I completely disagree here.  Any device which sends the OPC_IMM_PACKET is supposed to pay attention to the possibility of a LONG_ACK (Fail) code.  But Digitrax throttles do not!  I've seen it with DT40x throttles!)  

Is JMRI supposed to make up for the failings of the DT40x throttles?  That isn't reasonable.  Just like it isn't reasonable for JMRI to try to make up for the failings of LocoNet devices which issue "turnout control" LocoNet messages but do not pay attention to LONG_ACK (Fail).  It doesn't work well turnouts, and it isn't going to work well for OPC_IMM_PACKET messages, either.  

You should not simply delay all transmit activity after sending a LocoNet message which might get a LONG_ACK (Fail) message.  This will slow down JMRI's LocoNet thruput incredibly!  And if you did, how long would you wait for that potential LONG_ACK (Fail) message?  It cannot just be a milliscond or two, because some USB implementations/drivers have a much-longer "polling" time - 16mSec is not uncommon when using a USB-RS232 converter plus LocoBuffer (or, unfortunately, USB-RS232 plus MS-100!). 

Even when the normal USB "polling" time is short (i.e. 1mSec, with LocoBuffer-USB drivers), there is no guarantee that the USB Host implementation is not overloaded with USB "Isochronous" traffic, which can change the USB scheduling for devices which normally have very good "polling" time.  This is a real-world case that needs to be considered.  

Example: USB-based speakers for "Virtual Sound Decoder" and USB-based WiFi for WiThrottleServer and USB-connected LocoNet interface, running on an early R-Pi.  There are going to be some serious USB throughput challenges there!

Similarly, slowing down JMRI's transmit activity will break JMRI's half-hearted attempt to ensure sequencing of LocoNet turnout control messages.

Pugwash1, 2018-02-15:
If we create and send a message, then it behooves us to check any response code and take appropriate action. What Digitrax does with a return code for messages they generate is their concern, also they expect a human to do their checking for them. There is no human to check with automated trains.
To ignore a "reject busy", or as dcs50's are want just "busy" messages is very silly as any message sent can be ignored.

If we do not delay until we get an expected return, then if we get a bad return, or none, how do we know which message we sent failed and needs to be resent? Logs I have checked have a very short time between send and good/bad return codes. And the next message may be worthless without the previous one. There is no point being fast and wrong.

As for your Early Pi, which runs Linux by the way, so if talking from experience you must have access to a Linux system and could try the little test.

The slowest Pi I had access to was a B-2 and with 4 WiFi throttles and 4 VSDs there was no significant difference in the current master and "my wait for responses", except that my signals were a bit messed up using the master and a turnout required more than try to get it to set... I never thought there would be a real problem, the shared bus is 350Mps and Sound, loconet and throttles combined don't get much passed 100Kps. However when attempting to add 6 autotrains to the mix, the latest master version caused a head on collision in less that 3 minutes, my version went for about 2 hours before being stopped by a derailment. 

I have noticed that we always send a switch as 2 messages, an "On" and an "Off". If you look at the output from Routes in the DCS240 or DS64 you will see it just send 1, an "On". Can that be configured in the turnout table, and if not, what about adding it?

closed 2021-04-14 by stale[bot]