JMRI/JMRI #4832: Unable to change board id for PM42 using new roster feature

PUlvestad, 2018-01-28:
When creating pm42 roster entries, I am unable to change the board ID it is stuck at 1.

If I manually edit the roster file to the correct board address it still shows in the roster as being 1 (roster index was rebuilt).

Tested with 4.11.2 and 4.11.3 using LocoNet simulator

devel-bobm, 2018-01-29:
The "basic" page for the PM42 Roster Entry states that the tool cannot work correctly if the address is incorrect when the Roster Entry is opened.  As such, the XML for the PM4x devices (and for the BDL16x and SE8c and DS64) has the address "field" marked as "information only".

This means that the user CANNOT change the roster entry's "Board Address" value using the GUI tools.  This was a specific choice I made during development.  Perhaps that needs to be revisited.

A workaround is to CREATE a Roster Entry with the correct address.  Using the "Roster", when creating the entry, do not double-click on "PM42" in the list of possible decoders.  Instead, single-click on it.  In the "sidebar" that opens, enter the correct Board Address value (Board ID).  Then save the roster entry and then you can open the roster entry.

PUlvestad, 2018-01-29:
I will try but shouldn't manually editing the roster file change it?

PUlvestad, 2018-01-29:
Your workaround worked thanks.

PUlvestad, 2018-01-29:
I compared the new file with one I made before and see I missed manually changing the address in one of the 4 locations. Line #10 in the file.

dheap, 2018-01-29:
> The "basic" page for the PM42 Roster Entry states that the tool cannot work correctly if the address is incorrect when the Roster Entry is opened.

This is also true of every decoder definition, locomotive, accessory or signal when using Ops mode programming.

The user has to change the address and then save, close and reopen. Otherwise programming fails. (See my user warning instructions on the Basic Pane of the NCE Light-It decoders).

> As such, the XML for the PM4x devices (and for the BDL16x and SE8c and DS64) has the address "field" marked as "information only".

We don't lock the address field(s) for every other decoder definition, where the same issue arises if using Ops mode (the only mode available for some decoders).

PUlvestad, 2018-01-31:
Spoke too soon.
I tried the roster entries I made at home with simulator at the club and when I pulled up any of the files, they all showed an ID of 1 (none of our pm42's use 1, we use 2-5).
When I read the file that uses ID 5, it appeared to read correctly but when I saved the results it changes the CV 1 to 1 as well as any other location the ID number is stored.
The results were the same for 2 others (the last one wasn't tried as it has known issues that require at Florida vacation).
I did not try writing as we have a big OP session planned for Saturday.

closed 2021-04-14 by stale[bot]