JMRI/JMRI #4842: Dispatcher AutoActiveTrain not honouring speed limits

Pugwash1, 2018-02-02:
Automatic trains under dispatcher not honouring speed limits set for route when running with signal masts.

mstevetodd, 2018-02-02:
I checked on my simulator setup and the block and turnout speeds seem to be working fine. Note that these are set in Tools. Tables, Turnouts (enable Show Turnout Speeds) and Tools, Tables, Block (column Speed).

Pugwash1, 2018-02-02:
I just ran it and it worked on one route but not another so I am looking into it. It may be the way the layouts done. 
Steve G.

On February 2, 2018 8:50:39 AM EST, SteveT <notifications@github.com> wrote:
>I checked on my simulator setup and the block and turnout speeds seem
>to be working fine. Note that these are set in Tools. Tables, Turnouts
>(enable Show Turnout Speeds) and Tools, Tables, Block (column Speed).

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

Pugwash1, 2018-02-02:
It appears that we get the route speed from the signal mast aspect being displayed, so if it cannot display the speed that is detected for the route it substitutes something else, normally "approach" which equates to "medium", even though the speed is "restricted"

Pugwash1, 2018-02-03:
@mstevetodd  What happens is that when we convert a speed to a throttle for non-profiled (the loco has no profile or use of profile has not been checked) we divide the speed for the route by the maxlinespeed for that route to create a percentage throttle. We should divide by the max speed specified in warrents else the  percentage throttle for a given route between two mast varies for a given speed. I can set up layout where "Restricted slow" as the slowest speed in a route varies between 35% to 64% depending on the maximum speed for a block in that route.

Pugwash1, 2018-08-14:
From User:
In the code, setSpeedBySignal is used when the train enters a section to determine and set a new target speed. The software compares the speed dictated by the next signal aspect to the maximum speed allowed for the next signal’s path and chooses the lesser speed. Then this "speed”, the variable in the code, is compared to "prevSpeed", a saved value of "speed" which was determined when the train entered the last section. The lower value of these two is chosen as the target speed. The value of "prevSpeed" is then reset to the speed just calculated.
 
A problem occurs in multi-block sections because setSpeedBySignal is invoked each time a block is entered. After the first block "prevSpeed" has been reset to the maximum speed allowed for the next section. It no longer is the maximum speed allowed for the current section. In the case of a train being in a section with a limited path speed and a clear aspect on the controlling signal, the train speeds up to normal speed after it enters the second block, even though it is still in the speed limited section.

With SML, only use setSpeedBySignal when a section is entered rather than when every block is entered.

closed 2018-09-14 by Pugwash1