JMRI/JMRI #4846: Media Image Drag& Drop broken

MARCFOUR, 2018-02-03:
Windows 10,  JMRI 4.11.2 and 4.11.3.   The Roster Media image Drag & Drop  feature is broke in 4.11.2 and 4.11.3.  User unable to complete Drag&Drop of any media (JPG, PNG, GIF)  from a folder to  Roster Image frame.  Greeted with a Windows red circle of denial.  4 incidents reported

KenC57, 2018-02-10:
Right at the start of JMRI you will see this in the log:

2018-02-10 09:09:31,704 dnd.FileDrop                          DEBUG - FileDrop: Drag and drop is not supported with this JVM [main]

No idea when that started or why. But it does prevent adding media to roster entries.

MARCFOUR, 2018-02-10:
Which  log ?  I viewed the log of my system and do not see any such entry.  Windows here if it matters.   JAVA 1.8.0_161 with JMRI 4.9.5 does not show this D&D issue but same JAVA version with 4.11.3 does. Might that not indicate the problem is within JMRI itself.

KenC57, 2018-02-10:
Where I saw the message I sent was right in the first 10 lines of the
console log.
-ken c

MARCFOUR, 2018-02-10:
2018-02-10 11:52:40,742 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2018-02-10 11:52:40,789 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Marc\JMRI\log\messages.log [main]
2018-02-10 11:52:40,789 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Marc\JMRI\log\session.log [main]
2018-02-10 11:52:40,805 apps.AppsBase                         INFO  - DecoderPro version 4.11.3ish+jenkins+20180202T0146Z+R9762e1f starts under Java 1.8.0_161 on Windows 10 x86 v10.0 at Sat Feb 10 11:52:40 EST 2018 [main]
2018-02-10 11:52:47,086 gui3.Apps3                            INFO  - Starting with profile Digitrax_Loconet_Simulator.3ee94b4e [main]
2018-02-10 11:52:47,696 node.NodeIdentity                     INFO  - Using jmri-001195D4C000-3ee94b4e as the JMRI Node identity [main]
2018-02-10 11:52:52,024 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [Thread-7]
2018-02-10 11:52:55,071 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files\JMRI\ [main]
2018-02-10 11:52:55,071 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Marc\JMRI\Digitrax_Loconet_Simulator\ [main]
2018-02-10 11:52:55,071 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Marc\JMRI\Digitrax_Loconet_Simulator\ [main]
2018-02-10 11:52:55,071 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Marc\JMRI\ [main]
2018-02-10 11:52:55,071 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Marc\ [main]
2018-02-10 11:52:55,086 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files\JMRI\jython\ [main]

MARCFOUR, 2018-02-10:
IS it possible you see different because of the JVM.. ( I read about Hotspot, if any relation)

MARCFOUR, 2018-02-10:
Could that simply indicate that the " JVM "  itself does not support D&D.

MARCFOUR, 2018-02-10:
JVM is what in this example/case   ?   Is JVM = JAVA 1.8.0_161 ?   then it makes no sense as JMRI 4.10 has no D&D issue under this same JVM.

bobjacobsen, 2018-02-11:
Note that the message below is a DEBUG message:  Most people won’t see it.

The message is emitted if supportsDnD() returns false in jmri.util.iharder.dnd.FileDrop

That in turn is a static method that looks for a class called “DndConstants”.  That should be present, see https://docs.oracle.com/javase/8/docs/api/java/awt/dnd/DnDConstants.html

I’m not sure that this message is the same problem that Marc is reporting.

> On Feb 10, 2018, at 7:55 AM, Ken Cameron <notifications@github.com> wrote:
> 
> Right at the start of JMRI you will see this in the log:
> 
> 2018-02-10 09:09:31,704 dnd.FileDrop DEBUG - FileDrop: Drag and drop is not supported with this JVM [main]
> 
> No idea when that started or why. But it does prevent adding media to roster entries.
>

MARCFOUR, 2018-02-12:
For what it is worth. Windows 10, 32 bit. I turned  on /DEBUG and /NOISY. Do not see any message go by re " FileDrop: Drag and drop is not supported  "  in any log generated.  

CommandLine: "C:\Program Files\JMRI\LaunchJMRI.exe" /debug /noisy apps.gui3.dp3.DecoderPro3
Default options: -J-Djogamp.gluegen.UseTempJarCache=false
AppName: DecoderPro3
Class: apps.gui3.dp3.DecoderPro3
Parameters: 
Noisy: 1
Force32bit: 0
Profile: 
Testing for x64...
Result: 0
JavaPath: C:\Program Files\Java\jre1.8.0_161\bin\java.exe
Create folder: C:\Users\Marc\AppData\Local\Temp\LaunchJMRI
CopyFiles: C:\Program Files\Java\jre1.8.0_161\bin\java.exe C:\Users\Marc\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
Result: 0
JExePath: C:\Users\Marc\AppData\Local\Temp\LaunchJMRI\DecoderPro3.exe
PhysicalMemory: 2047m
Check if more than 4GB memory
Less than 4GB
Check if more than 1GB memory
More than 1GB
InitHeap: 96m
MinMemory: 192m
MaxMemory: 1023m
Checking maximum heap size...
Running x86 JVM
...finished
Checking user.home...
Reading Desktop Shell Folder registry key...
...read: C:\Users\Marc\Desktop
Checking if equal to %USERPROFILE%...
Comparing: C:\Users\Marc
to: C:\Users\Marc
user.home OK

Removed all remaining messages...

KenC57, 2018-02-12:
/debug is an option to the launcher. It does not set the log level to debug
in the program.

You need to edit the default.lcf file to enable debug for all of jmri.

-ken c

MARCFOUR, 2018-02-12:
Learning....   So I changed default.lcf to request DEBUG, Al, T, R   . If I did it right, I still do not see any "not supported" or anything related to " drop " message in the 862 line log.  I took the CLIPBOARD and pasted it to notepad++ and searched for "DROP". No hit.
- - -
2018-02-12 17:43:05,491 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2018-02-12 17:43:05,601 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Marc\JMRI\log\messages.log [main]
2018-02-12 17:43:05,601 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Marc\JMRI\log\session.log [main]
2018-02-12 17:43:05,663 apps.AppsBase                         INFO  - DecoderPro version 4.11.3ish+jenkins+20180202T0146Z+R9762e1f starts under Java 1.8.0_161 on Windows 10 x86 v10.0 at Mon Feb 12 17:43:05 EST 2018 [main]
2018-02-12 17:43:05,695 util.FileUtilSupport                  DEBUG - Attempting to find resources/logo.gif in [] [main]
2018-02-12 17:43:11,288 apps.AppsBase                         DEBUG - Config file set to: DecoderProConfig3.xml [main]
2018-02-12 17:43:11,288 util.Log4JUtil                        DEBUG - initLog4J already initialized! [main]
2018-02-12 17:43:11,882 util.FileUtilSupport                  DEBUG - Using C:\Users\Marc\JMRI\Digitrax_Loconet_Simulator/ [main]
2018-02-12 17:43:11,913 util.FileUtilSupport                  DEBUG - Using C:\Users\Marc\JMRI\Digitrax_PR3_DCS100/ [main]
2018-02-12 17:43:11,913 util.FileUtilSupport                  DEBUG - Using C:\Users\Marc\JMRI\Digitrax_PR3_Programmer/ [main]
2018-02-12 17:43:11,929 util.FileUtilSupport                  DEBUG - Using C:\Users\Marc\JMRI\Digitrax_LBSerial_DCS100/ [main]
2018-02-12 17:43:11,929 util.FileUtilSupport                  DEBUG - Using C:\Users\Marc\JMRI\ [main]
2018-02-12 17:43:11,960 profile.ProfileManager                DEBUG - Using config file C:\Users\Marc\JMRI\DecoderProConfig3.properties [main]
2018-02-12 17:43:12,195 profile.ProfileManager                DEBUG - Looking for profile Digitrax_Loconet_Simulator.3ee94b4e, found Digitrax_Loconet_Simulator.3ee94b4e [main]

silverailscolo, 2018-02-12:
On macOS 10.13.3 High Sierra with Java 8 Update 161 I see the same behavior when trying in vain to drag a PNG or JPEG image to the Roster Media rects (DEBUG on for path jmri.jmrit.roster):
```
2018-02-13 00:18:39,225 swing.ResizableImagePanel             DEBUG - Image path is now : null [AWT-EventQueue-0]
2018-02-13 00:18:39,228 dnd.FileDrop                          DEBUG - FileDrop: Drag and drop is not supported with this JVM [AWT-EventQueue-0]
```

KenC57, 2018-02-12:
Here is where the warnings is showing in my log. About 10 lines down with
full debug. I repeated the test using a normal build of 4.11.2 and achieved
the same message about drag/drop not being supported.

2018-02-12 18:39:30,255 util.FileUtilSupport                  DEBUG -
Attempting to find resources/logo.gif in [] [main]
2018-02-12 18:39:30,399 util.Log4JUtil                        INFO  - ******
JMRI log ******* [main]
2018-02-12 18:39:30,400 util.Log4JUtil                        INFO  - This
log is appended to file: C:\Users\kcameron\JMRI\log\messages.log [main]
2018-02-12 18:39:30,402 util.Log4JUtil                        INFO  - This
log is stored in file: C:\Users\kcameron\JMRI\log\session.log [main]
2018-02-12 18:39:30,414 apps.Apps                             INFO  -
PanelPro version 4.11.3ish+kcameron+20180210T2106Z+R9a15198262 starts under
Java 1.8.0_161 on Windows 10 amd64 v10.0 at Mon Feb 12 18:39:30 EST 2018
[main]
2018-02-12 18:39:30,438 util.Log4JUtil                        DEBUG -
initLog4J already initialized! [main]
2018-02-12 18:39:30,446 util.Log4JUtil                        DEBUG -
initLog4J already initialized! [main]
2018-02-12 18:39:30,449 util.FileUtilSupport                  DEBUG -
Attempting to find resources/logo.gif in [] [main]
2018-02-12 18:39:30,510 dnd.FileDrop                          DEBUG -
FileDrop: Drag and drop is not supported with this JVM [main]
2018-02-12 18:39:30,519 swing.FontComboUtil                   DEBUG -
Prepare font lists... [PrepareFontListsThread]

closed 2018-02-13 by bobjacobsen

MARCFOUR, 2018-02-13:
I also have a second system and it is also running Windows 10 Home N 1709 and  JAVA 1.8.0_161. It is running JMRI 4.10 at the present time. The drag and drop works perfectly. So I submit the issue is not from JAVA 1.8.0_161.  And Ken. I am running a 32 bit OS here if it matters any.

MARCFOUR, 2018-02-13:
I now see a tentative fix in 4.11.3 was done.

PR #4893 (2018-02-13): Fix Drag and Drop of Files
Remove no longer needed checks for Java 1.0 and 1.1 compatibility and cleanup Javadocs.
Fixes #4846.