JMRI/JMRI #4848: Bad OpenLCB table performance in JMRI 4.11

bobjacobsen, 2018-02-03:
From a [JMRI-developers thread]():

I have tried to use 4.11.2 at David Parks' layout and got into some serious slowness when trying to scroll in the Turnouts table.
I am attaching the XML file that produces the problem. I stripped away everything to the minimum needed to produce the problem. Here are the ingredients:
- a lot of turnouts
- a lot of OpenLCB sensors
- open the panel file from https://sites.google.com/site/balazsracz/jmri-1/cw-turnouts-and-sensors.xml?attredirects=0&d=1 (sorry too big to attach)
- open the turnouts table
- tick "show feedback information" checkbox in the bottom (if it is not checked the problem goes away)
- try to scroll down or up, page down/page up or resize the window

What I see is that it takes about 30 CPU-seconds to scroll one page down. During these 30 seconds JMRI is unresponsive to any UI action, so the AWT thread seems to be blocked by some computation.

Does anyone know what changed between 4.11.2 and 4.11.1 that might have caused this slowness?

thanks,
Balazs

bobjacobsen, 2018-02-03:
There are several things going on here to make this very slow:
- [ ] The OlcbAddress class uses regular expressions to check & parse all accesses to the String address form. That's slow.
- [ ] JmriBeanComboBox is calling for a new list of the available beans as part of each update. That's slow.
- [ ] JmriBeanComboBox is doing more updates than needed.  That's slow.
- [ ] AbstractManager is sorting the entire list of beans on each request for the list. That's slow.

So there's more than enough slow to go around.  I'll fix at least some of these (probably as multiple PRs) this weekend, but probably not all of them.

bobjacobsen, 2018-02-04:
PR #4849 fixes the 1st item.

rhwood, 2018-02-11:
Would it make sense to make JmriBeanComboBox use a type generic that extends NamedBean instead of making it only use Strings?

bobjacobsen, 2018-02-11:
"make JmriBeanComboBox use a type generic" - definitely yes. There's at least one place that'd make things cleaner, see the cry of desperation in the comment at line 679.

But first there needs to be some clarity in what it's doing with the UI.  I'm going to post to jmri-developers about that shortly to try to figure it out.

The class also needs a better (more custom) DataModel.  It now uses the default model to hold two things:  The (occasionally huge) list of String names, perhaps filtered by a custom filter, plus optionally a blank item, plus the usual info on portion and selection from a List model.  There are two significant problems:  
- We create a lot of JmriBeanComboBox items, all of them duplicating the list of names; in e.g. a BeanTable, that becomes effectively a O(N*N) storage issue when it should be O(1) by using the existing list in the manager. (See my `bj-abstractmanager-performance` branch for work toward making the manager lists easily available; ProxyManagers make it a bit tricky)
- It should be working with NamedBeans, not Strings, to simplify a lot of the code. C.f. the switch statements throughout.

So more work will follow this...

bobjacobsen, 2020-08-19:
Still working on this, expect deprecations will be complete in the current cycle.

closed 2021-04-14 by stale[bot]