JMRI/JMRI #4851: Operations Error Dialog causes test failures

rhwood, 2018-02-04:
The `jmri.jmrit.operations.ExceptionDisplayFrameTest` causes CI tests to hang and timeout on an intermittent basis.

This timeout is now one of the most common, if not the most common, reason existing CI tests do not just pass. I have tried to fix the tests and the underlying code, but have been unable to do so.

DanielBoudreau, 2018-02-17:
Test don't fail on my IDE

pabender, 2018-02-17:
@DanielBoudreau the fact that it doesn't fail on your IDE doesn't mean there isn't a problem.

Part of the reason we run the tests on the CI servers is to make sure tests compile and run on multiple platforms.

I'm checking in code that adds a timeout to the tests in the test class, but I don't know that this will completely solve the issue.

DanielBoudreau, 2018-02-17:
Paul,
I like your solution to add a global timeout, is there a reason we don't do this for every test case that has to wait for a response?  And I should have said "Not able to reproduce test failure on my IDE".

    @Rule
    public Timeout globalTimeout = Timeout.seconds(10); // 10 second timeout for methods in this test class.

rhwood, 2018-02-17:
The timeout still fails the test, so it ensures total failure does not occur, but random failures on the Linux GUI servers will continue.

rhwood, 2018-03-16:
OP is no longer concerned.

closed 2021-04-14 by stale[bot]