JMRI/JMRI #4873: Are throttles independent?

bobjacobsen, 2018-02-07:
@devel-bobm raised points in the discussion for #4870:

> For example, in my work with LocoNet throttles, I've found that, in the "big scheme of things", there can be more than one "higher-level" JMRI throttle controlling a single LocoNetThrottle object (i.e. a single LocoNet-controlled loco or consist). So, when a "LocoNetThrottle" object is "released", it can have an impact on more than just the JMRI "higher-level" throttle which issued the "release". If two JMRI GUI throttles have control of a given loco address, then ideally, releasing the loco from one of the JMRI GUI throttles does not "release" the loco from the other JMRI GUI throttle. In the LocoNet case, the GUI throttle "release" action will actually make a change to the associated "slot", but that change will not have any effect on the second GUI throttle's ability to control the loco.
> 
> Bringing this back to your work: on a LocoNet-based JMRI install, will your "release" from GUI throttle #1 affect other GUI throttles which happen to have control of the same loco as was controlled by GUI throttle #1?
> 
> (I don't have the foggiest idea, as I do not truly understand how the higher-level JMRI "throttle" implementations manage multiple throttles working with identical loco addresses.)

With reply from @SteveRawlinson:
> I use Loconet too, and have seen the behaviour you describe, a release performed in once place causing another to lose control of the loco as well.
> 
> ...
>
> However, if a GUI throttle window is operating a loco which is also being controlled by a script, and the script calls releaseThrottle() then it might blat the GUI throttle too. (I'll test that tomorrow.) I suspect the same would be true if the script called throttle.release() so this is not a new issue.
> 
> Of course if you're trying to control a loco with a GUI throttle while it's also being controlled by a script you're likely to hit other issues too :)

devel-bobm, 2018-02-07:
A related question: _should_ JMRI's throttles "be independent" from each other?

The answer may be system-dependent.

With LocoNet, it is entirely acceptable to have two physical throttles controlling the same Loco (i.e. LocoNet slot) - that's what the (misleadingly-named) LocoNet "Steal" protocol is about.  JMRI allows multiple GUI (or Wi) throttles (and physical throttles) to dial-up the same loco address, and allows any of them to control the loco; any single JMRI GUI (or Wi) throttle can "release" the loco without affecting the other throttles' ability to continue to control the loco.

I am not sure about other systems, but I seem to recall having some difficulty with an NCE layout that having two throttles dialed-up to the same loco address can cause difficulties.

KenC57, 2018-02-07:
One of the concepts for JMRI and multiple throttles controlling one loco was
to automatically trigger sounds as it moved around. So it would do the grade
crossing sounds automatically as an example. Another was the concept of a
Big Brother, it would watch and send a stop to the loco if it entered a
block with a stop signal (presuming the operator wasn't watching), or maybe
do the stop if you passed the red signal, I don't recall which was discussed
more often. Last idea was a way to smoothly transition between automated
running and manual. Here I think the typical case was automatic running
would take a train to say an interchange point and then an operator would
pick it up and do the switching. Then the operator would release the train
back to the automated running and the operator would go switch something
else.

At least those were some of the ideas kicked around over the last 15 years
I've been around this. Did they ever work that way, I don't remember. Is it
super important, I don't know. But the 'auto stop' idea comes up on the list
at regular intervals. As I see it only the auto-sound mode really required
concurrent access from two higher level parts. The rest would work ok with
some form of steal and return, I think. But until lately we've haven't had
any systems that supported the idea of when a steal happened, the first
throttle would lose control. Until lately you would have concurrent but
fighting access between the two (stop-go-stop...) which seldom was really
what people wanted.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

pabender, 2018-02-07:
On 02/07/2018 11:56 AM, Bob Jacobsen wrote:
> @devel-bobm <https://github.com/devel-bobm> raised points in the
> discussion for #4870 <https://github.com/JMRI/JMRI/pull/4870>:
>
>     For example, in my work with LocoNet throttles, I've found [...]

The way this was designed to work, there would only ever be one JMRI 
object talking to the throttle on the system (this is true no matter 
what system we are talking about).  The ThrottelManager is responsible 
for returning the same object to each throttle.

Multiple Throttle Manager users (GUI objects, scripts, other parts of 
the code, etc, all of which implement the ThrottleListener interface ) 
are allowed to ask for the same address.  During the acquisition 
process, there are the possibilities:

1) There is the first user asking for an address.  The throttle manager 
does what it needs to acquire the throttle and sends an appropriate 
message through the ThrottleListener interface to the user.
2) There is the nth user asking for an address for which an acquisition 
is already in progress.  The throttle manager adds the user to the list 
of ThrottleListeners waiting for the address and returns an appropriate 
message when the acquisition
3) This is the nth user asking for an address for which the acquisition 
process is complete.  the throttle manager immediately sends the user an 
appropriate response through the ThrottleListener interface.

(There is also a provision for "single use" throttles, but I'm not aware 
of any systems that use that flag).

Once acquisition is complete, objects using the throttle should register 
as property change listeners.  Note that the actual throttle users do 
not need to be the same object that requested the throttle.  (i.e. I can 
have an object request a throttle and then create another object that 
actually uses that throttle and pass the throttle over to the using 
process).

(note that this is all described in a UML sequence diagram associated 
with the ThrottleListener interface:
http://jmri.org/JavaDoc/doc/jmri/ThrottleListener.html )
> If two JMRI GUI throttles have control of a given loco address,
>     then ideally, releasing the loco from one of the JMRI GUI
>     throttles does not "release" the loco from the other JMRI GUI [...]
<snip>
>
> With reply from @SteveRawlinson <https://github.com/steverawlinson>:
>
>     However, if a GUI throttle window is operating a loco which is
>     also being controlled by a script, and the script calls
>     releaseThrottle() then it might blat the GUI throttle too. (I'll
>     test that tomorrow.) I suspect the same would be true if the
>     script called throttle.release() so this is not a new issue. 
>

The release side of things was never as well thought out as the 
acquisition side, but that is because  LocoNet is the major system where 
this really matters.

That said,  I think some of the recent changes have corrupted the usage 
of "release" and "dispatch" in the LocoNet case (LocoNet is the primary 
example of a "dispatch" functionality.  The concept doesn't exist on 
either XPressNet or NCE.)

I've always thought of the LocoNet release button as being the 
equivalent of selecting another address on the same LocoNet physical 
throttle without dispatching the previous address.   In other words, on 
the right hand side of a DTx00 throttle, dial up loco "100", operate 
that for a while, then (on the same side of the throttle) dial up 
address "200" and operate that for a while.  To the best of my 
knowledge, doing this makes no changes to the first slot.

"dispatch" on the other hand does change the slot.

>     Of course if you're trying to control a loco with a GUI throttle
>     while it's also being controlled by a script you're likely to hit
>     other issues too :)
>

"Control" can actually take more than one form.  Many times, we just 
listen to throttles for various things.  This is the way the Virtual 
Sound Decoder works, and also the way the Throttle Bridge scripts work.  
(plus, Ken's examples).  All of these are users of the throttle.

Now, to address the concern about what to do when a particular throttle 
user calls release (or dispatch), I see two possibilities:
1) The throttle (not the throttle manager) already keeps track of the 
property change listeners.  If the object calling the release (or 
dispatch) function is registered as a property change listener, remove 
it from the list.  If that was the last property change listener, 
dispose of the throttle appropriately (release doesn't change the slot 
on LocoNet, dispatch does).
2) Elsewhere we have implemented the concept of a vetoable property 
change listener.  This simply means we ask all the property change 
listeners if taking the action is ok.  If it is, then we proceed, if 
not, we abort the requested operation.

Paul

bobjacobsen, 2018-02-07:
I have to check details, but I basically agree with Paul 100%.

When you ask for the throttle for a locomotive, that’s what you’re getting: _the_ throttle in JMRI that talks to that locomotive. Think of being handed a little box with a knob. If some other code asks for the throttle, they’re functionally sharing the same little box. They can cooperate: “oh, I’ve noticed you changed the speed, ok, I’ll change my display”. Or they can fight: “no, you can’t set speed that high, I’m setting it back”. That’s up to the code using the throttle. But functionally its the da,e throttle. 

(As an implementation issue, perhaps they get different implementation objects, but they function as one)

If we know of exceptions to this, we should track down the cause and fix it.

SteveRawlinson, 2018-02-08:
> 1) The throttle (not the throttle manager) already keeps track of the
property change listeners.  If the object calling the release (or
dispatch) function is registered as a property change listener, remove
it from the list.  If that was the last property change listener,
dispose of the throttle appropriately (release doesn't change the slot
on LocoNet, dispatch does).

In the case of something like the virtual sound decoder which 'listens in' to the throttle but doesn't make any changes, is there a danger that a throttle might end up hanging around (with the slot still in use) if we don't dispatch() because there's still a property change listener registered? (Sorry if I've completely misunderstood.)

pabender, 2018-02-08:
On 02/08/2018 06:09 AM, Steve Rawlinson wrote:
> 1. The throttle (not the throttle manager) already keeps track of the
>         property change listeners. If the object calling the release (or
>         dispatch) function is registered as a property [...]

My point above is that we can't dispose of the throttle object until 
every last listener is done with it.

In the specific case when we have listen only users of a throttle, those 
listeners SHOULD work whether or not JMRI has control of the actual 
device.  Whether that occurs because of a dispatch/pickup cycle or the 
throttle being stolen isn't a concern of the property change listener. 
Those details ARE a concern of LocoNet specific throttle objects.

So, it may very well be that a throttle object is "dispatched" but we 
still retain a listener that is watching for changes on the throttle. 
That is a valid use case that we have supported for a very long time on 
systems that provide a way to monitor throttle activity (in other words, 
not just on LocoNet, but LocoNet is the only system that really supports 
the "dispatch" concept).

Paul

SteveRawlinson, 2018-02-08:
If that was the last property change listener, dispose of the throttle appropriately (release doesn't change  the slot on LocoNet, dispatch does).

[...]

> So, it may very well be that a throttle object is "dispatched" but we
still retain a listener that is watching for changes on the throttle.

On the one hand you seem to be saying that we shouldn't 'dispatch' a throttle while there are any property change listeners registered, but on the other hand that we might 'dispatch' a throttle and retain a listener. I'm obviously missing something.

pabender, 2018-02-08:
On Thu, Feb 8, 2018 at 9:48 AM, Steve Rawlinson <notifications@github.com>
wrote:

> When you ask for the throttle for a locomotive, that’s what you’re
> getting: the throttle in JMRI that talks to that locomotive.
> If we know of exceptions to this, we should track down the cause and fix
> it.
>
> I'm not sure if this counts as an exception in the sense you meant it but
> if you request a WiThottle on a loconet system it basically yanks the
> throttle away from any GUI throttles, who lose control. That may be the
> intended behaviour but it doesn't match your description of what ought to
> happen.
>
>
If that is happening, yes, that is a problem.  The question then becomes is
the issue in the throttle objects (WiThrottle and the GUI ) or is it in the
LocoNet throttle code.

I'll try to see what happens on an XpressNet system tonight.

Paul

bobjacobsen, 2018-02-08:
> On the one hand you seem to be saying that we shouldn't 'dispatch' a throttle while there are any property change listeners registered, but on the other hand that we might 'dispatch' a throttle and retain a listener. I'm obviously missing something.

I think LocoNet language is confusing here.

Consider the case of “Somebody else is controlling this locomotive, I have no interest in changing its speed, but I want to be told when the speed changes”. 

That’s not a LocoNet hardware-throttle concept.  But it’s a valid thing for a JMRI software Throttle to do.

devel-bobm, 2018-02-09:
@bobjacobsen wrote
`
Consider the case of “Somebody else is controlling this locomotive, I have no interest in changing its speed, but I want to be told when the speed changes”. That’s not a LocoNet hardware-throttle concept. But it’s a valid thing for a JMRI software Throttle to do.
`

Actually, that _is_ a LocoNet hardware-throttle concept.  It's called "slot following", and is an implicit function of the (misnamed) Digitrax "steal" concept.  (A new exception applies, see below.)

Just because a throttle has "stolen" a loco, that throttle does not need to make any changes to the speed, direction, or functions.  Any other throttle which previously "had" the loco can still send control messages to the throttle.

This changes somewhat with the use of "expanded" slots.  Expanded slots implement the "steal-zap" feature which is enabled via expanded slot messaging.  When an expanded slot is in-use, a successful "steal" removes control of the loco from the original expanded-slot throttle, in favor of the new throttle.  

It is unclear to me if a loco in an expanded-slot can be "slot-followed" in the traditional sense.

pabender, 2018-02-09:
> On Feb 8, 2018, at 7:04 PM, Bob Milhaupt <notifications@github.com> wrote:
> @bobjacobsen wrote
> Consider the case of “Somebody else is controlling this locomotive, I have no interest in changing its speed, but I want to be told when the speed changes”. That’s not a LocoNet hardware-throttle concept. But it’s a valid thing for a JMRI software Throttle to do.
> 
> Actually, that is a LocoNet hardware-throttle concept. It's called "slot following", and is an implicit function of the (misnamed) Digitrax "steal" concept. (A new exception applies, see below.)
> 
Yes and no.

Yes because that is the expected behavior when a non-JMRI throttle steals, but no because JMRI can't ( or at least shouldn't ) steal the throttle from itself.  In other words, there is a distinction between what happens inside of a JMRI instance vs what happens when more than one hardware device is involved.

rhwood, 2018-02-09:
> On Feb 8, 2018, at 20:46, Bob Milhaupt <notifications@github.com> wrote:
> 
> I am totally confused now. Probably because of the nebuluous nature of the term "steal".
> 
> @pabender <https://github.com/pabender>,
> 
> What do you mean by ... but no because JMRI can't ( or at least shouldn't ) steal the throttle from itself?
> 
> I am hoping to end up with a good understand exactly what there is a distinction between what happens inside of a JMRI instance vs what happens when more than one hardware device is involved is supposed to mean.
> 
> What does that mean for JMRI and what does it mean for JMRI's LocoNet implementation?
> 
> I think I'm going to need example cases and expected behavior.
> 
> a) Start with all "empty" slots. A JMRI-based throttle requests and gets loco 1234. LocoNet slot is now "in-use". A second JMRI-based throttle requests and gets loco 1234. What is the intended status of the first JMRI-based throttle (I presume "still alive and still selected and still controllable")? What is the intended status of the second JMRI-based throttle (I presume alive and selected and can control”)?
> 
It should not be possible for a second JMRI throttle to be created for any given address. Once JMRI has a throttle for a given address, anything within JMRI interested in having a throttle for that address should become a listener to and, if not only passively interested, controller of that first throttle.

This means that if a JMRI WiThrottle requests address 1234, and no Throttle exists for 1234, a Throttle is created for 1234 and the WiThrottle becomes a listener to that Throttle. When a ThrottleFrame is created and requests 1234, it also becomes a listener for the existing Throttle, and both the WiThrottle and ThrottleFrame listen to and control the same Throttle.

pabender, 2018-02-09:
Sent from my iPad

> On Feb 8, 2018, at 8:47 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> I am totally confused now. Probably because of the nebuluous nature of the term "steal".
> 
> @pabender,
> 
> What do you mean by ... but no because JMRI can't ( or at least shouldn't ) steal the throttle from itself?
> 
> I am hoping to end up with a good understand exactly what there is a distinction between what happens inside of a JMRI instance vs what happens when more than one hardware device is involved is supposed to mean.
> 
> What does that mean for JMRI and what does it mean for JMRI's LocoNet implementation?
> 
You have to remember what I said earlier.  There should NEVER be more than one object that is actually talking to the hardware from within JMRI, but there can be more than one JMRI object that uses that throttle.

I am not going to answer your questions about the UI below because the user of the throttle may not have an associated UI.  ( scripts and other code are equivalent to the throttle UIs as far as this discussion is concerned )
> I think I'm going to need example cases and expected behavior.
>
> a) Start with all "empty" slots. A JMRI-based throttle requests and gets loco 1234. LocoNet slot is now "in-use". A second JMRI-based [...]
You should end up with one LocoNet throttle objects and two users in this case.   Both users have the ability to control the throttle object.
> b) Continuing from a) above, the user of the first JMRI-based throttle selects the "Release" button. What is the expected state of the first JMRI-based throttle (I presume "no address selected and ready to request a new loco")? What is the expected state of the second JMRI-based throttle (I presume "still selected, can control")? What is the expected state of the associated LocoNet slot? I expect "common"; this would be consistent with the behavior of similar operations using two Digitrax physical throttles.
> 
In this case, I expect the slot to still be in use.  

The user that released the throttle should be dropped from the property change listeners for the LocoNet throttle object.  JMRI should retain control of the slot, because there is still one user of the LocoNet Throttle object, so it should not become common.
> c) Continuing from b) above, the user of the second JMRI-based throttle selects the "release" button. What is the state of the second JMRI-based throttle (I presume "no address selected and ready to request a new loco")? What is the status of the LocoNet slot? Assuming no other throttles have the loco(slot), I presume that the slot would be "common".
> 
That one is a good question.  The answer boils down the answer to a question I asked earlier in this thread.  Here is the example I gave earlier.

Using a DTx00:
1) select loco 100 and control it for a while using the right throttle
2) select loco 200 and control it for a while using the right throttle

What is the status of the slot for 100 after I select 200 ( note that we specifically do not press the dispatch button at any time ).

If we only ever release the throttle, whatever happens to the slot for 100 in the case above should happen when the last user releases the throttle object.
> d) Continuing from a) above, the user of the first JMRI-based throttle selects the "Dispatch" button. What is the expected state of the first JMRI-based throttle (I presume "no address selected and ready to request a new loco"). What is the expected state of the second JMRI-based throttle (I presume "still selected, can control"). What is the expected state of the associated LocoNet slot? I expect "common" and "dispatched" (LocoNet slot move to slot 0); this would be consistent with the behavior of similar operations using two Digitrax physical throttles.
> 
That I could agree with.  The dispatch functionality on LocoNet has always confused me though.  The user that dispatches the throttle should go away ( at least as far as the throttle object is concerned ).  The user that did not dispatch would need to retain the ability to listen to the slot, and, if desired, take control.
> e) Start with all "empty" slots. A JMRI-based throttle requests and gets loco 1234. LocoNet slot is now "in-use". A Digitrax physical throttle requests and "steals" loco 1234. What is the intended status of the JMRI-based throttle (I presume "still alive and still selected and still controllable", as would be true all throttles in question were Digitrax physical throttles)? What is the intended status of the Digitrax physical throttle (I presume alive and selected and can control")?
> 
Yes, I believe that is correct.  
> f) Continuing from e) above, the Digitrax physical throttle "releases" the loco. What is the status of the JMRI-based throttle (I presume "still alive and still selected and still controllable", as would be true if all throttles in question were Digitrax physical throttles)? What is the expected status of the LocoNet slot? I expect "common".
> 
Since the throttle was stolen, the slot would be in whatever state the physical throttle left it in, just like if the throttle stole from another physical throttle.
> g) Continuing from e) above, the JMRI-based throttle "releases" the loco. What is the status of the JMRI-based throttle (I presume "no loco selected"). What is the status of the Digitrax physical throttle (I presume "still alive and still selected and still controllable", as would be true if all throttles in question were Digitrax physical throttles)? What is the status of the LocoNet slot (I presume "common", as would be true of all involved throttles were Digitrax physcial throttles)?
> 
We should mimic the behavior of the physical throttles.

Note that if the release never modifies the slot, you get the behavior you describe here automatically.

plocher, 2018-02-09:
Another perspective :-)

From OUTSIDE of JMRI (as viewed from some other loconet connected
device...), you have no clue/concept of any internal relationships within
JMRI's software event distribution stack:

   1. JMRI's Throttle Manager requests/gets a slot for "1234".
   - HIDDEN: Threads of control within JMRI register as event listeners on
      that throttle instance
      - HIDDEN: Threads give the throttle manager "work" to do - set speed,
      direction, ...
      - HIDDEN: JMRI's throttle manager combines all the "work" from all
      the threads
      - VISIBLE: ThrottleManager sends the combined traffic to the Loconet
      bus
   2. JMRI's Throttle Manager releases the slot /or/ dispatches the slot*

* Based on the confusion surrounding what happens when a JMRI-Throttle
user/'bot tells the throttle manager to "release" or  "dispatch", JMRI may
well be inconsistent or buggy :-)

----

What should happen?  and how does this differ from "what actually does
happen"?  IMO, JMRI should handle all throttles the same, across all
hardware platforms.  This means NOT exposing the Loconet peculiarities
(dispatch, steal...) to the throttles directly:

Assume

   - Throttle Manager does reference counting, and has a policy of
   auto-dispatching a slot on last listener de-registration.
   - The JMRI Throttle Manager is the "real" loconet throttle, and any
   listener throttles are simply internal constructs that obey the JAVA
   subscribe/unsubscribe lifecycle
   - The Loconet specific attach/release/steal/dispatch lifecycle is NOT
   exposed to the JMRI throttles directly
   - For completeness, we need to consider hypothetical passive listeners
   (JMRI-3...) that never issue throttle commands, but only listen for, and
   react to, events related to "1234".

a) JMRI-based throttle requests 1234. A second JMRI-based throttle does the
same

JMRI creates a single Throttle Manager instance handling 1234 for the first
caller, and increments the refcount to 1

The second's request simply adds a subscription to the Throttle Manager
created by the first and increments the refcount to 2

Both JMRI throttles see every event associated with 1234, and both can
actively control 1234.

(These JMRI throttles are not constrained by any exclusivity requirements)

b) Continuing,  JMRI-1 "Release"s

This should be taken as an unsubscription or de-registration of JMRI-1 from
the Throttle Manager instance for 1234

The refcount in the ThrottleManager for 1234 drops to 1

JMRI-1 no longer gets notifications of changes to 1234, and can no longer
control 1234.

JMRI-2 is still getting notifications and can control 1234

There is no change to the slot data in the command station, and no special
loconet traffic caused by this "Release", as the JMRI Throttle Manager has
not itself released anything.

c) Continuing, JMRI-2 "release"s

Another unsubscription.

No JMRI throttle listeners are registered to get any notifications, and
none can control 1234 any longer.

Throttle Manager's reference count drops to 0, so it does a Release on the
loconet, and deletes/recycles itself.

d) Continuing from a) above, JMRI-1 selects "Dispatch"

If there are other JMRI listeners (refcount > 1), then this is the same as
"release" (above).

else, this was the only listener, and JMRI could/should? perform a real
"dispatch" and slot movement dance

e) "empty"  -> JMRI-1 -> Digitrax-2 -> "steal"

same as in 'a':

JMRI creates a single Throttle Manager instance handling 1234 for JMRI-1

Digitrax-2 and JMRI-1 both are in control and see each other's
changes/notifications

f) Continuing from e) above, Digitrax-2 "releases"

JMRI-1 is still in control, but there are no more events coming in from
Digitrax-2 for it to react to

g) Continuing from e) above, JMRI-1 "releases"

JMRI-1 unsubscribes, the ThrottleManager sees the refcount drop to 0, it
"releases" and terminates (same as "c" above)

Digitrax-2 is still in control, but there are no more events coming in from
JMRI-1 for it to react to

  -John

devel-bobm, 2018-02-09:
@pabender, 
You wrote `There should NEVER be more than one object that is actually talking to the hardware from within JMRI, but there can be more than one JMRI object that uses that throttle.` 

 I am not convinced that is achievable for LocoNet.  Consider "light" versus "turnout".  Consider SE8c "signal head" versus "light" versus "turnout".  Consider how Jython scripting allows the user to directly control LocoNet resources, independantly from JMRI's infrastructure.  I'd argue that these things all break the idea of "only one object that is actually talking to the hardware".

Perhaps this point is simply a question of semantics...

rhwood, 2018-02-09:
No, we can’t control how raw packets are dumped to any system, and if a users wants to do that by a Jython script, it is out of our hands.

However, if a user wants to use a JMRI provided object (be it Throttle, Turnout, Light, etc) either through scripting, or other JMRI-managed objects (in panels, tables, etc), JMRI must only provide a single instance of that object. If there are two different objects (Lights and Turnouts) that get the same representation on the LocoNet, that does not change that JMRI has one Light and one Turnout (even if they respond to the same message from the LocoNet), especially since other protocols do make that distinction.

Therefor lights and turnouts being the same in LocoNet is an implementation detail of LocoNet, but the concept that there is only one Light per address in JMRI or one Turnout per address in JMRI remains a rule within JMRI.

Randall Wood

devel-bobm, 2018-02-09:
Continuing on the theme "semantics", let me re-phrase my set of questions.  When I type "JMRI GUI* throttle" below, please consider it to mean "a JMRI GUI throttle, a JMRI throttle listener, a JMRI automation throttle, a script using a JMRI throttle, the user-thingy of a WiTrottleServer, etc."

a) Start with all "empty" slots. A "JMRI GUI* throttle" requests and gets loco 1234. One `LocoNetThrottle` object is created. LocoNet slot is now "in-use". A second "JMRI GUI* throttle" requests and gets loco 1234, via a reference to the pre-existing `LocoNetThrottle` object. What is the intended status of the first "JMRI GUI* throttle" (I presume "still alive and still selected and still controllable")? What is the intended status of the second "JMRI GUI* throttle" (I presume alive and selected and can control")?

b) Continuing from a) above, the user of the first "JMRI GUI* throttle" selects the "Release" button. What is the expected state of the first "JMRI GUI* throttle" (I presume "no address selected and ready to request a new loco")? What is the expected state of the second "JMRI GUI* throttle" (I presume "still selected, can control")? Is the single 'LocoNetThrottle' "disposed" (I think not, as it still has a reference from the second "JMRI GUI* throttle")?  What is the expected state of the associated LocoNet slot? I expect "common"; this would be consistent with the behavior of similar operations using two Digitrax physical throttles, and would still allow the existing `LocoNetThrottle` object to control the slot.

c) Continuing from b) above, the user of the second "JMRI GUI* throttle" selects the "release" button. What is the state of the second "JMRI GUI* throttle" (I presume "no address selected and ready to request a new loco")? What is the status of the `LocoNetThrottle` object (I presume it would be "disposed" when no "JMRI GUI* throttle" users were associated with it)?  What is the status of the LocoNet slot? Assuming no other (non-JMRI) throttles have the loco(slot), I presume that the slot would be "common".

d) Continuing from a) above, the user of the first "JMRI GUI* throttle" selects the "Dispatch" button. What is the expected state of the first "JMRI GUI* throttle"  (I presume "no address selected and ready to request a new loco"). What is the expected state of the second "JMRI GUI* throttle" (I presume "still selected, can control"). What is the expected state of the `LocoNetThrottle` object (I presume still able to control the loco/slot and still with one "JMRI GUI* throttle" object attached)?  What is the expected state of the associated LocoNet slot? I expect "common" and "dispatched" (LocoNet slot move to slot 0); this would be consistent with the behavior of similar operations using two Digitrax physical throttles.

e) Start with all "empty" slots. A "JMRI GUI* throttle" requests and gets loco 1234. LocoNet slot is now "in-use". A Digitrax physical throttle requests and "steals" loco 1234. What is the intended status of the "JMRI GUI* throttle" (I presume "still alive and still selected and still controllable", as would be true all throttles in question were Digitrax physical throttles)? What is the state of the associated `LocoNetThrottle` object (I presume still extant and able to control the slot)?  What is the intended status of the Digitrax physical throttle (I presume alive and selected and can control")?

f) Continuing from e) above, the Digitrax physical throttle "releases" the loco. What is the status of the "JMRI GUI* throttle" (I presume "still alive and still selected and still controllable", as would be true if all throttles in question were Digitrax physical throttles)? What is the expected state of the `LocoNetThrottle` object (I presume still alive and connected to the slot)? What is the expected status of the LocoNet slot? I expect "common".

g) Continuing from e) above, the "JMRI GUI* throttle" "releases" the loco. What is the status of the "JMRI GUI* throttle" (I presume "no loco selected"). What is the expected state of the `LocoNetThrottle` object (I presume "disposed", since no "JMRI GUI* throttle" objects are attached)?  What is the status of the Digitrax physical throttle (I presume "still alive and still selected and still controllable", as would be true if all throttles in question were Digitrax physical throttles)? What is the status of the LocoNet slot (I presume "common", as would be true of all involved throttles were Digitrax physcial throttles)?

devel-bobm, 2018-02-09:
@pabender,

You stated, and asked:
```
That one is a good question. 

The answer boils down the answer to a question I asked earlier in this thread. Here is the example I gave earlier. 
Using a DTx00: 
1) select loco 100 and control it for a while using the right throttle 
2) select loco 200 and control it for a while using the right throttle 
What is the status of the slot for 100 after I select 200 ( note that we specifically do not press the dispatch button at any time ).
```

Here are the sequence details, key being step "i)":

a) The DTx00 user rotates right knob to get it "active"
b) The user presses "Loco"
c) Some loco number appears on the display, might be 0, might be some previous number used on that throttle on that knob.
d) The throttle requests the slot for that address; the command station replies with a slot with that address.
e) The user dials in or keys in the number "100".  (intermediate LocoNet traffic ignored for convenience)
f) the throttle requests the slot for the address "100", and the command station returns the slot for that address.
g) The user presses either "Loco" or "enter" to "accept" the address.  The throttle attempts to put the slot "in-use" and writes the ThrottleID.
h) The user presses "Loco" again, to select a new loco on the same side of the throttle.

**i) as soon as the user has done this, the throttle immediately sends a "slot status write" to change the slot status to "common" (from whatever it was before).  The throttle is NO LONGER able to control address "100".**

j) the user dials in or keys in "200".
k) the throttle requests the slot for the address "200", and the command station returns the slot for that address.
l) The user presses either "Loco" or "enter" to "accept" the address.  The throttle attempts to put the slot "in-use" and writes the ThrottleID.

Remember that the slot status "common" is pretty much identical to slot status "in-use".  In both cases, LocoNet speed, direction, and function control messages to the slot will affect the slot and will be "refreshed" on the DCC track signal.

pabender, 2018-02-09:
Sent from my iPad

> On Feb 9, 2018, at 8:35 AM, Bob Milhaupt <notifications@github.com> wrote:
> 
> @pabender,
> 
> You stated, and asked:
> 
> That one is a good question. 
> 
> The answer boils down the answer to a question I asked earlier in this thread. Here is the example I gave earlier. 
> Using a DTx00: 
> 1) select loco 100 and control it for a while using the right throttle 
> 2) select loco 200 and control it for a while using the right throttle 
> What is the status of the slot for 100 after I select 200 ( note that we specifically do not press the dispatch button at any time ).
> Here are the sequence details, key being step "i)":
> 
> a) The DTx00 user rotates right knob to get it "active"
> b) The user presses "Loco"
> c) Some loco number appears on the display, might be 0, might be some previous number used on that throttle on that knob.
> d) The throttle requests the slot for that address; the command station replies with a slot with that address.
> e) The user dials in or keys in the number "100". (intermediate LocoNet traffic ignored for convenience)
> f) the throttle requests the slot for the address "100", and the command station returns the slot for that address.
> g) The user presses either "Loco" or "enter" to "accept" the address. The throttle attempts to put the slot "in-use" and writes the ThrottleID.
> h) The user presses "Loco" again, to select a new loco on the same side of the throttle.
> 
> i) as soon as the user has done this, the throttle immediately sends a "slot status write" to change the slot status to "common" (from whatever it was before). The throttle is NO LONGER able to control address "100".
> 

Ok.
> j) the user dials in or keys in "200".
> k) the throttle requests the slot for the address "200", and the command station returns the slot for that address.
> l) The user presses either "Loco" or "enter" to "accept" the address. The throttle attempts to put the slot "in-use" and writes the ThrottleID.
> 

So here is are a few follow up questions. (I can't try these myself, I only have an anchient DT100 that barely works and a UT4D ).

What happens when you try to select the same throttle on both sides of a DTx00 throttle?  Are you required to steal?  Is the same throttleID set in the slot? 

What about consists?  What happens to the slots relative to being in use after you add a locomotive to a consist and then select another address on the left side throttle?
> Remember that the slot status "common" is pretty much identical to slot status "in-use". In both cases, LocoNet speed, direction, and function control messages to the slot will affect the slot and will be "refreshed" on the DCC track signal.
> 
I did not know that.

SteveRawlinson, 2018-02-09:
Since I seem to have got the right end of the stick w.r.t. withrottle I have another example of behaviour that might not be what we expect. 

If a script obtains a throttle using AbstractAutomaton's getThrottle, and subsequently a GUI throttle asks for it, they both share it, with changes in state being made by the script being reflected on the GUI. However, if you press 'release' in the GUI the throttle object held by the script throws NPE if any changes are made,  and getLocoNetSlot() returns null.

So Im guessing that's also not intended?

devel-bobm, 2018-02-09:
Answers to @pabender 's questions:

q 1) "What happens when you try to select the same throttle on both sides of a DTx00 throttle?"

With a DT500 and a Zephyr Xtra:

1a) The throttle requests the slot for the address.

1b) The command station replies with the slot containing the address - which was previously assigned to "the other side" and was left with status "in-use" and with the throttle's ThrottleID set.

1c) The user presses "Loco" or "Enter" to atttempt to acquire the throttle on the second knob.

1d) the throttle decides that since the slot to be "in-use", it must issue the "Steal?" message.  NOTE that the throttle did not notice that the slot's ThrottleID was IDENTICAL to its own.

1d) the user "accepts" the Steal.

1e) The throttle sends an OPC_WR_SLOT message which re-writes the same information which was previously sent when ThrottleID was written to the slot when initially acquiring the loco.  In other words, "ThrottleID" for the "left" knob is exactly identical to "ThrottleID" for the "right" knob.

q 2) "Are you required to steal?"
Yes, per step 1d), above.

q 3) "Is the same throttleID set in the slot?"

Yes, per step 1e), above.

q4) " What about consists? What happens to the slots relative to being in use after you add a locomotive to a consist and then select another address on the left side throttle?"

This gets a little more involved, with the key item being 4h), below.  Assume the throttle has no loco on the "left" knob, and has loco 4567 acquired on the "right knob".

4a) User dials up loco "5678" on the left knob.  Assuming it is not already "in-use" and is not already consist-mid or consist-sub, the throttle will mark 5678's slot as "in-use" and write ThrottleID to the slot.  (Nothing new here).

4b) User hits "MU" then "+" to add loco 5678 as a "sub-consist" member of loco 4567.

4c) Throttle sends the LocoNet "consist" message, asking that 5678 be added as a slave of 4567.

4d) The command station creates the consist, and returns 5678's slot with consist status of "consist-sub" and "in-use".

4e) Note that no LocoNet messages show up for 4567's slot, even though it will now have a consist-status of "consist-top" - this is a "blind" change by the command station; JMRI ought to track such a change; I'm not sure if it does, but I don't think there is any real problem if it does not.

4f) The throttle writes the slot with ThrottleID information; all other info in the slot remains unchanged.

4g) The user now wants to dial-up a different loco on the left knob.  He hits "Loco".

**4h) The throttle immediately sends an OPC_WR_SL_STAT message to set 5678's slot to "common" state (from "in-use" state).**

4i) The user dials-up or keys in a different loco for the left knob, getting a slot report each time the address changes.

4k) The user tries to acquire a given address by pressing "Loco" or "Enter", and the throttle goes thru the normal processing of "is it already in-use" (i.e. determine if "Steal" must be asked), set slot "in-use", 

Note that the throttle behaves a little differently when acquiring a loco that is already marked as a member of a consist (and not the top-most, controlling member, of the consist).  One would potentially need to do this to change the loco's functions, say, douse the headlamp for a loco that is in the middle of the consist.

pabender, 2018-02-09:
On Fri, Feb 9, 2018 at 11:50 AM, Bob Milhaupt <notifications@github.com>
wrote:

> q 2) "Are you required to steal?"
>
> Yes, per step 1d), above.
>

What that indicates to me is that Digitrax is really treating both sides of
the DT500 throttle you were using as though they are two independent
throttles, even though they are not independent.  That is a design
decision, and even if it is a reference example, that doesn't make it the
best design decision.

I think @bobjacobsen's analogy of the throttle manager handing the throttle
user a little control box is a better way to think about this.  If you take
that to the physical world, this would be the equivalent of two users
literally sharing the same throttle.  Sometimes those users will fight over
the throttle, sometimes they will cooperate, but ultimately, there is only
one throttle.  This is the way that JMRI should handle this intenrally.

After all this discussion, I think @plocher's comments provide the best
insight to all this,

My thought on all of this is still the same, but to summarize and restate
my opinion:

When a throttle user releases a throttle, it should be removed from the
property change listeners for that throttle.  After the user is removed
from the property change listener's list, if the list is empty, then the
JMRI should do whatever is required of the hardware to actually relinquish
control (so in the loconet case, change the slot to common) and dispose of
the throttle object.   All of the internal JMRI details are hidden from the
command station.  Ultimately, the command station doesn't really care if
JMRI has 1 or 20 users of the throttle, only that JMRI is controlling the
throttle.

Because dispatch is a context sensitive operation (it only really has
meaning in the context of a LocoNet system...), it should just remove the
listener and perform the required operation(s) to dispatch the throttle.
It should only dispose of the throttle object if there are no more property
change listeners.  The dispatch case has the specific use case of what if
we want to listen to the actions of a UT1 (do I have that right?) so I can
see the need to retain that.  (If it were not for that one use case, I
would argue we should remove the dispatch function altogether.  We
certainly don't need to expose that loconet specific detail to users).

pabender, 2018-02-09:
Just one more note, jmri.jmrix.AbstractThrottleManager already handles handing out an existing address:

https://github.com/JMRI/JMRI/blob/bcad0f61e9109be723c2ad28dcbcacec4117f664/java/src/jmri/jmrix/AbstractThrottleManager.java#L219-L226

So unless a specific Throttle Manager has overridden the requestThrottle(LocoAddress, BasicRosterEntry, ThrottleListener) method, the "hand the same box to all users" is happening now.

devel-bobm, 2018-02-10:
I believe that current JMRI/JMRI "head" code is behaving in a way that is consistent with "desires" as described by @pabender and @plocher.  If it is not, I have missed the disconnect; I will review and work to correct any shortcoming that I am notified of.

I have not yet wrapped my head around the behavior described in @SteveRawlinson 's 09Feb problem.  I would expect a 'LocoNetThrottle` to handle a "release" request to simply mark the slot as having "common" status.  But perhaps it is coded to do more than that.  It needs some thought.  

There may need to be some clarification (at least in my mind!) of the "release" request process interactions between `AbstractThrottleManager` and LocoNet-related code.  

Note that I do not expect the act of changing the LocoNet slot status from "in-use" to "common" (as part of the "release") to cause such a problem.  There must be something more going on.  

Given the tenor of the thread, I do not expect the `LocoNetThrottle` code or the `LnThrottleManager` code to have to understand anything about the many users of the "Throttle" object and behave differently when there is only one user versus more than one user.  I would expect `AbstractThrottleManager` to hide those differences from the LocoNet-related code.

bobjacobsen, 2018-02-11:
Sorry to be away for a bit, but I've read all the posts and tried to catch up.

I really do thing that the LocoNet terminology is causing problems.  LocoNet's "release" and "dispatch" operations have very specific meanings for a single, hardware throttle.  But neither one is the "dispose" that's common to JMRI objects. And I'm not sure that any of those are on-target for the discussion here.

* Disposing an object (i.e. a JFrame) says that the program, including all its parts, are done with the object: Remove resources and throw away.   
   * But `throttleReference.dispose()` is not what you want if you're saying "This piece of code is done with the Throttle, but I'm really not sure whether anybody else is": If a ThrottleManager has given the same Throttle object to multiple requestors, how are they to handle a dispose-type operation?
* Invoking a LocoNet release operation is a lot like `dispose()`, with the addition of some LocoNet-specific context.  It's an "I'm done with this (physical) throttle's connection to that decoder, and the rest is up to the command station" associated with particular command station actions (perhaps queued by LocoNet operations). 
* Invoking a LocoNet dispatch operation is all that, with more command station contextual operations.

The question at the top is "Are throttles independent?"

I think that the answer is "no, not now, but they were intended to be". What do I mean by that?

Originally (and I wrote this code over 15 years ago, so memory is dim), the idea was that there were (if possible, see below) an infinite number of fully-independent Throttle objects.  Think of an infinite rack of little boxes on an infinite wall in the layout room.  If two parts of the program asked for a throttle for decoder address 25, they could get _separate_ throttles: separate Throttle objects representing their own individual little boxes.  That way they could independently manage the lifetime of their own throttle.  Those might interact, just like two physical LocoNet DT100 (it was a long time ago) throttles would, if they were aimed at the same address. But they were separate objects.

When EasyDCC (IIRC) came along, this didn't make sense.  The EasyDCC protocol really didn't have a way for two independent-but-interacting "throttles" to work a single address the way that LocoNet did. And it was pretty clear that more systems were going to be like that than like LocoNet.  For a while, there were two parallel implementations, with a single-use flag that let using code tell whether it had one-of-many Throttles or the-one-and-only Throttle, but I think Glenn Oberhauser removed all that as part of building JMRI's graphical throttle, which was based on the shared-single-object approach that's assumed through all the discussion above.

AFAIK, LocoNet is the only system were there's a lot riding on Throttle end-of-life.  It's not a big deal for NCE, EasyDCC, TMCC; I think it's not a big deal for SPROG (though I might be wrong about that for the mode that can only drive one address) or the XpressNet-based systems; I don't know anything recent about the [status of OpenLCB's traction](https://i.pinimg.com/564x/f4/c6/43/f4c6431353723320dd8742b6b7459b3d.jpg) protocol. The issue for LocoNet is that slots are a scarce resource in many cases, so Throttles should release them as soon as they can. And that's a problem for JMRI, because it's not easy to determine "as soon as they can".

One way to do it is to centralize operations in the ThrottleManager:  Remove release(), dispatch() and dispose() from Throttle, and have those operations require talking to the ThrottleManager _with_ a clue to which request is now terminated.  Could use the ThrottleListener from the original request, could use some other token (including giving each requestor a different object so they can be told apart, i.e. something that's really a handle to the underlying object).  Lots of ways to do that.

Or LocoNet could go back to providing unique Throttles when asked, which mediate their own interactions.  That'd work too.

pabender, 2018-02-11:
Sent from my iPad
> On Feb 11, 2018, at 12:58 AM, Bob Jacobsen <notifications@github.com> wrote:
> AFAIK, LocoNet is the only system were there's a lot riding on Throttle end-of-life. It's not a big deal for NCE, EasyDCC, TMCC; I think it's not a big deal for SPROG (though I might be wrong about that for the mode that can only drive one address) or the XpressNet-based systems; I don't know anything recent about the status of OpenLCB's traction protocol.
> 
XpressNet throttles could work as either independent objects or as single shared objects, but shared objects are more efficient, due to the polling required when another device ( typically a hand held controller ) takes over.
> One way to do it is to centralize operations in the ThrottleManager: Remove release(), dispatch() and dispose() from Throttle, and have those operations require talking to the ThrottleManager with a [...]
I have been looking at the code, and someone made an effort a few years ago to centralize the release in the ThrottleManager, but I don't believe that work was ever truly finished. ( I looked at git blame, and remember the timeframe, but would neeed to go back to that to look at who did the work )

That code intermediates the release ( or dispatch ) and requires passing a throttle listener in, but, there are several problems with that.

1) Because throttle objects can be passed by the original ThrottleListener to a delegate, that Throttle listener may be associated with more than one throttle. ( some of the servers may do this, but I can't remember for certain, and haven't looked )
2) we are not actually keeping a mapping between throttle listeners and the throttles
3) there is no part of the throttle listener interface that actually deals with release
4) the code in question isn't doing anything with the passed ThrottleListener ( because of 3 )

I really do think a better measure of reference count is provided by the property change listeners.  Those can also be registered via the manager ( though I don't believe they must be ).

SteveRawlinson, 2018-02-11:
> I really do think a better measure of reference count is provided by the property change listeners.  Those can also be registered via the manager ( though I don't believe they must be ).

Does every throttle object in use have at least one property change listener? I don't think the throttles provided by AbstractAutomaton register one.

devel-bobm, 2018-02-11:
I'd think that `AbstractThrottleManager` ought not allow "disposal" (in the JAVA sense) of a `Throttle` if there are any `ThrottleListeners` or any 'PropertyChangeListener`s registered against that `Throttle`.

devel-bobm, 2018-02-11:
I've opened a somewhat-related issue, #4884, to address propagation of layout-level "kill the throttle(s)" operations back to `Throttle` objects.

pabender, 2018-02-11:
@SteveRawlinson 

jmri.jmrit.automat.AbstractAutomaton does not register a property change listener, but things that use the returned throttle certainly can and do.

The reason for this is that in a script, we might simply be asking for a throttle to perform one operation, and then the throttle is forgotten about by the script.

Exactly what happens there is out of our control, as are many things users do with scripts.

All we can do is make a best effort and recommend to users how to work with these objects.

pabender, 2018-02-11:
@devel-bobm 

If the throttle is being disposed and there are ThrottleListeners waiting for the throttle, then there is something wrong with the code.

Remember that ThrottleListeners are only valid while an object is waiting for a specific address.

Once a throttle is returned, several things can happen (these are not all mutually exclusive):

1. The actual ThrottleListener object may be disposed of.
2. The actual ThrottleListener object may be re-used for another request.
3. The actual ThrottleListener object may hand the throttle off to some other object or objects.

That last point is actually important.

If you've looked under the hood of the GUI throttle, you will see that it actually does  both 2 and 3.
The address panel is the ThrottleListener object.   It is retained, but just displays the address, while the throttle is active, but once the throttle is released, could be used to select another address.

The AddressPanel passes throttle objects to the ControlPanel, the SpeedPanel, and the FuncitonPanel objects (and not always the same throttle object.  There are provisions for a consist address and a regular address to be set so that the FunctionPanel and the ControlPanel to have different addresses.  This is currently only used in throttles created by the ConsistTool however).  

jmri.jmrit.automat.AbstractAutomaton does 1.  It produces a ThrottleListener that bound to a local variable in a single method, so it goes away when the method returns a value to the requesting object.

SteveRawlinson, 2018-02-11:
> jmri.jmrit.automat.AbstractAutomaton does not register a property change listener, but things that use the returned throttle certainly can and do.

But also sometimes they don't. Using the number of property change listeners as a measure of reference count might result in the disposal of a throttle object that is in fact still being used.

devel-bobm, 2018-02-11:
@SteveRawlinson , Can you give an example of an AbstractAutomation user which acquires a `Throttle` object but does not register a property change listener?

devel-bobm, 2018-02-11:
It seems that we have a sort of "anarchy" with respect to `Throttle` objects.  An invoking method can, effectively, acquire a throttle, effectively "anonymously" (i.e. without registering a property change listener).  This leaves `AbstractThrottleManager` with no way of knowing whether the `Throttle` object is in-use or may be retired.

Maybe we should give that some attention?  (Perhaps this should be handled in a separate issue?)

We already have the "property change listener" approach to "is in use by some object" - if the property change listener list has more than 0 entries, then it must be "in-use".  But that falls apart in the "anonymous user" (no property change listener) scenario.  

Perhaps we could "require" registration of a "property change listener" for the duration that a `Throttle` object is being used?  The listener doesn't need to do anything, just be added at "begin of use" and removed at "end of use".  AbstractThrottleManager could "dispose" of any `Throttle` object which doesn't have any "property change listeners" registered.  If the "user" code doesn't add a "property change listener" when acquiring a `Throttle` object, then the user code is at risk of losing controllability/snoop-ability.

pabender, 2018-02-11:
Sent from my iPhone

> On Feb 11, 2018, at 12:53 PM, Bob Milhaupt <notifications@github.com> wrote:
> 
> @SteveRawlinson , Can you give an example of an AbstractAutomation user which acquires a Throttle object but does not register a property change listener?
> 
https://github.com/JMRI/JMRI/blob/master/jython/BackAndForth.py

And most of the rest of the example scripts we distribute that use a throttle.

pabender, 2018-02-11:
Sent from my iPad

> On Feb 11, 2018, at 1:02 PM, Bob Milhaupt <notifications@github.com> wrote:
> Perhaps we could "require" registration of a "property change listener" for the duration that a Throttle object is being used? The listener doesn't need to do anything, just be added at "begin of use" and removed at "end of use". AbstractThrottleManager could "dispose" of any Throttle object which doesn't have any "property change listeners" registered. If the "user" code doesn't add a "property change listener" when acquiring a Throttle object, then the user code is at risk of losing controllability/snoop-ability.

I have been thinking about this point today, and I am actually thinking more along the lines that we should try not to actually release the throttle object until the throttle is able to be garbage collected.

Since we have who knows how many user created scripts that we don't control that run throttles, requiring a user script to register as s property change listener provides an unnecessary burden on the user and represents a change that isn't backwards compatible with existing user files.

We still want throttle objects that have property change listeners to dispose of those listeners, and when the throttle manager sees no lire property change listeners, I think it should remove any references it has to those throttles, which will may allow that throttle to be garbage collected, but only if there are no other references.

bobjacobsen, 2018-02-11:
Can we set aside the history and decide from scratch what's needed? There seem to be two choices:

1) Truely independent throttles:  If there are multiple requests within JMRI, there are multiple throttles. Each is responsible for its cleanup, either manually (the program requests it) or as part of finalization (i.e. if it gets garbage collected).  Then it's up to the using code to use them efficiently.

2) Shared throttles:  When you request one, you get a handle so the throttle manager can determine when no more are in use.  (For LocoNet:  the code doesn't automatically have the ability to release, which is not a problem AFAIK, or dispatch, which might be)

3) Both

Personally, I think (2) is the right choice; I'm willing to accept limitations and/or special-case handling for Digitrax systems.

pabender, 2018-02-12:
Sent from my iPad

> On Feb 11, 2018, at 5:31 PM, Bob Jacobsen <notifications@github.com> wrote:
> 
> Can we set aside the history and decide from scratch what's needed? There seem to be two choices:
> 
> Truely independent throttles: If there are multiple requests within JMRI, there are multiple throttles. Each is responsible for its cleanup, either manually (the program requests it) or as part of finalization (i.e. if it gets garbage collected). Then it's up to the using code to use them efficiently.
> 
> Shared throttles: When you request one, you get a handle so the throttle manager can determine when no more are in use. (For LocoNet: the code doesn't automatically have the ability to release, which is not a problem AFAIK, or dispatch, which might be)
> 
> Both
> 
> Personally, I think (2) is the right choice; I'm willing to accept limitations and/or special-case handling for Digitrax systems.
> 

I agree, I think 2 is the right choice.

SteveRawlinson, 2018-02-12:
Definitely 2.

SteveRawlinson, 2018-02-12:
>  if you request a WiThottle on a loconet system it basically yanks the throttle away from any GUI throttles

I was wrong about this, it was fixed 5 months ago. WiThrottle plays nice.

SteveRawlinson, 2018-02-12:
I'm going to put this on record before somebody else works it out.

In 2011 @kevin-dickerson modified AbstractThrottleManager to deliver an existing throttle when another request is made for the same address. He tracked the number of 'users' of the object by incrementing a count (and decrementing it when it's released). It basically does exactly what we want in option 2 per Bob's list.

There was a problem with loconet throttles raised in Nov 2015 in issue #159 where previously released addresses were getting slots not set to 'in-use'. 

In Jan 2016 some nitwit decided that the problem was with with the use count in AbstractThrottleManager being incremented even when a new throttle wasn't needed. He suppressed the increment when an existing throttle was used . Since there's only ever one throttle per address that count never goes above 1 and AbstractThrottleManager has lost its notion of the number of users of a throttle.

The actual problem is that LnThrottleManager's release() implementation does not bother to consult the use count in its parent class before setting the slot status to COMMON.  

I will unfix the fix and fix it properly this time. Since WiThrottle is behaving and this will fix the only other noted exception to the desired behaviour we seem to have reached a consensus on, we might be sitting pretty soon.

(The fact that suppressing that count didn't break a lot more things might suggest that other (non-loconet) implementations might not be using it either.)

pabender, 2018-02-13:
Sent from my iPad

> On Feb 12, 2018, at 11:34 AM, Steve Rawlinson <notifications@github.com> wrote:
> In Jan 2016 some nitwit decided that the problem was with with the use count in AbstractThrottleManager being incremented even when a new throttle wasn't needed. He suppressed the increment when an existing throttle was used . Since there's only ever one throttle per address that count never goes above 1 and AbstractThrottleManager has lost its notion of the number of users of a throttle.

That is some good detective work.  Thanks for digging this out.  So the first step is to put that back in. 
> The actual problem is that LnThrottleManager's release() implementation does not bother to consult the use count in its parent class before setting the slot status to COMMON.
>
> I will unfix the fix [...]
Thanks.
> (The fact that suppressing that count didn't break a lot more things might suggest that other (non-loconet) implementations might not be using it either.)
> 
That may be true, but I can also tell you that other systems behave considerably differently than Loconet systems do.

I just checked, the XpressNet throttle manager’s releaseThrottle does nothing, disposeThrottle calls the super method in the superclass to check if the dispose can continue ( dispatch has no meaning on XpressNet...).  Kevin added that as part of his throttle refactoring effort back in 2011.  I suspect you will find this on other systems as well.

Paul

devel-bobm, 2018-02-13:
Removing that "suppression" seems to make LocoNet behave better, at least when using "Tools->Throttles->New Throttle" to open multiple throttles for the same loco.  For this level of testing, it appears to be acting with "option 2" behavior.

With that suppression removed from `AbstractThrottleManager`, if I have multiple GUI throttles all controlling the same address, and Close one of the GUI throttles, I don't see any LocoNet traffic.  Once I Close the last GUI throttle, I see the LocoNet message to mark the slot as "Common".

Similarly, with that suppression removed, if I have multiple GUI throttles all controlling the same address, and "Release" one of them, I don't see any LocoNet traffic as a result of that "Release" operation.  Once the last GUI throttle is "Release"d, I see the LocoNet message to mark the slot as "Common".

This is similar to, but not identical to, Digitrax throttle behavior when multiple physical throttles have acquired the same Loco address and then "release" them sequentially.  The Digitrax throttle will always set the slot to "Common" when releasing the loco address, regardless of whether other throttles have acquired the address.

closed 2021-04-14 by stale[bot]