JMRI/JMRI #4884: AbstractThrottleManager cannot propagate "kill the throttle" events from layout to Throttle users

devel-bobm, 2018-02-11:
If some action (from JMRI or from an external source) causes a `Throttle` to become unusable at the layout-system-level, there does not appear to be any mechanism for informing `AbstractThrottleManager` of this case, and there does not appear to be any mechanism for `AbstractThrottleManager` to inform the `Throttle`'s users of same.

This can be a problem with on LocoNet-based systems.  The JMRI user can "clear" slots via the Slot Manager. Or the user can issue the DT500 "Slot Zero" function.  Or the "Loco Reset" button on the DCS240 or DCS210 can be pushed.  

When any of these situations are observed by LocoNet code, there is no way to inform `AbstractThrottleManager` or any of its managed `Throttle` objects that they no longer have control of (or visibility of) the loco(s) they previously had control of (or visibility of).

I envision that some sort of new `ThrottleListener` event and some sort of new `PropertyChangeListener` event could be added to inform `Throttle` users and listeners to know of these events.  Where existing code uses `Throttle` objects, but does not understand the new events, behavior would not be any different from current behavior.  It would be good to add support for the new events to existing `Throttle` use code, to provide the information to the user, where appropriate.

pabender, 2018-02-11:
@devel-bobm 

I don't think most systems generate any kind of event that can be propagated.

For XPressNet, there is an XpressNet command to delete a Command Station stack entry, but the process of recreating that entry is trivial (The next command send for a locomotive address will recreate it).

(The Lenz Command Station serves the purpose of holding the information that must be refreshed periodically on the rails.  This is the model assumed on XpressNet systems.)

I highly recommend against using the throttle listener for this purpose.  The throttle listener represents objects that are currently waiting for a new throttle.  Once that throttle is delivered, the ThrottleManager removes that from it's list.

The ThrottleManager does maintain a list of throttles, and through that list, could trigger a property change event.

devel-bobm, 2018-02-11:
@pabender, 
With the `ThrottleListener` lifetime limited to the initial "acquisition" time, and not repeated upon subsequent additional "acquisitions", the `ThrottleListener` need not be hardened against (informed of) these events; it would be sufficient to harden `LnThrottleManager`.  

If `LnThrottleManager` sees an indication of a slot getting "killed", it could simply respond to the requestor via the `LnThrottleManager.notifyRefused` method.

That would resolve the issue during loco acquisition, but would still need to be resolved at the `AbstractThrottleManager` level and the `Throttle` object user's level.

devel-bobm, 2018-02-11:
I just found that this is a more-important issue to "solve" than I had previously thought.  For some Digitrax command stations, JMRI's "shared" `Throttle` premise will work ok.  But for other Digitrax command stations, it will fail.

Details:
When a Digitrax throttle "releases" a loco (i.e. user presses "Loco" button on the throttle, when the active "side" has an active Loco), it sends OPC_SLOT_STAT1 with the slot status of "Common".  

- Working case: 
With a DB150 and DCS51 command stations, the slot is changed to the "Common" state.  This is confirmed by a subsequent OPC_RQ_SL_DATA of the slot.  When the "slot" is in "Common" state, the slot's speed/direction/function controls are passed to the DCC track signal and track signal investigation.

- Failing case: 
With a DCS100, DCS240 and DCS210 command stations, the OPC_SLOT_STAT1 request to place the slot in "Common" state is implemented by the command station as the "Idle" state.  When the "slot" is in "Idle" state, the slot's information is neithr passed to nor refreshed on the DCC track signal.  This is confirmed by a subsequent OPC_RQ_SL_DATA of the slot and track signal investigation.

Summary:

This means that, with the DCS100, DCS240 and DCS210 command stations, if _any_ throttle "releases" or "dispatches" its loco, _all_ JMRI users of a `Throttle` object related to that loco are no longer valid.  Any attempt by a JMRI `Throttle` user  to control the slot's loco will "fall on deaf ears" and will not reach the loco.  But JMRI will still see the echo of any related Loconet messages it sends and think it is in control. 

Perhaps there is a command station OpSw setting where the DCS100/DCS240/DCS210 doesn't exhibit this behavior.  I will check with Digitrax...

icklesteve, 2019-05-26:
Hi @devel-bobm , 

This can be closed as in #6932 there's the forceDisposeThrottle(LocoAddress la) method in AbstractThrottleManager for situations where a normal Throttle release is not possible,

Steve.

devel-bobm, 2019-05-26:
I will need to examine #6932 before closing this, to ensure that it covers the specific boundary-conditions which are problematic to me.

closed 2021-04-14 by stale[bot]