JMRI/JMRI #4886: LoadLibrary error 87

ewstoub, 2018-02-11:
I'm running JMRI version 4.6-R81496dc with Java version 1.8.0_161 (en_US) on my HP Pavilion using Windows 10 Pro. I can launch and run DecoderPro without a problem using the built-in graphics card connected to a monitor.

When I disconnect that monitor from the PC video card and instead connect the PC to a "pluggable USB 3.0 docking station", with its own monitors etc, launching DecoderPro yields the little splash screen, then the Set Active Profile dialog is presented, followed after a few moments by an Error dialog stating "LoadLibrary failed with error 87: The parameter is incorrect".

I'd sure like to see DecoderPro fixed to handle this situation.

KenC57, 2018-02-12:
Everett S,

Right after you see this, and the exit from JMRI (the window is gone),
please check the JMRI/log/session.log file. We need to see that. I may give
us much better clues of what you are seeing.

-ken c

KenC57, 2018-02-12:
A web search says this is a graphic driver issue with your computer. What I'm reading would say that if you run a number of other programs that use intensive graphics will give you the same error. So this has nothing to do with JMRI itself.

ewstoub, 2018-02-12:
OK, thanks. It may well be a Java JRE problem then.

rhwood, 2018-02-17:
This appears to most likely be an interaction between the JRE and the drivers for the graphics chipset on your computer. Please ensure you have current drivers for all devices on your system.

danielb987, 2020-08-19:
I'm closing this issue since there has been no comment on the problem for over two years.

closed 2020-08-19 by danielb987