JMRI/JMRI #4897: `LocoNetThrottle` can be disposed even though prop change listeners attached

devel-bobm, 2018-02-14:
`LnThrottleManager` will disposee a `LocoNetThrottle` even though there are `PropertyChangeListener`s attached to the throttle.

(problem noted by @SteveRawlinson as part of pull-request #4894)

devel-bobm, 2018-02-14:
This problem applies to both `LnThrottleManager.disposeThrottle()` and `LnThrottleManager.releaseThrottle()`.  Both of these probably can be "protected" against a non-zero length list of `PropertyChangeListener`s.

It is unclear what actions are appropriate in the case of non-zero length `PropertyChangeListener` list for `LnThrottleManager.dispatchThrottle()`.  Will leave this for future consideration...

devel-bobm, 2018-02-14:
**BEWARE**: With current `AbstractThrottle` and `AbstractThrottleManager` implementations, if the user of an object of a class which extends `AbstractThrottle` does not register any `PropertyChangeListener`s against that object, and some other user of that object registers one or more `PropertyChangeListener`, and, at some later time, all `PropertyChangeListener`s of the object, then `AbstractThrottle` automatically disposes of the object.  

This is done by `AbstractThrottle` invoking `*ThrottleManager.disposeThrottle()` with the `ThrottleListener` parameter set to  `NULL`.  This seems to cause `AbstractThrottleManager` to dispose the throttle without regard to the number of throttle users.

So, if another throttle user exists (which hasn't registered a `PropertyChangeListener`), its throttle could disappear unexpectedly if some other agent removes the last `PropertyChangeListener` associated with that throttle.

SteveRawlinson, 2018-02-14:
> and some other user of that object registers one or more PropertyChangeListener, and, at some later time, all PropertyChangeListeners of the object ... then AbstractThrottle automatically disposes of the object.

I have a feeling there are some words missing in that paragraph where I've inserted the ellipsis. I can't work out from the context what they are.

pabender, 2018-02-14:
@devel-bobm 
please point us to the part of the code that causes an automatic dispose if there are no ProertyChangeListeners.

SteveRawlinson, 2018-02-14:
> LnThrottleManager will dispose a LocoNetThrottle even though there are PropertyChangeListeners attached to the throttle.
> (problem noted by @SteveRawlinson as part of pull-request #4894)

This is not the problem I noted, by the way. The issue I mentioned is not related  to PropertyChangeListeners.

devel-bobm, 2018-02-14:
@pabender, perhaps I am mistaken about the source of the "automatic dispose" when there are no property change listeners.  I infer this behavior from the following code and log, from a testcase I am working on in LnThrottleManagerTest.

I have manually added notes to the lot to highlight a few lines that are related to my inference.

```
    @Test
    public void testCreateLnThrottleRunAndReleaseWithPCLScenario2() {
        ThrottleListener throtListen = new ThrottleListener() {
            @Override
            public void notifyThrottleFound(DccThrottle t) {
                throttle = t;
                log.error("created a throttle");
            }

            @Override
            public void notifyFailedThrottleRequest(jmri.LocoAddress address, String reason) {
                log.error("Throttle request failed for " + address + " because " + reason);
                failedThrottleRequest = true;
            }

            @Override
            public void notifyStealThrottleRequired(jmri.LocoAddress address){
                // this is a never-stealing impelementation.
                InstanceManager.throttleManagerInstance().stealThrottleRequest(address, this, false);
            }
        };

        tm.requestThrottle(1205, throtListen);

        Assert.assertEquals("address request message",
                "BF 09 35 00",
                lnis.outbound.elementAt(lnis.outbound.size() - 1).toString());

        LocoNetMessage cmdStationReply = new LocoNetMessage(new int[] {
                0xe7, 0x0e, 0x11, 0x00, 0x35, 0x0, 0x0, 0x7, 0x0, 0x09, 0x0, 0x0, 0x0, 0x53});
        lnis.sendTestMessage(cmdStationReply);
        Assert.assertEquals("null move",
                "BA 11 11 00",
                lnis.outbound.elementAt(lnis.outbound.size() - 1).toString());

        cmdStationReply = new LocoNetMessage(new int[] {
                0xe7, 0x0e, 0x11, 0x30, 0x35, 0x0, 0x0, 0x7, 0x0, 0x09, 0x0, 0x0, 0x0, 0x00});
        lnis.sendTestMessage(cmdStationReply);
        Assert.assertEquals("write Throttle ID",
                "EF 0E 11 30 35 00 00 07 00 09 00 71 02 00",
                lnis.outbound.elementAt(lnis.outbound.size() - 1).toString());

        cmdStationReply = new LocoNetMessage(new int[] {
                0xb4, 0x6f, 0x7f, 0x5B});
        lnis.sendTestMessage(cmdStationReply);

        Assert.assertNotNull("have created a throttle", throttle);
        Assert.assertEquals("is LnThrottle", throttle.getClass(), jmri.jmrix.loconet.LocoNetThrottle.class);
        Assert.assertEquals("throttleId is set", ((jmri.jmrix.loconet.LocoNetThrottle) throttle).slot.id(),0x171);
        jmri.util.JUnitAppender.assertErrorMessage("created a throttle");

        PropertyChangeListenerScaffold pcListener = new PropertyChangeListenerScaffold();

        Assert.assertEquals("property change listener list empty", 0,
                throttle.getListeners().size());

        throttle.addPropertyChangeListener(pcListener);
        Assert.assertEquals("no pcListener call",0,pcListener.getCallCount());
        Assert.assertEquals("property change listener list size 1", 1,
                throttle.getListeners().size());

        throttle.setSpeedSetting(0.125f);

        Assert.assertEquals("set speed to one eighth",
                "A0 11 1A 00",
[... 80 lines cut ...]
2018-02-14 16:07:50,690 loconet.LocoNetThrottle               DEBUG - getLocoAddress replying address 1205 for slot 17 [main]

Here's where I know that the dispose is actually happening:

2018-02-14 16:07:50,690 loconet.LnThrottleManager             DEBUG - disposeThrottle - throttle 1205(L) [main]
2018-02-14 16:07:50,690 loconet.LocoNetThrottle               DEBUG - getLocoAddress replying address 1205 for slot 17 [main]
2018-02-14 16:07:50,690 jmrix.AbstractThrottleManager         DEBUG - AbstractThrottleManager.disposeThrottle: 1205(L), jmri.jmrix.AbstractThrottle$1@6a1aab78 [main]
2018-02-14 16:07:50,690 loconet.LocoNetThrottle               DEBUG - getLocoAddress replying address 1205 for slot 17 [main]

Here's the message that tells me that AbstractThrottleManager is entirely ok with disposal of the throttle:

2018-02-14 16:07:50,690 jmrix.AbstractThrottleManager         DEBUG - Loco Address not found in the stack 1205(L) [main]

Here's LocoNetThrottle actually disposing the throttle:

2018-02-14 16:07:50,690 loconet.LocoNetThrottle               DEBUG - throttleDispose - disposing of throttle (and setting slot = null) [main]

Here's where LocoNetThrottle "releases" (sets the slot to "Common" slot status):
2018-02-14 16:07:50,690 loconet.LocoNetThrottle               DEBUG - throttleDispose is releasing slot jmri.jmrix.loconet.LocoNetSlot@462d5aee [main]

And, while LocoNetThrottle is "cleaning up", it stops its slot refresh timer:
2018-02-14 16:07:50,690 loconet.LocoNetThrottle               DEBUG - Stopped refresh timer for slot 17 address 1205 as part of throttleDispose [main]
```

(Note - the 2nd-to-last DEBUG message shown has been corrected in my local environment to display as shown; in JMRI/JMRI master, it says "dispatchThrottle is dispatching ..." instead of "throttleDispose is releasing ...").

devel-bobm, 2018-02-14:
@SteveRawlinson, ok, I am out-of-the-loop.

But I do not have a problem with a "common" slot being unresponsive.  What is the situation where you have that problem?  Please give the details in a new "issue", as this is the best way for me to keep focused.

pabender, 2018-02-14:
@devel-bobm here is the cause

https://github.com/JMRI/JMRI/blob/f1fb13cecd2c93e123229b15d933d2b6b06ad6a9/java/src/jmri/jmrix/AbstractThrottle.java#L435-L461

So the question now is, since we have the counter working in the throttle manager, should that if statement at line 443, and the body that follows, be removed.  I think it probably should be.

SteveRawlinson, 2018-02-15:
> But I do not have a problem with a "common" slot being unresponsive. What is the situation where you have that problem? Please give the details in a new "issue", as this is the best way for me to keep focused.

I haven't had it either for a long time and I wouldn't expect to see it until these changes are merged in. Once they are, I'll check if the problem has come back.

closed 2021-04-14 by stale[bot]