JMRI/JMRI #4903: SourceForge services still intermittent

bobjacobsen, 2018-02-16:
SourceForge’s data center move isn’t going well. 

- The JMRI website in still (mostly) offline
- Mail through the mailing lists, including JMRI-developers, has been delayed to the point that some is bouncing When it does start getting through it’s likely to in odd order with some messages missing. 

If something is urgent, you can get to (most) of the developers by replying to this message. 

More info on SF status will apparently be posted to Twitter: https://twitter.com/sfnet_ops/

bobjacobsen, 2018-02-16:
As of Friday 5:30AM California time, SourceForge reports that mailing lists are up and have caught up; there may be email coming through for about four hours or so that's been delayed at the sender.

"Project web services", which is where the JMRI web site is hosted, are still down.

For more info:
https://twitter.com/sfnet_ops
https://sourceforge.net/p/forge/site-support/new/

rhwood, 2018-02-17:
The website appears to have recovered.

Am livid that the only notice of this potential downtime was [a tweet](https://twitter.com/sfnet_ops/status/962041749030952962) and the duration of the downtime suggests a lack of testing of the new hardware before traffic was redirected.

rhwood, 2018-02-17:
The website is coming and going, so reopening.

SteveRawlinson, 2018-02-17:
http://95.172.10.202/ is up and running if you want to point the DNS there until sourceforge sort it out.

bobjacobsen, 2018-02-17:
> On Feb 17, 2018, at 8:43 AM, Steve Rawlinson <notifications@github.com> wrote:
> 
> http://95.172.10.202/ is up and running if you want to point the DNS there until sourceforge sort it out.
> 

Thank you!  Done for jmri.org et al.  The A record has a 1-hour time to live, so it might take a little while for people to see it.

SteveRawlinson, 2018-02-17:
That seems to have propagated now.

https://dnschecker.org/#A/jmri.org

I got the help section my copying it from JMRI/JMRI and it doesn't look quite right The content is there but there's things missing at the top and on the left.

If anyone wants to tell me how to fix that, or indeed just needs credentials to the vm to fix it themselves just let me know.

rhwood, 2018-02-17:
Unfortunately SourceForge recently instructed us to use a different IP address, 216.105.38.10, since that is the address that should be working after the data center migration per the message they sent us.

Randall Wood
Alexandria Software
randall.h.wood@alexandriasoftware.com
http://alexandriasoftware.com

bobjacobsen, 2018-02-17:
I'm working through the Apache2 setup on the http://95.172.10.202/ machine now.  So far, SSI seems to be up (which gets the formatting, sidebars, etc), as is redirection (which is needed for some of the hardware pages).  More to follow...

SteveRawlinson, 2018-02-17:
> Unfortunately SourceForge recently instructed us to use a different IP address, 216.105.38.10, since that is the address that should be working after the data center migration per the message they sent us.

When the website is working on that sourceforge IP, by all means switch the DNS record to it. (It's not working yet.) 

I've been involved in data centre migrations and I have every sympathy for the people involved, some of them will have had no sleep for a very long time.

bobjacobsen, 2018-02-18:
Looks like SourceForge project hosting is back up.  I've moved the DNS back and kicked Jenkins to do an update of the static content.  (Generated content, like the Javadocs and pre-formatted decoder summaries, can't yet be uploaded because the SourceForge shell service isn't yet online)

bobjacobsen, 2018-03-04:
SF.net has been offline a couple times over the last week, and I've been moving the DNS entry back and forth.  (The current back up site is a few weeks out of date, that's being worked on)  Back to SF.net just now.

closed 2018-03-26 by bobjacobsen