JMRI/JMRI #4910: Closing throttle window bug - cannot re-aquire that loco

nigelcliffe, 2018-02-18:
JMRI 4.11.2, on Raspberry PI, connected to Digitrax Zephyr DCS50 via PR3 device.  

If a throttle window which controls a loco is closed (rather than the loco dispatched/released), then the loco disappears from the Slot Monitor window.     It is now no longer possible to get control of that loco address via a JMRI throttle again (including WiThrottle server to Android client).    Control via hardware (LocoNet) throttles is fine and normal.  

Nothing on the system console indicates an error.

pabender, 2018-02-18:
@nigelcliffe

That hopefully has been fixed by some recent changes.  Can you please try the latest daily build from 

http://jmri.tagadab.com/jenkins/job/Development/job/Packages/

And let us know if you still see the problem.

nigelcliffe, 2018-02-18:
Unfortunately, the "Jenkins" site given is not working properly from my computer, and I just end up with “Download unexpectedly terminated”, or “download was interrupted”.   Tried multiple times, on different computers.    I can’t get a download from there.  

Would 4.11.3 have the fix or not ?  ( original problem noted on 4.11.2 ).

pabender, 2018-02-18:
It should be in the 4.11.3 release.

Not sure why you can’t download, the daily builds, but it is possible that is a local problem of some kind.

nigelcliffe, 2018-02-19:
4.11.3 seems to fix the problem, thanks Paul.

devel-bobm, 2018-03-03:
Per comment from @nigelcliffe, 4.11.3 fixes the problem.  Closing this issue.

closed 2018-03-03 by devel-bobm