JMRI/JMRI #4911: Failures in Jenkins "Separate Tests" job

bobjacobsen, 2018-02-18:
Jenkins runs a daily [separate tests](http://jmri.tagadab.com/jenkins/job/Development/job/Separate_Tests/) job that checks to see if the JUnit test classes can be run independently.  The results are stored in a [test_results.txt](http://jmri.tagadab.com/jenkins/job/Development/job/Separate_Tests/ws/test_results.txt) file. There are multiple intermittent failures (it sometimes succeeds!)

Most recently:
```
There were 5 failures:
1) checkGetLocoDirectionAdvanced(jmri.jmrix.nce.NceConsistTest)
java.lang.NoClassDefFoundError: Could not initialize class jmri.jmrix.nce.NceMessage
	at jmri.jmrix.nce.NceConsist.sendNceMessage(NceConsist.java:278)
	at jmri.jmrix.nce.NceConsist.sendNceBinaryCommand(NceConsist.java:274)
	at jmri.jmrix.nce.NceConsist.killConsist(NceConsist.java:269)
	at jmri.jmrix.nce.NceConsist.dispose(NceConsist.java:61)
	at jmri.jmrix.nce.NceConsistTest.tearDown(NceConsistTest.java:92)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:33)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runners.Suite.runChild(Suite.java:128)
	at org.junit.runners.Suite.runChild(Suite.java:27)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:137)
	at org.junit.runner.JUnitCore.run(JUnitCore.java:115)
	at org.junit.runner.JUnitCore.runMain(JUnitCore.java:77)
	at org.junit.runner.JUnitCore.main(JUnitCore.java:36)
	at jmri.util.junit.TestClassMainMethod.main(TestClassMainMethod.java:27)
2) checkContainsAdvanced(jmri.jmrix.nce.NceConsistTest)
java.lang.NoClassDefFoundError: Could not initialize class jmri.jmrix.nce.NceMessage
	at jmri.jmrix.nce.NceConsist.sendNceMessage(NceConsist.java:278)
	at jmri.jmrix.nce.NceConsist.sendNceBinaryCommand(NceConsist.java:274)
	at jmri.jmrix.nce.NceConsist.killConsist(NceConsist.java:269)
	at jmri.jmrix.nce.NceConsist.dispose(NceConsist.java:61)
	at jmri.jmrix.nce.NceConsistTest.tearDown(NceConsistTest.java:92)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.runners.model.FrameworkMethod$1.runReflectiveCall(FrameworkMethod.java:50)
	at org.junit.internal.runners.model.ReflectiveCallable.run(ReflectiveCallable.java:12)
	at org.junit.runners.model.FrameworkMethod.invokeExplosively(FrameworkMethod.java:47)
	at org.junit.internal.runners.statements.RunAfters.evaluate(RunAfters.java:33)
	at org.junit.runners.ParentRunner.runLeaf(ParentRunner.java:325)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:78)
	at org.junit.runners.BlockJUnit4ClassRunner.runChild(BlockJUnit4ClassRunner.java:57)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
	at org.junit.runners.ParentRunner.runChildren(ParentRunner.java:288)
	at org.junit.runners.ParentRunner.access$000(ParentRunner.java:58)
	at org.junit.runners.ParentRunner$2.evaluate(ParentRunner.java:268)
	at org.junit.runners.ParentRunner.run(ParentRunner.java:363)
	at org.junit.runners.Suite.runChild(Suite.java:128)
	at org.junit.runners.Suite.runChild(Suite.java:27)
	at org.junit.runners.ParentRunner$3.run(ParentRunner.java:290)
	at org.junit.runners.ParentRunner$1.schedule(ParentRunner.java:71)
[... 202 lines cut ...]
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:109)
	at java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:184)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

Time: 0.769

OK (1 test)
```

rhwood, 2018-02-18:
These tests are disabled on Travis CI for the same reason. I have not been able to figure out why this is happening.

pabender, 2018-02-19:
It really concerns me that those NCE tests are failing with a class not found exception.  That implies something is not right in the linkng process.

pabender, 2018-02-19:
The ClassNotFoundException is being triggered by the dispose method in the NceConsist class.  I don't know why this is generating a ClassNotFoundException, but I think I know how to eliminate it.

I've submitted PR #4913 that I think does the right thing here for the NCE Consist tests.

The dispose method in the NCE Consist was calling a private method called killConsist which was causing the consist to send a message to the command station instructing it to delete the consist from it's memory (at least I think that is what it did.  @KenC57 @DanielBoudreau can you confirm?).

The problem with that is that with that in place, if my interpretation is correct, we are violating the purpose of the jmri.Consist and the jmri.ConsistManager.  The intention for those interfaces, when the command station provides methods to do this, is that we will manipulate consists on the command station so that they can be passed back and forth between JMRI and handheld throttles.

If the consist is being deleted from the command station's memory whenever we dispose of our internal representation of that, then the handheld throttles cannot access and manipulate those consists.

Also, the comment on the dispose method implies we are cleaning up our internal representation, but if that involves deleting something on the command station, then we are no longer keeping this local.

dheap, 2018-02-19:
@pabender I'ver just done some on-layout test to determine exactly what the `NceBinaryCommand.LOCO_CMD_KILL_CONSIST`does at both command station and track packet level.

Tea is served, so I'll be back with a detailed report later.

dheap, 2018-02-19:
@pabender
`sendNceBinaryCommand`(invoked by `killConsist`) is a private method that sends an `0xA2 <addr_h> <addr_l> 0x11 0x0` (kill consist) command to the NCE command station.

The command station processes this in the following way:
- The 'Lead Loco' and 'Rear Loco' columns of the internal ConsistTable are searched for a match to 'addr'. This occurs irrespective of whether 'addr' is a long (bits 6 & 7 are set in addr_h) or short address.
- If a match is found in any row (x), row x in the table is cleared _and_ an NMRA Consist Control 'set consist number of address x to 0' instruction  is sent to the track bus.
- If no match is found && 'addr' is a  short address, an NMRA Consist Control 'set consist number of address addr to 0' instruction is sent to the track bus anyway.
- If no match is found && 'addr' is a long address, the command is discarded.

Hope this clarifies what happens in the command station.

dheap, 2018-02-19:
There are a few odd things going on in NceConsist:
- NceConsist.dispose() certainly removes the consist from both command station memory and CV19 in all locos with a matching consist address - they don't have to be in the table.
- NceConsistManager.delConsist invokes NceConsist.dispose() to do part of its job.
- NceConsist.sizeLimit() incorrectly returns 6.
The NCE command station allows an unlimited number of mid-consist locos. You can just keep adding them and the appropriate NMRA Consist Control command is sent. Only the first four are recorded in the table, but you can tell the command station to remove any loco from a consist. It will simply send the appropriate NMRA Consist Control command to the loco address AND delete that address from any row in the table IF it finds a match. We aren't doing that correctly.

dheap, 2018-02-19:
> @pabender If the consist is being deleted from the command station's memory whenever we dispose of our internal representation of that, then the handheld throttles cannot access and manipulate those consists.

If Consist is only an representation of a consist, maybe in the NCE case it should be passing all the work up to a single ConsistManager instance and it alone should take care of all consist tracking and communication with the command station. Otherwise local representations will quickly get out of sync with the command station.

pabender, 2018-02-19:
Sent from my iPad

> On Feb 19, 2018, at 4:50 AM, Dave Heap <notifications@github.com> wrote:
> 
> There are a few odd things going on in NceConsist:
> 
> NceConsist.dispose() certainly removes the consist from both command station memory and CV19 in all locos with a matching consist address - they don't have to be in the table.
That is actually fine.
> NceConsistManager.delConsist invokes NceConsist.dispose() to do part of its job.
Ok, I. That case, the kill consist actually does need to be in place, but probably in the consist manager, not the consist itself.  The consist can’t tell the difference between “this con
> NceConsist.sizeLimit() incorrectly returns 6.
I am aware of that.  I did not put that limitation in place.

pabender, 2018-02-19:
Sent from my iPad

> On Feb 19, 2018, at 6:27 AM, Dave Heap <notifications@github.com> wrote:
> 
> @pabender If the consist is being deleted from the command station's memory whenever we dispose of our internal representation of that, then the handheld throttles cannot access and manipulate those consists.
> 
> If Consist is only an representation of a consist, maybe in the NCE case it should be passing all the work up to a single ConsistManager instance and it alone should take care of all consist tracking and communication with the command station. Otherwise local representations will quickly get out of sync with the command station.
> 
This doesn’t get out of sync as often as you would think.  In practice, most layouts tend to either use the handheld throttles or JMRI to manipulate consists and not both.  Sometimes you will see layouts switch, but usually only between sessions.

From a design perspective, my preference is that if a consist needs to be refreshed from the command station, it should be doing that for itself.  In this case, all that would mean is setting off the existing consist read thread periodically.

Paul

KenC57, 2018-02-19:
Paul,

My thought would have the specific details in the command station verified
(if possible) when you select the consist in the tool. Then you know if you
are in sync or not. The option would be effectively asking the user which is
the one they want: current command station or saved file version of the
consist. From a user view this would be reasonable. But I'm not sure how
well this would be from the code side. I've not had time to look at those
for a long time.

-Ken Cameron, Member JMRI Dev Team
www.jmri.org
www.fingerlakeslivesteamers.org
www.cnymod.com
www.syracusemodelrr.org

pabender, 2018-02-19:
Sent from my iPhone

> On Feb 19, 2018, at 10:42 AM, Ken Cameron <notifications@github.com> wrote:
> My thought would have the specific details in the command station verified
> (if possible) when you select the consist in the tool.

We do defer to the command station in cases where both JMRI's stored file and the command station know about the same consist. 

Paul

pabender, 2018-02-20:
I have updated PR #4913 so that it no longer removes the killConsist method, but it is now package protected instead of private so the tests can override.

If the only place dispose is called is when the NceConsistManager removes the consist, then that is ok.

pabender, 2018-02-20:
In PR #4920, @rhwood disabled the MergePromptTest.  But this shows something interesting....

I have a branch that is setup to update the test benchmarks, used on Travis and Appveyor.

The current goal is to add a check that there is at least one line of tested code in every source file.

I pulled in master after that PR was merged, which automatically kicks off Travis and Appveyor builds.

The appveyor build has not finished yet, but on the Travis build ( https://travis-ci.org/pabender/JMRI/jobs/343837065. ) jmri.jmrt.logix.MergePrompt is NOT listed as being uncovered.

That means that some other test is building and/or using something in the file.  An educated guess as to why the MergePromptTest has been intermittently failing is that some other occurrence of the window is on the screen when the test starts, and the Jemmy operator is identifying the wrong one.  The dialog box itself is modal, so that would mean this has to be occurring while the first window is closing ( the modal nature of the dialog is the reason there is a thread created by the test. )

closed 2020-01-08 by bobjacobsen