JMRI/JMRI #4925: Odd Loconet slot behaviour

SteveRawlinson, 2018-02-21:
I'm testing a DR5000 command station and this issue might have more to do with that than JMRI.

    11:06:33.402: [B5 12 13 4B]  Write slot 18 with status value 19 (0x13) - Loco is Not Consisted, Common and operating in 128 speed step mode.
    11:06:33.497: [B4 35 7F 01]  Unable to parse LocoNet message.
    contents: B4 35 7F 01
    11:06:33.657: [BA 12 00 57]  Mark slot 18 as DISPATCHED.
    11:06:33.657: [E7 0E 12 33 40 00 00 07 08 1E 00 00 00 66]  Report of slot 18 information:
    	Loco 3904 is Not Consisted, In-Use, operating in 128 SS mode, and is moving Forward at speed 0,
    	F0=Off, F1=Off, F2=Off, F3=Off, F4=Off, F5=Off, F6=Off, F7=Off, F8=Off
    	Master supports LocoNet 1.1; Track Status: On/Running; Programming Track Status: Available; 
    STAT2=0x08, ThrottleID=0x00 0x00 (0).

The first issue is the 'unable to parse Loconet message'. As far as I can see that message is valid: LONG_ACK (B4), for opcode B5 (35), accepted (7F). Is it perhaps that opcode B5 does not expect a LACK?

The second problem is that the slot is apparently still in use.

devel-bobm, 2018-02-21:
I don't think that either of these concerns has a direct impact upon JMRI.

Per LocoNet Personal Use Edition 1.0 spec, the OPC_SLOT_STAT1 message (0xB5) does not have a reply.  As such, a LONG_ACK (0xB4) reply is not interpreted in that context by JMRI's LocoNet monitor.

The fact that the DR5000 command station leaves the slot "in-use" rather than"common" or "downgraded to idle" upon a "dispatch" operation apparantly is DR5000-specific behavior, and should be taken up with the manufacturer.  There may be some specific reason for the DR5000 to behave this way.

If the slot is left "in-use", then this will be corrected in JMRI's view of the LocoNet slot when the OPC_SL_RD_DATA (0xE7) message arrives.

I find it curious, though, that STAT2=0x08.  LocoNet Personal Use Edition 1.0 does not include any definition for that encoding of STAT2.  Perhaps this is how the DR5000 encodes a "dispatched" slot?

SteveRawlinson, 2018-02-21:
Ok, just as I suspected. Thanks.

closed 2018-02-21 by SteveRawlinson